weewx/weewx #181: patch to add humidity stats picture in weewx default skin for weewx reports

yannick56, 2016-11-14:
Hello,

Severals weather stations like Davis Vantage Vue or Davis Vantage pro 2 have by default an embedded humidity sensor, but humidity stats picture in weewx reports are not available in weewx default skin. 
Humidity data are very important for farmers because with wind speed and humidity data they can choose the best time for using Pesticide/Biocide (https://en.wikipedia.org/wiki/Biocide) and reduce the quantity used for crop protection products.
Can you patch the latest weewx source code with this patch to add this improvement ?

Best regards

yannick56, 2016-11-14:
[weewx-3.6.2-humidity-stats-picture.patch.gz](https://github.com/weewx/weewx/files/589990/weewx-3.6.2-humidity-stats-picture.patch.gz)

tkeffer, 2016-11-14:
This seems like a reasonable request. 

Your patch won't work. The observation type is `outHumidity`, not `humidity`. 

But, don't worry about it. It's an easy enough change to the skin.

yannick56, 2016-11-14:
>Your patch won't work. The observation type is outHumidity, not humidity.
Strange, because this patch works fine for me with a Davis Vantage Vue and the latest weewx french standard skin for Weewx 3.5.0 ( https://github.com/weewx/weewx/wiki/weewx-french-standard-skin) .
The only change that I have included in patch for Weewx 3.6.2 is that I have added 2 lines that start with #.
For 3.6.2 skins/Standard/year.html.tmpl file, I have 
+         #if $year.humidity.has_data
+         [image]
+         #end if
 and 
for my 3.5.0 skins/Standard/year.html.tmpl file, I have ONLY:
 [image]
 
>But, don't worry about it. It's an easy enough change to the skin.
Ok, thank you.

tkeffer, 2016-11-15:
Fixed in commit https://github.com/weewx/weewx/commit/cab12d31c11745001f6b65ef168fe02cfa3558cd

I did not add `inHumidity` to the plots --- most stations do not support inside humidity.

closed 2016-11-15 by tkeffer

yannick56, 2016-11-16:
>I did not add inHumidity to the plots --- most stations do not support inside humidity.

OK.
It is perfect, the patch is enough for most users.
Thank you.