weewx/weewx #183: consider moving the examples directory and removing the experimental directory

matthewwall, 2016-11-22:
the examples directory is in the bin directory

weewx/bin/examples

that is implicitly confusing and/or misleading, since best practice says to put all customized code in the user directory.

imho, it would be more appropriately located in the same directory as extensions and util

weewx/examples
weewx/extensions
weewx/util

the experimental directory could go away (by definition, all of weewx is experimental), with its contents moved to the examples directory

helper programs (such as a tester for sending emails to verify the alarm functionality) would go in the examples directory if they are one-offs, or the util directory if they are production-ready.

tkeffer, 2016-11-29:
The examples directory is in the bin directory so a newbie can easily add a new service to `weewx.conf` without specifying a `PYTHONPATH` or copy the new service to the `user` subdirectory.

Implying a package where there really isn't one is horribly un-Pythonic but, OTOH, it's very simple to use. 

I see a couple of options:

1. Add instructions to copy whatever example you want to use to the `user` subdirectory; or
2. Package the examples as extensions.

I'd prefer the first option. The whole point of the examples is to illustrate how to write a new service. Installing as a package obfuscates this.

Agree completely about getting rid of the experimental sub-directory. The contents are pretty obsolete except for `mem.py`. For that one, we can pretty it up, add some instructions, and include it in the examples subdirectory as an example of recycling observation names (*e.g., `soilMoist1`, in this case) without extending the schema.

matthewwall, 2016-11-29:
with the move of the examples directory (and documentation about copying to user), i would like to put a few more examples in the examples directory, such as a really bare-bones driver, a bare-bones csv/xml service, and a bare-bones search list extension.

the samples in extensions are good for what they are.  however, the extension packaging adds complexity that while necessary for extensions, just gets in the way for teaching how to write a driver/service/sle.

matthewwall, 2016-12-06:
fixed at commit 7139033

closed 2016-12-06 by matthewwall