weewx/weewx #193: what to do about fake windGustDir

matthewwall, 2017-01-05:
some drivers insert a windGustDir that is simply the value for windDir: wmr100, wmr200, vantage, fousb

should we eliminate this from the driver?

if someone wants to retain this behavior, they could put windGustDir = windDir in the [Corrections] stanza

tkeffer, 2017-01-06:
I can't speak for the other drivers, but the vantage driver calculates a legit wind gust and direction, and inserts them into the LOOP packet (not archive). 

This was introduced in V2.4.0 (3-Aug-2013), but I can't for the life of me remember why.

gjr80, 2017-01-06:
I agree with the general thrust of the issue; but I think there is a bit more to the issue, especially for stations that can emit hardware archive records. For example, the Vantage stations emit highest wind speed and corresponding direction fields (ie `windGust` and `windGustDir`) in each archive record. If the user is using hardware archive generation then I believe the Vantage driver behaviour is absolutely appropriate; nothing is made up just decoded and passed on. If, on the other hand, the user is using software record generation then the loop packets only emit wind speed and direction, I believe the driver tracks these over each archive period and adds `windGust` and `windGustDir` based on the accumulated values. I could see an argument for this behaviour to be different, eg the gust info being derived from the accumulators. If this was the case I don't think `windGustDir=WindDir` correction would achieve an approriate outcome (unless the correction behaviour is altered).

tkeffer, 2017-01-06:
I'm thinking the reason why the vantage driver inserts `windGust` and `windGustDir` in the LOOP packets is for real-time displays.

But, if LOOP packets are intended to represent "instantaneous" values, that's just plain wrong. 

We can always take them out and see what happens. 1/2 :-)

Presently, the accumulators calculate `windGust` and `windGustDir`. If software record generation is specified, then those are used in new archive records. Otherwise, hardware values are used if available. If they are not available, then nothing is inserted into the archive record.

This seems like the appropriate behavior to me.

Matthew: in your research, were these values inserted into LOOP packets? Or, archive records?

matthewwall, 2017-01-06:
the drivers insert windGustDir into LOOP packets.  it looks like it was first done in wmr100, then that pattern was copied by wmr200 and fousb.  i'm not sure about the evolution of windGustDir in the vantage driver.

matthewwall, 2017-01-06:
wind gust is one of those issues that flares up from time to time in the forums.  the flareups usually happen because there is no definitive description of what wind gust means.  NWS has a definition, i think UKMET has a definition, then each type of hardware does something different.  we probably need a wind gust page on the wiki analogous to the pressure/altimeter/barometer page.

removing windGustDir will cause some support backlash when people do not see a gust dir in their real-time dials/gauges fed by LOOP packets.  if we do that we should have the rationale nailed down and the how-to-fake-data-because-your-station-cannot-provide-it workaround in place.

tkeffer, 2017-01-06:
As I said, we can take them out and see what happens. 

But, it's worth asking: what problem are we trying to fix?

matthewwall, 2017-01-06:
not really a problem, more a case of eliminating incorrectness and inconsistency.

i just noticed it while doing the sensor_map on the wmr drivers.

i removed it on the wmr100, wmr200, and fousb drivers, and noted in the changelog.  at some point i'll make a wiki page about wind speeds/gusts...

matthewwall, 2017-01-07:
also removed from the te923 driver.  wiki page has been created.

https://github.com/weewx/weewx/wiki/What-is-wind-gust%3F

matthewwall, 2017-01-07:
sometimes github can be very irritating...

https://github.com/weewx/weewx/wiki/windgust

tkeffer, 2017-01-24:
What did we decide, if anything, about whether the vantage driver should insert `windGust` and `windGustDir` in the LOOP packets?

As I said before, I have no idea why I added this. It feels incorrect.

gjr80, 2017-01-26:
As far as I know nothing was decided, the closest we came to a decision was your "let's try it and see" comment.

Just thinking aloud. Given the change to how `windSpeed` will be recorded in the archive, if `windGust` is dropped from the vantage loop packet what 'wind gust' info will the user have available? If `record_generation = hardware` then the vantage archive record will have a `windGust` field and value, this will be recorded in the archive and daily summary as required. The user will basically have the same 'data' as he had now, except that since loop packets are not contributing to `windGust`, the `maxtime` field in the `windGust` daily summary will now have no better resolution that `archive_interval` seconds. So a slight loss of granularity but data still available.

On the other hand, if the user has `record_generation = software`, the archive record will have no `windGust` field, so no `windGust` data is recorded in the archive or the daily summaries. `windSpeed` is available but that is now the true average wind speed seen across the archive period. So the user can now no longer display the highest wind speeds picked up by his station, the best he can do is, say, a five minute average. I am not sure this is a good outcome. Admittedly, it's a pretty small audience (I would expect most vantage users would use `record_generation = hardware`). To me it's almost the reverse of the fix applied for `windSpeed`.

I know this sounds bad and is overly complex, but it seems that what should be happening is that `windSpeed` should be ignoring loop `windSpeed` data but `windGust` should be using loop `windSpeed` data. 

Just looking through the hardware manual and I see a lot of stations emit a hardware loop `windGust`, that surprised me. That made me think again, is the solution just to change `windGust` and `windGustDir` to 'D' in the 'Loop' column of the `Vantage station data` table in the Hardware Guide? At the end of the day the data is being 'calculated' by the driver and is based upon all available data (admittedly only one loop `windSpeed` `windDir` value pair) so arguably not made up. Or does such a 'calculation' go into `StdWXCalculate`?

Sorry, getting a bit waffly...

tkeffer, 2017-01-26:
This is not quite how `windGust` is handled.

Unless `loop_hilo` is set to false, LOOP packet `windSpeed` *always* contributes to `windGust`. So, the max wind speed *and* it's precise time will be stored. This is true for all weather stations.

No need to waffle!

matthewwall, 2017-01-26:
i think the decision is to report only what the hardware emits.

note that some hardware reports only windSpeed and windDir.  other hardware reports windSpeed, windDir, windGust.  other hardware reports windSpeed, windDir, windGust, and windGustDir.

other than the vantage driver, the hardware.htm guide on the development branch accurately reflects which hardware emit gust info; the hardware guide at weewx.com is not accurate.

i'm still a bit confused about what should happen with accumulators/StdWXCalculate.

perhaps a matrix in the windgust page would help:

| hardware | observation | loop | archive | resulting loop | resulting archive |
| z | windSpeed | y | y | ? | ? |
| z | windDir | y | y | ? | ? |
| z | windGust | y | y | ? | ? |
| z | windGustDir | y | y | ? | ? |

matthewwall, 2017-01-26:
doh!  github markdown does not like to do anything other than headings and code!

tkeffer, 2017-01-26:
Sure it can. You just forgot the horizontal line.

| hardware | observation | loop | archive | resulting loop | resulting archive |
|----------|-------------|------|---------|----------------|-------------------|
| z        | windSpeed   | y    | y       | ?              | ?                 |
| z        | windDir     | y    | y       | ?              | ?                 |
| z        | windGust    | y    | y       | ?              | ?                 |
| z        | windGustDir | y    | y       | ?              | ?                 |

matthewwall, 2017-01-26:
hey, if i cannot blame the tool, then i have to blame *this* tool :)

tkeffer, 2017-01-26:
With the change in what max `windSpeed` means, I am satisfied with the way `windGust` and its friends are handled for the Vantage series. I recognize that while the magnitude of the wind gust comes from the hardware archive record, the timing comes from the time of the LOOP packet. It's all coming from the hardware, just from different sources.

matthewwall, 2017-01-26:
please see the tables at the end of the windgust wiki page:

https://github.com/weewx/weewx/wiki/windgust

feel free to make adjustments

gjr80, 2017-01-26:
>This is not quite how `windGust` is handled.

Seems my understanding of how things get into the accumulators is far better than my understanding of how things get out.

> With the change in what max `windSpeed` means, I am satisfied with the way `windGust` and its friends are handled for the Vantage series. I recognize that while the magnitude of the wind gust comes [...]

So I guess that vantage loop `windGust` and `windGustDir` are history...

tkeffer, 2017-01-26:
Is there a key for the wiki page as to what the symbols mean?

matthewwall, 2017-01-26:
there is now

matthewwall, 2017-04-16:
i'm pretty sure this was resolved in the 3.7 release

closed 2017-04-16 by matthewwall

gjr80, 2017-04-18:
I know this is closed but...

I notice that the `ws23xx` driver sets `windGust` and `windGustDir` to `None` whenever it generates a loop packet or archive record. Appreciate this has the same net effect as omitting `windGust` and `windGustDir` from the packets/records concerned, but should this go as well?