weewx/weewx #195: Max values for windSpeed and windGust are always equal

tkeffer, 2017-01-13:
See: https://groups.google.com/d/msg/weewx-user/wT29RgdTRHw/ecQGkjieBwAJ

tkeffer, 2017-01-13:
The following SQL query will select `max` from `windSpeed` and `windGust`, for the first 20 rows of the daily summaries:

~~~sql
SELECT archive_day_windSpeed.dateTime as dt, date(archive_day_windSpeed.dateTime,'unixepoch','localtime'), archive_day_windSpeed.max, archive_day_windGust.max
FROM archive_day_windSpeed
INNER JOIN archive_day_windGust
ON archive_day_windSpeed.dateTime=archive_day_windGust.dateTime order by dt asc limit 20;
~~~
I get:
~~~
192258800|2007-10-13|2.0|6.0
1192345200|2007-10-14|3.0|9.0
1192431600|2007-10-15|255.0|23.0
1192518000|2007-10-16|5.0|14.0
1192604400|2007-10-17|7.0|14.0
1192690800|2007-10-18|5.0|15.0
1192777200|2007-10-19|11.0|31.0
1192863600|2007-10-20|9.0|19.0
1192950000|2007-10-21|4.0|14.0
1193036400|2007-10-22|2.0|5.0
1193122800|2007-10-23|2.0|5.0
1193209200|2007-10-24|12.0|27.0
1193295600|2007-10-25|7.0|18.0
1193382000|2007-10-26|9.0|21.0
1193468400|2007-10-27|5.0|11.0
1193554800|2007-10-28|2.0|4.0
1193641200|2007-10-29|10.0|21.0
1193727600|2007-10-30|4.0|15.0
1193814000|2007-10-31|8.0|19.0
1193900400|2007-11-01|12.0|25.0
~~~
About what you'd expect, although there is a bad data point 3rd row in.

If I do the same query, but with descending order, so I'm selecting the end of the daily summaries, I get:
~~~
1483862400|2017-01-08|12.0|12.0
1483776000|2017-01-07|14.0|14.0
1483689600|2017-01-06|9.0|9.0
1483603200|2017-01-05|6.0|6.0
1483516800|2017-01-04|10.0|10.0
1483430400|2017-01-03|12.0|12.0
1483344000|2017-01-02|6.0|6.0
1483257600|2017-01-01|13.0|15.0
1483171200|2016-12-31|28.0|28.0
1483084800|2016-12-30|18.0|18.0
1482998400|2016-12-29|21.0|21.0
1482912000|2016-12-28|14.0|14.0
1482825600|2016-12-27|21.0|21.0
1482739200|2016-12-26|4.0|4.0
1482652800|2016-12-25|4.0|4.0
1482566400|2016-12-24|16.0|16.0
1482480000|2016-12-23|12.0|12.0
1482393600|2016-12-22|4.0|4.0
1482307200|2016-12-21|4.0|4.0
1482220800|2016-12-20|29.0|29.0
~~~
Here, the `max` values for `windSpeed` and `windGust` are identical.

I suspect that something happened when we switched from the standalone stats database to the daily summaries.

tkeffer, 2017-01-13:
In my own database, the values converge sometime in the first week of November, 2014, just about when V3.0 came out, which introduced the daily summaries.

Interestingly, if I rebuild the daily summaries from scratch, then they have the right answer.

gjr80, 2017-01-13:
Tom, I have been looking at the loop packets from the vantage driver and as you would expect there is `windSpeed` and `windDir` only (no gust). However, when I look at the loop packet that `StdConvert` 9Iwoudl have thought `StdConvert` was the first port of call for the loop packet) processes it already contains `windGust` and `windGustDir`. This doe snot seem right to me. I have tried working out where the gust obs are coming from but with no luck. Can you help?

gjr80, 2017-01-13:
Ah, OK, just found it in the vantage driver `VantageService`.

gjr80, 2017-01-13:
OK, I have a theory, it doesn't really explain the behaviour of the WMR180 system that kicked this of but it does fit Tom, Andrew and my situations....

Basically the issue arises when using the vantage driver with `loop_hilo` set True. The `VantageService` intercepts the loop packet from the station and adds `windGust` and `windGustDir`, these values come from monitoring the `windSpeed` and `windDir` in the loop packet and keeping track of the max `windSpeed` from the loop packets during the archive period. `VantageService` adds the highest `windSpeed` seen as `windGust` and the corresponding `windDir` as `windGustDir`. So if we consider a fictitious archive period that saw loop packet `windSpeed` values of say, 3, 0, 5, 4, 2 and 1 successive loop packets would have had `windGust` values of 3, 3, 5, 5, 5 and 5. Now when these hit the accumulator at the end of the archive period the `windSpeed` accumulator will have a max value of 5 and the `windGust` accumulator will also have a max value of 5. If no more wind speeds are seen during the day that are >5 then the max fields in `archive_day_windSpeed` and `archive_day_windGust` for the day will both be 5 and `$day.windSpeed.max` and `$day.windgust.max` will be the same.

If we factor in (hardware) archive records from the station. `windSpeed` will be the average wind speed seen over the archive period so it will certainly be less than the max loop `windSpeed` seen. `windGust` will be the max wind speed seen by the station over the archive period, which I suspect will be the same as the max of the loop `windSpeed` seen over the archive period. Bottom line, I don't believe the archive record will influence the daily summary max fields.

Now if we turn off `loop_hilo` the loop packets are ignored for highs and lows and the max fields will be populated from the archive records. So `archive_day_windSpeed` would have the max archive record `windSpeed` for the day and `archive_day_windGust` would have the max archive record `windGust` for the day. These would be different values.

Now to explain Tom's situation where the values converge around when v3.0 came out. This does not tally with my vantage driver theory as the `VantageService` `windGust` code has been in there since well before v3.0. I wonder if perhaps the introduction of v3.0 and the force `rebuild` of the stats into daily summaries actually caused a rebuild from the archive data at the time and hence the `correct` max values were put in the new daily summaries. The archive data is good, it is only the daily summary max values that are incorrect.

Andrew reported that his data appeared to be fine. This could be because he has rebuilt his daily summaries but he showed the 13 Jan 17 daily summary, I doubt that would have been rebuilt. I think the real reason is that his station uses th FOUSB driver and whilst I haven't checked I suspect it does not treat wind the same as the vantage driver.

The one flaw in my theory is that this does not explain the WMR180 situation that kicked this off. I had a cursory look at the wmr100 driver, it seems pretty basic. Matthew may be able to add more.

tkeffer, 2017-01-13:
How stupid of me. 

The code is actually correct. Using your example, the max daily wind speed is, in fact, 5. So, that's the right answer.

But, of course, if you rebuild the summaries, you're going to get a different answer, because you've lost the resolution of the loop packets.

Unfortunately, this means it's hard to say what `$day.windSpeed.max` means. If you get it from the daily summary, it's the max seen during the day, including from loop packets. If you get it from the archive table, then it's the max of the archive records, which are themselves averages, so you get a lower number.

The same situation applies with other variables, but they don't have a "Gust" type observation type, so the max seen in each archive interval is never recorded separately.

We have two choices:
1. Leave it the way it is. We're getting the correct answer, but the answer depends on whether or not the higher resolution of the daily summaries is available.
2. Treat wind differently.

I don't really like either choice. Any others?

gjr80, 2017-01-13:
I guess I have a slightly different view, for wind I don't think if it as a case of the extra resolution we are getting from loop packets, rather I think of it as in the daily summaries we are treating both `windSpeed` and `windGust` as if they are instantaeous or snapshot values whereas in the archive we are treating `windSpeed` as an average over a period of time and `windGust` as the instantaneous or snapshot value. Davis defines what we call `windSpeed` in an archive record as the average wind speed over an archive period. Yet the value we put into `archive_day_windSpeed` max is not the max average wind speed over an archive period in the day but rather the max instantaneous wind speed over the day. If you look at some of the other weather software they do this very thing, they provide a max average wind speed for the day and a max wind speed for the day, with both being different values.

I guess I want my cake and eat it to. I wantr the giger resolution of the loop packets give the daily summaries in terms of peaks and times. But I think windSpeed needs to be treated differently to how it is now (well in terms of the vantage driver). I'm firmly for 2.

gjr80, 2017-01-13:
For mine, implementing something like

```
if obs_type == 'windSpeed'
    loop_hilo = False
else:
    loop_hilo = True
```
would give an approriate result

tkeffer, 2017-01-13:
How about the following patch?

~~~
diff --git a/bin/weewx/accum.py b/bin/weewx/accum.py
index 0b29e06..6f7957c 100644
--- a/bin/weewx/accum.py
+++ b/bin/weewx/accum.py
@@ -328,19 +328,16 @@
         if weewx.debug:
             assert(obs_type == 'windSpeed')
         
-        # First add it to regular old 'windSpeed', then
-        # treat it like a vector.
-        self.add_value(record, obs_type, add_hilo)
+        # First treat windSpeed as a regular old scalar. However, 
+        # max wind speed is handled by wind gusts, so don't add high/lows.
+        self.add_value(record, obs_type, add_hilo=False)
         
-        # If the type has not been seen before, initialize it
+        # Now treat wind as a vector. First initialize the vector accumulator.
         self.init_type('wind')
         # Then add to highs/lows, and to the running sum:
         if add_hilo:
             self['wind'].addHiLo((record.get('windGust'), 
                                  record.get('windGustDir')), 
-                                 record['dateTime'])
-            self['wind'].addHiLo((record.get('windSpeed'), 
-                                 record.get('windDir')),
                                  record['dateTime'])
         # Now add to the running sums/counts, passing a weight of 'interval' if 
         # the value is an aggregate over some interval (eg archive record) and 

~~~

tkeffer, 2017-01-13:
Incidentally, the reason why the daily summaries started showing this issue when v3.0 was introduced is not because the code changed, but  because the daily summaries had to be rebuilt at that time.

gjr80, 2017-01-13:
>Incidentally, the reason why the daily summaries started showing this issue when v3.0 was introduced is not because the code changed, but because the daily summaries had to be rebuilt at that time.

That supports my theory; it was always like that. One other thing, until steel series gauges came along that essentially put `$day.windSpeed.max` and `$day.windGust.max` next to each other who would have noticed?

gjr80, 2017-01-13:
>How about the following patch?

Seems to make sense to me. I can't think of any reason why someone would want to add high/lows for `windSpeed`, so I think it is quite valid hard coding `add_hilo=False` rather than making it user selectable through a config option.

Actually, just thinking aloud, my analysis focussed on the vantage driver, it's probably better defined and more stable than some of the others. Are we creating any issues for any lessesr capable stations that say only push out `windSpeed`? I am afraid I am not familiar with the 'fleet'.

tkeffer, 2017-01-13:
So, do we want to add this to the patch? It would not be hard to do. Just set `archive_day_windSpeed.max` to the max for the day from the archive data.

> Actually, just thinking aloud, my analysis focussed on the vantage driver, ...

I don't see why this is specific to the vantage driver. It may be specific to Vantage weather stations, because they are one of the few that emit meaningful `windGust` values.

matthewwall, 2017-01-13:
this hardware emits windGust:

cc3000
fousb
te923
vantage
wmr100
wmr200
wmr300
wmr9x8
ws28xx
cm1
lw30x
gw1000u
ip100
netatmo
wh1000
wh23xx

this hardware emits windGustDir:

vantage
wmr300
wmr9x8
ws28xx
meteostick
cm1
gw1000u
ip100
netatmo
wh1000

gjr80, 2017-01-13:
Sorry, still thinking aloud. So what would be the result of a station that (1) only emits loop packets and (2) only provides `windSpeed` not `windGust`...

Archive records would be software generated and the archive `windSpeed` would be the average of the loop `windSpeed` values seen. That is correct. Archive `windGust` would be `None` as no `windGust` loop data was accumulated. In terms of the daily summary the `archive_day_windSpeed` max field would be set to the archive `windSpeed` (if it was greater than the existsing max). The `archive_day_windGust` max field would be set to `None` since there was no `windGust` data.

Is this consistent with what we do now for stations that do not emit `windGust`?

gjr80, 2017-01-14:
>this hardware emits windGust:

So Matthew, does an AcuRite station put anything in any `windGust` fields in the database?

matthewwall, 2017-01-14:
no.  every driver not in the list will not emit a windGust or windGustDir field.  except for ws23xx - it emits a None for each.  given our discussion about None, i consider that to be a bug that should be fixed when we do the None review.

gjr80, 2017-01-14:
>So, do we want to add this to the patch? It would not be hard to do. Just set `archive_day_windSpeed.max` to the max for the day from the archive data.

Ah, just realised what you were asking? I first thought you meant this patch:

>How about the following patch?

but now I realise you meant the #61 'patch'. In answer to your question I think 'yes' if the final #195 patch is as you posted above.

gjr80, 2017-01-14:
>no.  every driver not in the list will not emit a windGust or windGustDir field.  except for ws23xx - it emits a None for each.  given our discussion about None, i consider that to be a bug that should be fixed when we do the None review.

Ok, so for my thinking alound about loop only stations without `windGust` the patch Tom proposed above would not be taking away any 'data' for such users.

I say go ahead and do it.

matthewwall, 2017-01-14:
so the update-summaries code will now include repair-wind-calculations, repair-interval-calculations, and its namesake update-daily-summaries.  right?

tkeffer, 2017-01-14:
I'm not sure what you mean by update-daily-summaries. It will contain
- Fix for the interval weightings;
- Fix for max windSpeed.

Too much? Maybe we should go back to a user invoked patch...

gjr80, 2017-01-14:
>so the update-summaries code will now include repair-wind-calculations, repair-interval-calculations, and its namesake update-daily-summaries. right?

If you are asking will the 'interval weighting patch code' now include 'repair archive_day_windSpeed.max calculations' and 'interval weight the daily sumamries' then the answer is yes, that is what is on the table. Not sure what your 3 elements of 'update-sumamries code' refer to . Sorry.

tkeffer, 2017-01-14:
> Archive records would be software generated and the archive `windSpeed` would be the average of the loop `windSpeed` values seen. That is correct. Archive `windGust` would be `None` as no `windGust` [...]

That pretty much summarizes it. However, note that `archive_day_wind.max` would still contain the absolute max `windSpeed` seen in all loop packets. Don't know if that's useful, or even more confusion.

I think we need a consistent policy for `windGust`. Perhaps `StdWXCalculate` could take it on?

matthewwall, 2017-01-14:
seems appropriate for StdWXCalculate.  if a station provides windGust, weewx uses that.  if not, StdWXCalculate creates windGust from windSpeed - the maximum windSpeed seen in an archive interval; it would apply only to an archive interval.

it could also do windGustDir, but that is kind of fabricated since windDir might or might not correlate with a windSpeed.  perhaps better to not do windGustDir?

matthewwall, 2017-01-14:
> Too much? Maybe we should go back to a user invoked patch...

it seems to me that a user-invoked repair will be easier to implement and support.

then the operation is as simple as:

wee_database --drop-summaries
wee_database --build-summaries [--fix-weights] [--fix-wind]
wee_database --fix-strings

with diagnostic options:

wee_database --check-weights
wee_database --check-wind
wee_database --check-strings

that indicate whether a repair is necessary.

- the user is directly aware/responsible for making a backup
- the user can decide whether the repairs need to be done
- the user can decide when to do the repairs
- the user can apply the repairs to specific databases, e.g., other than wx_binding
- all of the startup checks can go away
- no need to track down all the corner cases for automatically doing repairs

gjr80, 2017-01-14:
>it could also do windGustDir, but that is kind of fabricated since windDir might or might not correlate with a windSpeed. perhaps better to not do windGustDir?

I presume that similar code to what is in VantageService would be used to 'calculate' windGust. If that is the case, and if there is a windSpeed and windDir in each loop packet, why would you not be able to 'calculate' windGustDir? If windSpeed and windDir are in a packet they must be related?

> it seems to me that a user-invoked repair will be easier to implement and support.
>
> then the operation is as simple as:
>
> wee_database --drop-summaries
> wee_database --build-summaries [--fix-weights] [...]

I must admit I was not a big fan of the manual approach before, I saw it a leaving who knows how many databases in who knows what state. I am less hard and fast now. If a manual approach is taken I guess there are two approaches (1) some level of 'minimum' fix is required to move to 3.7.0 and beyond, eg. you must move to version 2.0 if you want to use 3.7.0 or beyond but you must do the 'fix' and here is the tool to use, or (2) no fixes are mandatory to move to 3.7.0. Not sure what you are proposing Matthew, I feel better about (1).

In terms of the manual process, there should be no need to --drop-daily. You mention --build-summaries, a new verb? what about --check-summaries instead with optional --fix-weighting and --fix-wind.   --check-strings should not really come into it, understand it is another diagnostic tool but it is not a consequence of interval weighting or windSpeed.max 'fixing'.

matthewwall, 2017-01-14:
sorry to use different terms - i'm spitballing alternatives for 'patch' :)  'build-summaries' is instead of what was known as 'backfill-daily'.  is it 'build-daily' now?

so are you proposing this:
```
# see if the database has bogus strings
wee_database --check-strings
# see if the database has any incorrect weightings
wee_database --check-weights
# see if the database has any incorrectly calculated wind values
wee_database --check-wind

# repair any fields that have funky strings
wee_database --fix-strings
# (A) recalculate any interval calculations
wee_database --fix-weights
# (B) recalculate any wind calculations
wee_database --fix-wind

# clear the summaries then build the summaries (the nuclear option)
wee_database --drop-daily
wee_database --build-daily
```
there are probably some caveats for operations (A) and (B) that need to be clarified

tkeffer, 2017-01-14:
> it could also do windGustDir, but that is kind of fabricated since windDir might or might not correlate with a windSpeed. perhaps better to not do windGustDir?

If `StdWXCalculate` keeps track of the max `windSpeed` seen during an archive interval, then it can also record the wind direction that goes along with it.

Come to think of it, the accumulators already do that. But, `StdWXCalculate` doesn't have access  to them (they live in `StdArchive`). So, maybe the place to do it is when extracting a record out of the accumulators.

matthewwall, 2017-01-14:
for stations that emit windGust but no windGustDir, that would result in data skew, no?

gjr80, 2017-01-14:
>for stations that emit windGust but no windGustDir, that would result in data skew, no?

But that's a slightly different case to what (I understood) you first mentioned:

>it could also do windGustDir, but that is kind of fabricated since windDir might or might not correlate with a windSpeed. perhaps better to not do windGustDir?

I would have thought something like the following would be doable, seen as extracting the most infor we could whilst not fabricating data:

```
if windGust in record and windGustDir in record:
    pass  #windGust and windGustDir already exist, no need to calculate
elif windGust in record:
    pass  #windGust already exists but no windGustDir, just use windGust nothing more we can do
else:
    record[windGust] = loop_max_windSpeed
    record[windgustDir] = loop_max_windDir
```

Assuming of course the station provides windSpeed and windDir

tkeffer, 2017-01-14:
I don't see a problem with reporting a `windGust`, but no `windGustDir`, if that's what the instrument provides. Gary's solution should work.

Instruments with limited capabilities will provide limited, or even compromised, information. Our job is to just write it down.

tkeffer, 2017-01-14:
Fixed windSpeed problem. Turned out to be a lot more involved, because the value has to be merged into the daily summaries as well. This required new merge functions.

While I was at it, I implemented a way of specifying accumulator configuration from config files, instead of dictionaries. See issue #115. Then used that to specify the special merge function for windSpeed.

closed 2017-01-14 by tkeffer