weewx/weewx #197: Need to patch max windSpeed

tkeffer, 2017-01-14:
This is a residue from issue #195. See the discussion there.

While the code has been fixed, the old values of windSpeed in the daily summaries remains.

gjr80, 2017-01-15:
Put the first cut of the code to recalculate the `windSpeed` daily summary max value in the new `weecfg.database`. Intend to review and move the existing `weecfg.patch` content to `weecfg.database`. #61

gjr80, 2017-01-15:
Just to give some idea of time for this patch. RPi1 model B is taking about 280 seconds to patch 10 years of data with 50 days/transaction. This is consistent across MySQL and SQLite. Increasing transaction days to 200 brought this down to 230 seconds for MySQLite (did not test MySQL).  Going to 500 days/transaction made no difference, stayed at 230 seconds.

tkeffer, 2017-01-29:
Fixed in V3.7.0

closed 2017-01-29 by tkeffer