weewx/weewx #198: Consistent treatment of windGust

tkeffer, 2017-01-14:
This is left over from issue #195.

Not all weather stations generate `windGust`. We should provide it if it's missing. 

Not clear where to do this. `StdWXCalculate` is the obvious answer, but it would have to accumulate loop packets, something the accumulators already do. But, they live in `StdArchive`, and `StdWXCalculate` has no access to them. Two choices.
1. StdWXCalculate accumulates the loop packets anyway, resulting in some duplication.
2. Have the record extraction function in accum.py augment the record it is building with `windGust` if it is missing.

I'm slightly favoring option 1, even though it means duplicate functionality. The reason is that I'd like to give the user an option of how it's calculated, like the more traditional derived values like `dewpoint`.

matthewwall, 2017-01-14:
from a strict only-what-the-hardware-emits point of view, weewx should not add windGust or windGustDir if the hardware does not provide them.  it seems like windGust is not as cut-and-dried as dewpoint or heatindex or windchill or apptemp.

on the other hand, if you say that windGust is just the maximum of windSpeed over a period, then windGust does smell a lot like windchill or dewpoint - just another derived variable.

it would be nice if StdWXCalculate could have a 'NOAA gust' algorithm for calculating windGust as an option instead of max(windSpeed)

matthewwall, 2017-04-16:
some hardware reports windGust and windGustDir.  some reports only windGust.  some report neither.

loop packets should reflect exactly what the driver is spitting out, e.g., no windGustDir for fousb or wmr200 hardware, and no windGust or windGustDir for acurite hardware.

with record_generation=software, archive packets should contain windGust and windGustDir (derived from windSpeed and windDir values).

with record_generation=hardware, archive packets should contain only what the hardware reports, e.g., no windGustDir for fousb.  wmr200 and acurite have no logger, so record_generation=hardware is not applicable to them.

closed 2017-04-16 by matthewwall