weewx/weewx #200: extreme clock skew has undesirable side effects

matthewwall, 2017-01-15:
if there is significant skew between the station clock and the system clock, weewx runs the report generator on every loop packet.  if the reports take more than a loop interval to run, then things get real ugly real fast.

i hit this behavior when the station was 5 hours off from the computer (station in local time, computer in utc).

we should figure out how to make weewx work better in this case, either fail loudly or not run reports on every loop.

tkeffer, 2017-03-18:
Right now, this is the basis for deciding whether to break the main event packet loop:

~~~python
        # Has the end of the archive delay period ended? If so, break the loop.
        if event.packet['dateTime'] >= self.end_archive_delay_ts:
            raise BreakLoop
~~~

If the incoming loop packet has a timestamp greater than the end of the archive delay (generally 15 seconds after the end of the archive interval), it breaks the event loop, and launches a report thread. If the station clock is way ahead of the system clock, this would definitely cause a premature launch.

I see two possible fixes.

1. In the above, I don't think we want to start using the wall clock time instead of the loop packet timestamp, but what we could do is introduce a new timestamp, the wall clock time the packet was received. This would then be used to decide whether to break the loop and launch the reporting thread.

    But, there may be lots of downstream consequences that I'm not seeing. It would require some experimenting.

2. Alternatively, when a loop packet arrives, we can check its timestamp to see if it is very different from the wall clock time. If it is, reject the packet, or do a clock sync, or ... ?

    This has the advantage that it addresses the problem at its root. There is just no good way to compensate for a bad clock on the hardware.

tkeffer, 2017-03-25:
Matthew: do you see a problem with my proposal 2? That is, reject LOOP packets with very old timestamps?

tkeffer, 2019-11-28:
Thinking about this one, your situation was clearly a configuration problem. So, the right thing to do is to fail loudly.

We could check for packets with a timestamp more than, say, 60 seconds ahead of the wall clock time. If found, exit with an error.

It's true that this is a harsh treatment for just a couple of errant LOOP packets with timestamps in the future in a stream of otherwise OK packets. But, I don't see a scenario where that could happen. Most likely, the timestamps are either all OK, or none of them is.

closed 2024-07-03 by github-actions[bot]