weewx/weewx #201: Test skin for testing templates fails when local date format not M/D/Y

dcapslock, 2017-01-15:
While generating a new skin test case I have found that the use of .format("%x %X") in index.html.tmpl means that index.html fails when the local date format is not M/D/Y. The template should use an explicit format, like those in the skin.conf for datetime contexts.

tkeffer, 2017-01-15:
I assume the reason is that you are testing in a different locale? 

Two choices:
1. Set the locale to something consistent before running. I don't know how consistent `%x` and `%X` are across operating systems in a given locale, so this may not solve the problem completely.
2. Do as you say, and use an explicit format. 

I'd say option 2 is the clear favorite.

dcapslock, 2017-01-15:
Yes. Testing on my default locale (Aus English).

We could set the locale with LC_ALL - see http://stackoverflow.com/questions/12896509/forcing-date-to-use-a-locale-other-than-the-machine-default

I agree that 2. is the favourite as we are not wanting to test forcing a locale.

dcapslock, 2017-01-21:
PR #205 fixes this issue.

closed 2017-01-21 by dcapslock