weewx/weewx #202: interval being set to zero

overpaul, 2017-01-16:
I have weewx set to update every 60 seconds.
The calculation to determine the interval is integer so if the time is slightly less than 60 seconds it becomes zero.

I fixed the problem with the following code:

#        record['interval'] = int((now_ts - last_ts) / 60)      # REPLACED
        record['interval'] = int(round(((now_ts - last_ts) / 60.0),0))      #  to correct for times slightly less than interval

while this works for me, a better solution might be

#        record['interval'] = int((now_ts - last_ts) / 60)      # REPLACED
        record['interval'] = int((now_ts - last_ts + 5) / 60)      # to correct for times slightly less than interval

where 5 is the maximum jitter

tkeffer, 2017-01-16:
You lost me. I can find no place in the code that looks like your line. What are you patching? What hardware are you using?

overpaul, 2017-01-16:
I must have posted it on the wrong thread.  The problem was in the cmon
extension code not base weewx code.
Where should I post it?

*Paul Oversmith*

E-Mail: pgo@overweb.com
Phone: (941) 266-3600

tkeffer, 2017-01-16:
If it's a legitimate bug, no problem with posting it here. But, you need to provide some context!

I'm sure Matthew will be by. :-)

overpaul, 2017-01-16:
Sorry I thought it was a bug list for cmon.  Here is a copy of the database
entries that identified the problem.  Could you pass it along to Matthew?
[image: Inline image 1]

*Paul Oversmith*

E-Mail: pgo@overweb.com
Phone: (941) 266-3600

matthewwall, 2017-01-16:
cmon 0.16 includes a fix to ensure that the interval is non-zero

closed 2017-01-16 by matthewwall

overpaul, 2017-01-16:
thanks - for the quick fix, much better than my patch😀

*Paul Oversmith*

E-Mail: pgo@overweb.com
Phone: (941) 266-3600