weewx/weewx #203: cheetahgenerator file not found error

overpaul, 2017-01-19:
I have my Raspberry Pi set up to store the web files in memory rather than on the SD card.

When booting weewx has to rebuild all the files.  This heavy load created two problems at start up:

The first problem was with imagegenerator getting a database lock which I fixed, per your suggestion, by changing the SQLite timeout to 15 seconds

The second problem was in generating the NOAA reports (I have over 3 years of data).  I would get several 'file not found errors' when trying to copy the temp file. (may times on the same NOAA file)
I suspect this happened because of the Pi being overloaded with file activity

I did two things to fix this:

1. First I added more information to the error message (temp file and target file) to locate which file caused the problem
2. Second I tested to see if the file existed before copying and if it didn't. I would wait 1 second before copying.  If the file was there in a second, I let it rase an error

My code changes follow:

            try:
                text = Cheetah.Template.Template(
                    file=template,
                    searchList=searchList,
                    filter=encoding,
                    filtersLib=weewx.cheetahgenerator)
                with open(tmpname, mode='w') as _file:
                    print >> _file, text
                    
##### Modified by PGO (START) ########################################                

                if not (os.path.isfile(tmpname)):             # Allow a delay, if file not closed yet
                  time.sleep(1)
                
##### Modified by PGO (END) ######################################## 
  
                os.rename(tmpname, _fullname)
            except Exception, e:
                # We would like to get better feedback when there are cheetah
                # compiler failures, but there seem to be no hooks for this.
                # For example, if we could get make cheetah emit the source
                # on which the compiler is working, one could compare that with
                # the template to figure out exactly where the problem is.
                # In Cheetah.Compile.ModuleCompiler the source is manipulated
                # a bit then handed off to parserClass.  Unfortunately there
                # are no hooks to intercept the source and spit it out.  So
                # the best we can do is indicate the template that was being
                # processed when the failure ocurred.
                logerr("Generate failed with exception '%s'" % type(e))

##### Modified by PGO (START) ########################################
                
                logerr("**** Temp file: %s Report: %s" % (tmpname, _fullname))     ### PGO to better log error
                
##### Modified by PGO (END) ########################################   
             
                logerr("**** Ignoring template %s" % template)
                logerr("**** Reason: %s" % e)
                weeutil.weeutil.log_traceback("****  ")

tkeffer, 2017-01-19:
I'm not following your logic. This section of Python code is completely synchronous. Are you saying that after the operating system finishes writing the temporary file, it is not aware that it exists for a second, so we must wait for it to appear?

Or, am I misunderstanding something?

overpaul, 2017-01-19:
Appears that way.

Attached is the log before I added the logging of the file name.

When I did, I found it was the NOAA files.  It also looked like it tried
several times to write the same file before success.

The check for existence of the file and delay if not found  I added seemed
to fix it.

The only thing I can figure is the operating system was busy and had not
closed the NOAA file before the copy request.

It is a Pi 3, but the imagewriter thread was very busy.  I create about 160
images, divided equally between day, week, month and year.  This takes
several minutes.

In any case, I don't think the changes I made hurt anything, since the
delay will not occur if the system close the file promptly.
And the additional log data is helpful in determining what file is causing
the problem.

Once all the files are built there is no timing issue and I suspect the
problem would not occur if the web files were on a SD as they would be
there already during a reboot.

*Paul Oversmith*

E-Mail: pgo@overweb.com
Phone: (941) 266-3600

tkeffer, 2017-01-20:
I am not enthusiastic about this patch. Linux includes lots of code, especially for the default EXT4 partitions used by Raspbian, to insure file integrity. If it's not working, there are far bigger problems.

I suspect that your SD is close to failing, leading to erratic behavior.

What is an "imagewriter thread?"

Also, no log was attached.

tkeffer, 2017-01-29:
No activity.

closed 2017-01-29 by tkeffer