weewx/weewx #207: AAG 1-Wire station - odd looking data

clickworkorange, 2017-01-24:
I've got the venerable AAG TAI 8515 weather instrument, which is connected to a Raspberry Pi running the latest version of Rasbpian with OWFS and the latest firmware. My interface is an [AB Electronics 1-Wire Pi Zero](https://www.abelectronics.co.uk/p/76/1-Wire-Pi-Zero), which connects to the Pi over the I2C bus. My boot.config contains `dtoverlay=w1-gpio` and lsmod returns 

```
Module                  Size  Used by
ip6table_filter         1349  0 
ip6_tables             11701  1 ip6table_filter
xt_tcpudp               2269  1 
nf_conntrack_ipv4      12912  3 
nf_defrag_ipv4          1428  1 nf_conntrack_ipv4
xt_conntrack            3167  3 
nf_conntrack           77460  2 xt_conntrack,nf_conntrack_ipv4
iptable_filter          1409  1 
ip_tables              11445  1 iptable_filter
x_tables               13165  6 ip6table_filter,ip_tables,xt_tcpudp,xt_conntrack,iptable_filter,ip6_tables
snd_bcm2835            20447  0 
snd_pcm                75762  1 snd_bcm2835
snd_timer              19288  1 snd_pcm
snd                    51908  3 snd_bcm2835,snd_timer,snd_pcm
bcm2835_wdt             3225  0 
w1_gpio                 3657  0 
i2c_bcm2708             4834  0 
bcm2835_gpiomem         3040  0 
wire                   25219  1 w1_gpio
cn                      4374  1 wire
uio_pdrv_genirq         3164  0 
uio                     8000  1 uio_pdrv_genirq
i2c_dev                 5859  2 
fuse                   84037  1 
ipv6                  347556  34 
```
WeeWX has been configured to use the 1-Wire interface with: 

[Station]
    station_type = OWFS

```
[OWFS]
    interface = /dev/i2c-1
    [[sensor_map]]
        outTemp = /uncached/10.8B401B000800/temperature
        windDir = /uncached/20.E63307000000     # /volt.ALL
        windSpeed = /uncached/1D.005A01000000   # /counter.A
    [[sensor_type]]
        windDir = aag_windvane
        windSpeed = aag_windspeed
```
And sudo PYTHONPATH=/home/weewx/bin python /home/weewx/bin/user/owfs.py --iface=/dev/i2c-1 shows: 
```
--readings
E63307000000
_usePath: /20.E63307000000
_type: DS2450
_path: /20.E63307000000
_useCache: True
  /20.E63307000000/family: 20
  /20.E63307000000/set_alarm/: high.ALL,high.A,high.B,high.C,high.D,low.ALL,low.A,low.B,low.C,low.D,unset,volt2high.ALL,volt2high.A,volt2high.B,volt2high.C,volt2high.D,volt2low.ALL,volt2low.A,volt2low.B,volt2low.C,volt2low.D,volthigh.ALL,volthigh.A,volthigh.B,volthigh.C,volthigh.D,voltlow.ALL,voltlow.A,voltlow.B,voltlow.C,voltlow.D
  /20.E63307000000/CO2/: Vdd,ppm,status
  /20.E63307000000/locator: FFFFFFFFFFFFFFFF
  /20.E63307000000/PIO.D: 1
  /20.E63307000000/r_locator: FFFFFFFFFFFFFFFF
  /20.E63307000000/id: E63307000000
  /20.E63307000000/PIO.B: 1
  /20.E63307000000/PIO.C: 1
  /20.E63307000000/pages/: page.ALL,page.0,page.1,page.2,page.3
  /20.E63307000000/volt.D: 4.73749
  /20.E63307000000/alarm/: high.ALL,high.A,high.B,high.C,high.D,low.ALL,low.A,low.B,low.C,low.D
  /20.E63307000000/volt.C: 4.74148
  /20.E63307000000/volt.A: 0.0660166
  /20.E63307000000/volt2.D: 0
  /20.E63307000000/volt2.C: 0
  /20.E63307000000/volt2.B: 0
  /20.E63307000000/volt2.A: 0.0343755
  /20.E63307000000/PIO.A: 1
  /20.E63307000000/memory: p
  /20.E63307000000/type: DS2450
  /20.E63307000000/power: 0
  /20.E63307000000/PIO.ALL: 1,1,1,1
  /20.E63307000000/volt.ALL: 0.0660166,4.72429,4.74148,4.73749
  /20.E63307000000/address: 20E6330700000038
  /20.E63307000000/8bit/: volt.ALL,volt.A,volt.B,volt.C,volt.D,volt2.ALL,volt2.A,volt2.B,volt2.C,volt2.D
  /20.E63307000000/r_address: 380000000733E620
  /20.E63307000000/volt2.ALL: 0.0343755,0,0,0
  /20.E63307000000/crc8: 38
  /20.E63307000000/volt.B: 4.72429
  /20.E63307000000/r_id: 0000000733E6
8B401B000800
_usePath: /10.8B401B000800
_type: DS18S20
_path: /10.8B401B000800
_useCache: True
  /10.8B401B000800/scratchpad: �
  /10.8B401B000800/power: 0
  /10.8B401B000800/family: 10
  /10.8B401B000800/locator: FFFFFFFFFFFFFFFF
  /10.8B401B000800/type: DS18S20
  /10.8B401B000800/crc8: DA
  /10.8B401B000800/temphigh: 75
  /10.8B401B000800/templow: 70
  /10.8B401B000800/r_id: 0008001B408B
  /10.8B401B000800/address: 108B401B000800DA
  /10.8B401B000800/errata/: die,trim,trimblanket,trimvalid
  /10.8B401B000800/r_address: DA0008001B408B10
  /10.8B401B000800/r_locator: FFFFFFFFFFFFFFFF
  /10.8B401B000800/id: 8B401B000800
  /10.8B401B000800/temperature: 85
005A01000000
_usePath: /1D.005A01000000
_type: DS2423
_path: /1D.005A01000000
_useCache: True
  /1D.005A01000000/pages/: count.ALL,count.0,count.1,count.2,count.3,count.4,count.5,count.6,count.7,count.8,count.9,count.10,count.11,count.12,count.13,count.14,count.15,page.ALL,page.0,page.1,page.2,page.3,page.4,page.5,page.6,page.7,page.8,page.9,page.10,page.11,page.12,page.13,page.14,page.15
  /1D.005A01000000/mincount: 12124
  /1D.005A01000000/family: 1D
  /1D.005A01000000/locator: FF40FFFFFFFFFFFF
  /1D.005A01000000/type: DS2423
  /1D.005A01000000/crc8: 64
  /1D.005A01000000/memory: 
                           ��
  /1D.005A01000000/counter.ALL: 12134,12133
  /1D.005A01000000/r_address: 64000000015A001D
  /1D.005A01000000/r_id: 000000015A00
  /1D.005A01000000/address: 1D005A0100000064
  /1D.005A01000000/counter.A: 12134
  /1D.005A01000000/counter.B: 12133
  /1D.005A01000000/r_locator: FFFFFFFFFFFF40FF
  /1D.005A01000000/id: 005A01000000
```
So far, so good - but I'm getting some pretty weird measurements when I run WeeWX (lines shortened for brevity): 
```
LOOP: 17:10:57 UTC (1485277857) outTemp: None, windDir: None, windSpeed: None
LOOP: 17:11:07 UTC (1485277867) outTemp: None, windDir: 270.0, windSpeed: 7.11422073645
LOOP: 17:11:19 UTC (1485277879) outTemp: None, windDir: 270.0, windSpeed: 6.03073309555
LOOP: 17:11:30 UTC (1485277890) outTemp: 61.1375, windchill: 61.1375, windDir: 270.0, windSpeed: 6.57898155879
LOOP: 17:11:41 UTC (1485277901) outTemp: None, windDir: 270.0, windSpeed: 6.02144074872
LOOP: 17:11:51 UTC (1485277911) outTemp: None, windDir: 270.0, windSpeed: 6.50092584538
LOOP: 17:12:01 UTC (1485277921) outTemp: None, windDir: 270.0, windSpeed: 6.62358482359
LOOP: 17:12:12 UTC (1485277932) outTemp: 61.25, windchill: 61.25, windDir: 270.0, windSpeed: 6.46747339677
LOOP: 17:12:23 UTC (1485277943) outTemp: 61.1375, windchill: 61.1375, windDir: 270.0, windSpeed: 6.46747339677
LOOP: 17:12:34 UTC (1485277954) outTemp: None, windDir: 270.0, windSpeed: 5.90993258671
LOOP: 17:12:44 UTC (1485277964) outTemp: None, windDir: 270.0, windSpeed: 7.11422073645
LOOP: 17:12:55 UTC (1485277975) outTemp: None, windDir: 270.0, windSpeed: 6.57898155879
LOOP: 17:13:07 UTC (1485277987) outTemp: None, windDir: 270.0, windSpeed: 5.41743820448
LOOP: 17:13:17 UTC (1485277997) outTemp: None, windDir: 270.0, windSpeed: 6.50092584538
LOOP: 17:13:27 UTC (1485278007) outTemp: None, windDir: 270.0, windSpeed: 6.50092584538
LOOP: 17:13:37 UTC (1485278017) outTemp: 61.1375, windchill: 61.1375, windDir: 270.0, windSpeed: 7.23687971467
LOOP: 17:13:49 UTC (1485278029) outTemp: 61.25, windchill: 61.25, windDir: 270.0, windSpeed: 5.92851728038
LOOP: 17:14:00 UTC (1485278040) outTemp: None, windDir: 270.0, windSpeed: 5.90993258671
```
As you can see the `outTemp` only returns a value occasionally, and seems to be repeating itself - and the windSpeed fluctuates surprisingly wildly, considering that the anemometer is placed in front of a desk fan which runs at a steady speed, while it too shows some very odd value repetitions (e.g. 7.11422073645 appears twice, about 1 1/2 minute apart!?). This cannot be right I think, but what could be the problem?

matthewwall, 2017-01-24:
what is the resolution of the anemometer wrt windSpeed?

excellent description of the problem, but instead of here could you post this in the weewx forum:

http://groups.google.com/group/weewx-user

you'll probably get broader feedback there.

we use the github issue tracking system once a bug has actually been identified.

closed 2017-01-24 by matthewwall

clickworkorange, 2017-01-24:
Thank you - I did think Github was probably not the best place for this, but I did not know you had a Google Group. I have to say I'm not a fan of Google Groups, but I'll see if I can set up an account.