weewx/weewx #212: BadLogin connection to WU

paulkaras, 2017-02-02:
After many successful logins I keep funding that both Weewx and Wunderfixer return errors like "Bad login
INVALIDPASSWORDID|Password or key and/or id are incorrect" which would appear to an issue withe the Restx WU code.  This then leads to huge gaps in my WU data feed and the weather station goes off line.  Is it possible to make the code more resilient by, say, retrying the login again a few seconds later instead of waiting an hour (in weewx) or just crashing out the code in wunderfixer.  Thoughts?

tkeffer, 2017-02-03:
Hello

I doubt very much there is an issue in the restx code. Far more likely, this is the usual WU flakiness.

Nevertheless, allowing a user to adjust the retry delay is easy enough. I've added it to commit https://github.com/weewx/weewx/commit/8cac36d37a34e424a8278101dce733f363316488

Look for it in V3.7.0.

As far as wunderfixer is concerned, I'm leery about making it a longer-lived program. Most people treat it as a one-shot thing, not something that can execute indefinitely.

closed 2017-02-03 by tkeffer

paulkaras, 2017-02-04:
Thanks  - i tweaked the local copy of restx.py to give a 2 second delay after a badlogin feedback.  This seems to have worked.  It would be great to do the same for wunderfixer so that I can fill the old gaps.

tkeffer, 2017-02-04:
This would better be done in a shell script.

paulkaras, 2017-02-04:
Two issues -
Normal weewx RF update to wunderground seems to have a less than 90% chance of a successful login so the wait 2 seconds stops 60 minute or often longer gaps appearing in the data and wunderground then take the station offline.  So a 2 second wait keeps it going.  it would be good if this were in the 'proper' version of the code.
Wunderfix is more of a problem as it just crashes out after every few uploads so it takes a lot of rerunning just to get the data uploaded.  So - I'm looking for a quick hack to make it more error tolerant.  the code is well documented so it shouldn't be too hard to fix for a one off catch up.
Thanks
P

tkeffer, 2017-02-04:
> Normal weewx RF update to wunderground seems to have a less than 90% chance of a successful login so the wait 2 seconds stops 60 minute or often longer gaps appearing in the data and wunderground [...]

As I said, this will appear in v3.7.0. If you can't wait that long, then download the version of [restx.py from the repository](https://raw.githubusercontent.com/weewx/weewx/development/bin/weewx/restx.py).

> Wunderfix is more of a problem as it just crashes out after every few uploads so it takes a lot of rerunning just to get the data uploaded. So - I'm looking for a quick hack to make it more error tolerant. the code is well documented so it shouldn't be too hard to fix for a one off catch up.

Is this a call for help? If so, sorry, but it's just not a priority. You, of course, are welcome to change your own version of wunderfixer. As I stated before, I don't like the idea of a long-running wunderfixer, but one thing you could do is have it return a non-zero system error code in the event of a bad login. Then it would be easy to come up with a shell script that just reruns wunderfixer if it returns a non-zero code.