weewx/weewx #217: the upgrade process mangle weewx.conf

matthewwall, 2017-02-18:
doing a setup.py or deb or rpm upgrade inserts sections and comments into the weewx.conf file.

the upgrade process should make an old conf file compatible with the new release, but it should not reorganize the config file, add comments to the config where there were none before, or add sections other than those necessary for compatibility.

examples of bad behavior:

- sections added to StdReport section: StandardReport, FTP, Rsync
- comments inserted into every single stanza
- top-level stanzas rearranged

tkeffer, 2017-03-18:
How do we square this with issue #18?

matthewwall, 2017-03-18:
the only truly bad behavior is adding sections that were not there before, for example StandardReport, FTP, Rsync on an installation that has only custom reports.  inserting those stanzas results in unexpected, sometimes broken behavior.

"do no harm by inserting stanzas" should apply to *every* stanza.  for example, if a user does not have a StdTimeSync stanza, the update process should not insert one.

the only exception would be if a stanza is required for new, required functionality (e.g., DatabaseTypes)

the re-insertion of comments and re-arranging of stanzas does not affect functionality, so that should be ok (i find it irritating because i remove all the comments from my production configs, and i use a specific order that is different from the weewx default, but i'm a corner case).

tkeffer, 2017-03-21:
OK, although I'm a bit fuzzy about what constitutes a "new" stanza. At one point, `StdTimeSync` was a new service.

tkeffer, 2018-08-16:
The way it works now, any section that is missing in the user's `weewx.conf` file, but present in the incoming distribution `weewx.conf` file, is injected during the merge process. So, if a user takes `[[StandardReport]]` out of his config file, the merge process just puts it right back in.

We could get rid of this behavior, but there are a few "new" sections that truly need to be injected. This includes
```
StdRegistry
AWEKAS
WOW
RSYNC
StdWXCalculate
Calculations
```
They didn't exist before, so we depend on this merge process to include them.

So, it seems like the simplest solution is to regard these as "special" sections and inject them. If a section is not on this list, it does not get injected.

That means, of course, that the "special" list has to be maintained, but that doesn't seem like too big a burden.

Sound good?

tkeffer, 2018-09-06:
OK, I think I've fixed this as of commit https://github.com/weewx/weewx/commit/27bda02ad3e0c36b2ffeff3998fddabc6899908d

The strategy is to no longer do a merge. Instead, each upgrade adds or changes whatever it needs in `weewx.conf`. The only thing that gets merged is the final version number.

This means there is now a *lot* of logic in the upgrade code, but I think I've captured everything. 

It needs to be tested thoroughly!

gjr80, 2018-09-06:
> It needs to be tested thoroughly!

Yes, and I suspect its not fixed yet. Just went to install latest on a clean Stretch VM and received the following (included the whole thing from ./setup.py onwards, trace at the bottom). Wondered what was going on at first, I run a bash script to install from master and thought this sounded like your install work but that is in development. Checked my script and it turned out I had set it to install development and forgot to change it back. At least that makes a bit more sense.

```
./setup.py install
running install
running build
running build_py
creating build
creating build/lib.linux-x86_64-2.7
copying bin/daemon.py -> build/lib.linux-x86_64-2.7
creating build/lib.linux-x86_64-2.7/schemas
copying bin/schemas/wview.py -> build/lib.linux-x86_64-2.7/schemas
copying bin/schemas/__init__.py -> build/lib.linux-x86_64-2.7/schemas
creating build/lib.linux-x86_64-2.7/user
copying bin/user/extensions.py -> build/lib.linux-x86_64-2.7/user
copying bin/user/__init__.py -> build/lib.linux-x86_64-2.7/user
creating build/lib.linux-x86_64-2.7/weecfg
copying bin/weecfg/database.py -> build/lib.linux-x86_64-2.7/weecfg
copying bin/weecfg/config.py -> build/lib.linux-x86_64-2.7/weecfg
copying bin/weecfg/__init__.py -> build/lib.linux-x86_64-2.7/weecfg
copying bin/weecfg/extension.py -> build/lib.linux-x86_64-2.7/weecfg
creating build/lib.linux-x86_64-2.7/weedb
copying bin/weedb/sqlite.py -> build/lib.linux-x86_64-2.7/weedb
copying bin/weedb/__init__.py -> build/lib.linux-x86_64-2.7/weedb
copying bin/weedb/mysql.py -> build/lib.linux-x86_64-2.7/weedb
creating build/lib.linux-x86_64-2.7/weeimport
copying bin/weeimport/weeimport.py -> build/lib.linux-x86_64-2.7/weeimport
copying bin/weeimport/wuimport.py -> build/lib.linux-x86_64-2.7/weeimport
copying bin/weeimport/csvimport.py -> build/lib.linux-x86_64-2.7/weeimport
copying bin/weeimport/cumulusimport.py -> build/lib.linux-x86_64-2.7/weeimport
copying bin/weeimport/__init__.py -> build/lib.linux-x86_64-2.7/weeimport
creating build/lib.linux-x86_64-2.7/weeplot
copying bin/weeplot/utilities.py -> build/lib.linux-x86_64-2.7/weeplot
copying bin/weeplot/genplot.py -> build/lib.linux-x86_64-2.7/weeplot
copying bin/weeplot/__init__.py -> build/lib.linux-x86_64-2.7/weeplot
creating build/lib.linux-x86_64-2.7/weeutil
copying bin/weeutil/weeutil.py -> build/lib.linux-x86_64-2.7/weeutil
copying bin/weeutil/Moon.py -> build/lib.linux-x86_64-2.7/weeutil
copying bin/weeutil/ftpupload.py -> build/lib.linux-x86_64-2.7/weeutil
copying bin/weeutil/Sun.py -> build/lib.linux-x86_64-2.7/weeutil
copying bin/weeutil/__init__.py -> build/lib.linux-x86_64-2.7/weeutil
copying bin/weeutil/rsyncupload.py -> build/lib.linux-x86_64-2.7/weeutil
creating build/lib.linux-x86_64-2.7/weewx
copying bin/weewx/units.py -> build/lib.linux-x86_64-2.7/weewx
copying bin/weewx/engine.py -> build/lib.linux-x86_64-2.7/weewx
copying bin/weewx/accum.py -> build/lib.linux-x86_64-2.7/weewx
copying bin/weewx/uwxutils.py -> build/lib.linux-x86_64-2.7/weewx
copying bin/weewx/filegenerator.py -> build/lib.linux-x86_64-2.7/weewx
copying bin/weewx/wxservices.py -> build/lib.linux-x86_64-2.7/weewx
copying bin/weewx/tags.py -> build/lib.linux-x86_64-2.7/weewx
copying bin/weewx/manager.py -> build/lib.linux-x86_64-2.7/weewx
copying bin/weewx/crc16.py -> build/lib.linux-x86_64-2.7/weewx
copying bin/weewx/wxmanager.py -> build/lib.linux-x86_64-2.7/weewx
copying bin/weewx/reportengine.py -> build/lib.linux-x86_64-2.7/weewx
copying bin/weewx/__init__.py -> build/lib.linux-x86_64-2.7/weewx
copying bin/weewx/cheetahgenerator.py -> build/lib.linux-x86_64-2.7/weewx
copying bin/weewx/almanac.py -> build/lib.linux-x86_64-2.7/weewx
copying bin/weewx/qc.py -> build/lib.linux-x86_64-2.7/weewx
copying bin/weewx/wxformulas.py -> build/lib.linux-x86_64-2.7/weewx
copying bin/weewx/wxengine.py -> build/lib.linux-x86_64-2.7/weewx
copying bin/weewx/imagegenerator.py -> build/lib.linux-x86_64-2.7/weewx
copying bin/weewx/restx.py -> build/lib.linux-x86_64-2.7/weewx
copying bin/weewx/station.py -> build/lib.linux-x86_64-2.7/weewx
creating build/lib.linux-x86_64-2.7/weewx/drivers
copying bin/weewx/drivers/acurite.py -> build/lib.linux-x86_64-2.7/weewx/drivers
copying bin/weewx/drivers/fousb.py -> build/lib.linux-x86_64-2.7/weewx/drivers
copying bin/weewx/drivers/ultimeter.py -> build/lib.linux-x86_64-2.7/weewx/drivers
copying bin/weewx/drivers/ws23xx.py -> build/lib.linux-x86_64-2.7/weewx/drivers
[... 216 lines cut ...]
Traceback (most recent call last):
  File "./setup.py", line 649, in <module>
    'util/udev/rules.d/ws28xx.rules'])
  File "/usr/lib/python2.7/distutils/core.py", line 151, in setup
    dist.run_commands()
  File "/usr/lib/python2.7/distutils/dist.py", line 953, in run_commands
    self.run_command(cmd)
  File "/usr/lib/python2.7/distutils/dist.py", line 972, in run_command
    cmd_obj.run()
  File "/usr/lib/python2.7/distutils/command/install.py", line 613, in run
    self.run_command(cmd_name)
  File "/usr/lib/python2.7/distutils/cmd.py", line 326, in run_command
    self.distribution.run_command(command)
  File "/usr/lib/python2.7/distutils/dist.py", line 972, in run_command
    cmd_obj.run()
  File "./setup.py", line 154, in run
    install_data.run(self)
  File "/usr/lib/python2.7/distutils/command/install_data.py", line 74, in run
    (out, _) = self.copy_file(data, dir)
  File "./setup.py", line 137, in copy_file
    rv = self.process_config_file(f, install_dir, **kwargs)
  File "./setup.py", line 198, in process_config_file
    weecfg.reorder_to_ref(config_dict)
AttributeError: 'module' object has no attribute 'reorder_to_ref'
```

tkeffer, 2018-09-06:
Oops. Try it now.

gjr80, 2018-09-06:
development installs fine now.

closed 2018-09-06 by tkeffer