weewx/weewx #218: Monthly Climatological Summary report issue with 00:00

awcolley, 2017-02-24:
I just noticed that for this month one record:
08   31.9   44.1  00:00   25.8  00:00   33.1    0.0   0.00    2.6   14.0  01:14    304
shows the max 44.1 occurred at 00:00 _and_ the min 25.8 also occurred at 00:00.

I assume this is due to rounding the timestamp (for the report output) to the nearest minute. This may be alright for any other minute of the day, but it is confusing for the last minute of the day. Without looking at anything other than that one line in the report, do you know whether the day began at 44.1 and ended at 25.8 or vice versa?

I don't have a problem with the timestamp always being truncated to the minute; however, (if that is not acceptable to weewx) it should at least be truncated to the minute for 23:59:xx when constructing the Monthly Climatological Summary report.

gjr80, 2017-02-25:
Have you looked at the underlying data (ie the `outTemp` daily summary in the database) to see what timestamps are recorded for `max` and `min` (ie `maxtime` and `mintime`) on the 8th? You may well be right in your assumption, but if the daily summary ro for the 8th indeed has the 'midnight' timestamp against `maxtime` and `mintime` then the issue is not with the reports but rather the integrity of the underlying data.

The timestamp in the report is formatted with `%H:%m` so a timestamp representing 23:59.xx will display as 23:59, it will not round to 00:00.

awcolley, 2017-02-25:
sqlite> select
datetime(dateTime,'unixepoch','localtime'),min,datetime(mintime,'unixepoch','localtime'),max,datetime(maxtime,'unixepoch','localtime')
from archive_day_outTemp where datetime(dateTime,'unixepoch','localtime')
>= "2017-02-08" AND datetime(dateTime,'unixepoch','localtime') <
"2017-02-09";

2017-02-08 00:00:00 | 25.8 | 2017-02-09 00:00:00 | 44.1 | 2017-02-08
00:00:02

So, for the day 2017-02-08, the timestamp for mintime is indeed exactly
midnight on 02-09.

So apparently it isn't a rounding problem in the report, but a problem in
weewx's determining which data to include when determining the mix/max for
a date.

gjr80, 2017-02-26:
> 2017-02-08 00:00:00 | 25.8 | 2017-02-09 00:00:00 | 44.1 | 2017-02-08 00:00:02

Based on the above the monthly climatological report is doing exactly what is should. For the 8th the max `outTemp` occurred at 2 seconds after midnight at the the start of the day, the min `outTemp` occurred at midnight at the end of the day. So when those times are formatted in `%H:%m` format they both show 00:00.

Has weeWX correctly determined what `outTemp` data to display for the 8th; based on the above yes it has, a given days obs (ie a 'statistical day') includes everything after midnight at the start of the day up to and including midnight at the end of the day. 

Possible solutions:
1. Use `%H:%m:%S` rather than `%H:%m` to formats times, solves the ambiguity but will make an already cramped report even worse. I have no idea where the report format comes from, I see it referred to elsewhere as a 'NOAA format' report, so NOAA? I am not in the USA so I have no idea if changing the format is seen as breaking some cardinal rule.
2. Change what weeWX defines as a 'statistical day'. This is one of the foundation stones of weeWX' data recording and reporting and it affects far more than just this report. Besides, there is nothing wrong with what weeWX records/reports as it is correct and consistent, it is the interpretation of the results that is the issue. I don't see this changing. In any case, changing the 'statistical day' will still result in similar incidents like this occurring given the right circumstances, it just won't happen a midnight.
3. Don't include loop data in the calculation of min/max stats (ie set `loop_hilo = False` in `weewx.conf` `[StdArchive]`). This will prevent ambiguities like this (the 00:00:02 max would (most likely) have been recorded as 00:05:00) but has the downside of missing out on the extra 'precision' arising out of using the loop data.
4. Rebuild your daily summary for the 8th if using weeWX 3.7.0, or rebuild all daily summaries if using an earlier weeWX version. This has the same effect as retrospectively applying `loop_hilo = False` but of course has no affect on future data. Note, the extra 'precision' from the loop data will be lost for all obs for all days rebuilt.
5. Live with the issue accepting it will seldom occur.

I guess we will see what Tom says when he is back.

awcolley, 2017-02-26:
I agree with you that my original premise for this Issue was incorrect; the report generating function is clearly not doing anything wrong, merely presenting the information available to it in a predefined format.

Regarding your item 1: 
The report format seems to be a modification of the old NOAA Local Climatological Data Monthy Summary front page. Just FYI, the NOAA page includes e.g. snow stats, observed weather, and sky cover that isn't available on consumer weather instrument packages, and it omits the timestamps for max/min. The weewx report format is very similar to the one wview used.

I personally like having the timestamps for min/max (i.e. versus the actual NOAA omission of same).

Nonetheless, in this instance %H:%m:%S would not be less ambiguous; only %d:%H:%m:%S would be unambiguous. Personally, I think it would be better to avoid the ambiguity at the source (the **archive_day_outTemp** table).

Regarding your items 3 and 4:
Yes, it is an option to not include loop packets in the min/max stats.

I'm not sure how weewx handles the min/max temp provided in the Davis Vantage archive record (DMP); specifically if that information is used for the min/max stats when loop packets are not being used. However, I can see that the DMP min/max temp information is not stored in the weewx **archive** table; so that information is clearly not available for regeneration of the **archive_day_outTemp** table.

I would prefer to include loop packets in min/max stats because DMP does _not_ include timestamps for min/max temp (even if weewx uses those values somehow).

Regarding your item 5:
468 out of 4088 records in my **archive_day_outTemp** table contain an instance of "Day+1 at 00:00:00" for either mintime or maxtime. This is >10%, which is not "seldom".

If, however, you only meant "seldom" to apply to the occurrence of both mintime and maxtime being 00:00 (ignoring date and seconds), yes that is appropriately called "seldom". In my database, there is only one occurrence of this (02/08) because I have only been using weewx since July last year (because wview did not store a min/max temp or temp timestamp in its database, either).

Regarding your item 2:
Yes, "midnight" is an ambiguous term because it is an instant in time (i.e. of time span 0) that can refer to the instant after which today starts ("midnight last night") or before which today ends ("midnight tonight"). Legally, most contracts use 11:59pm or 12:01am to avoid this ambiguity.

However, the software world (and ISO) is unambiguous on which date 00:00:00 occurs; it is the same date as the following second 00:00:01, and the date following the previous second 23:59:59.

ISO and "military" time allow for "midnight tonight" being 24:00:00 or 2400, but programming language date-time functions/objects do not support an hour of 24.

Leap seconds may technically be 24:00:00 (and followed by 00:00:00), but the software world and consumer weather instruments ignore them (they just are off by a second until their clocks are next synced to some external time source).

Further consideration:
The loop packet (or DMP archive record) is reporting: (1) values that have occurred sometime since the last packet, e.g. precipitation and wind gust; (2) average values over the time period since the last packet, e.g. wind speed; and (3) values measured when the packet was generated, e.g. temperature, humidity, barometric pressure. (Note: some instruments report average temperature, etc. rather than instantaneous, so may not have any type 3.)

To me, for a loop/DMP generated at 00:00:00, values of type 3 should be assigned to the date associated with 00:00:00; while values of types 1 and 2 should be assigned to the date associated with 23:59:59 (i.e. the previous second).

If a loop/DMP spans 00:00:00, types 1 and 2 should be assigned the date which contained the larger portion of the loop/DMP timespan. If the loop/DMP timespan contains the same number of seconds prior to 00:00:00 as including and after 00:00:00, then this is just like rounding 0.5 (i.e. some arbitrary convention).

However, that approach definitely increases software complexity. If avoiding that complexity is the preferred approach, then a loop/DMP that would otherwise be timestamped 00:00:00 (and **archive** records at 00:00:00 when regenerating the daily summary tables) should, in my opinion, be assumed to be timestamped 23:59:59 to avoid ambiguity (since 24:00:00 is not an option).

gjr80, 2017-02-26:
A few comments

>Nonetheless, in this instance %H:%m:%S would not be less ambiguous; only %d:%H:%m:%S would be unambiguous. Personally, I think it would be better to avoid the ambiguity at the source (the archive_day_outTemp table).

There is no ambiguity in the daily summaries, the ambiguity is in the output layer. The daily summary for, say 8 February, will only have timestamps >= 1 seconds after midnight at the start of the 8th up to an including midnight at the end of the 8th. The reason being that an archive record for a station is rearward looking (historical) data rather than forward looking (future) data. It would be ideal if the archive record was forward looking so that weeWX' notion of an archive day would align with our notion of a day, but unfortunately those type of stations don't exist yet. At first glance there may appear to be ambiguity in the daily summaries but when you understand the weeWX notion of an archive or statistical day there is no ambiguity.

> I'm not sure how weewx handles the min/max temp provided in the Davis Vantage archive record (DMP); specifically if that information is used for the min/max stats when loop packets are not being [...]

WeeWX makes no use of max/min data in the Vantage archive record. max/mins (and their times) are derived from loop packets and archive records (if `loop_hilo = True`) and archive records only (if `loop_hilo = False`). Yes, regeneration (backfilling) the daily summaries effectively drops any loop high/low info, that is why it is effectively applying `loop_hilo = False` to your historical data

>I would prefer to include loop packets in min/max stats because DMP does not include timestamps for min/max temp (even if weewx uses those values somehow).

Likewise, I cringe each time I have to do some sort of backfill of the daily summaries as I know I am effectively losing data.

>Regarding your item 5:
468 out of 4088 records in my archive_day_outTemp table contain an instance of "Day+1 at 00:00:00" for either mintime or maxtime. This is >10%, which is not "seldom".

I was referring to both min/max effectively appearing of a %H:%M boundary, but I take your point having anything on a 00:00:00 +1 day boundary will lead to an ambiguous monthly report output.

>ISO and "military" time allow for "midnight tonight" being 24:00:00 or 2400, but programming language date-time functions/objects do not support an hour of 24.

Interesting, in my time in the military we were always trained to avoid 0000 or 2400 because of the possible ambiguity.

The data recorded by WeeWX is unambiguous regarding midnight as well, once you understand the archive or statistical day that is used. The ambiguity is this case is in the output layer.

> However, that approach definitely increases software complexity. If avoiding that complexity is the preferred approach, then a loop/DMP that would otherwise be timestamped 00:00:00 (and archive [...]

Definitely one for Tom's opinion. WeeWX effectively takes the view that (for straight up observations) archive records contain an average of the observation over the archive period and loop packets effectively contain instantaneous values (obviously rain is the exception where it is the sum over the period concerned in both cases).

awcolley, 2017-02-27:
Another observation ... the dateTime value stored in the archive_day_* tables is the epoch seconds at Date:00:00:00 (Local Time) ... so one presumes that record summarizes the data over the 86400 seconds beginning at Date:00:00:00 (LT), the last second of record being Date:23:59:59 (LT). And, thus, one presumes a mintime/maxtime of 00:00 to be "at the beginning of the day".

tkeffer, 2017-03-18:
Throughout weewx, intervals are timestamped with their *ending time*. So, a 5 minute interval that ends at midnight is timestamped `0000`. With this convention, it does sound like an event, such as max temperature, that happened at `0000` happened in the following day.

I suppose as an alternative, we could have taken the convention that intervals are timestamped with their *starting* time. Then such an event would have been timestamped `2355`, which is less ambiguous. 

But, the reason why the former convention was chosen is because that's the way every weather station I've seen works. They all timestamp an interval with the ending time. Your Vantage certainly does. 

Imagine the confusion trying to reconcile archive records coming off your weather station with a database that timestamps them 5 minutes earlier.

Neither convention is ideal, but it's too late to go back now.

closed 2017-03-18 by tkeffer