weewx/weewx #219: accumulator exception in weewx 3.7.0b2

matthewwall, 2017-02-26:
i had been running the ws3000 driver for awhile when i got this exception:

```
Traceback (most recent call last):
  File "./bin/weewxd", line 64, in <module>
    weewx.engine.main(options, args)
  File "/Users/Shared/weewx/bin/weewx/engine.py", line 879, in main
    engine.run()
  File "/Users/Shared/weewx/bin/weewx/engine.py", line 201, in run
    self.dispatchEvent(weewx.Event(weewx.POST_LOOP))
  File "/Users/Shared/weewx/bin/weewx/engine.py", line 223, in dispatchEvent
    callback(event)
  File "/Users/Shared/weewx/bin/weewx/engine.py", line 570, in post_loop
    self._software_catchup()
  File "/Users/Shared/weewx/bin/weewx/engine.py", line 633, in _software_catchup
    record = self.old_accumulator.getRecord()
AttributeError: 'NoneType' object has no attribute 'getRecord'
```

these are the loop packets just prior to the exception:

```
LOOP:   2017-02-25 20:19:04 EST (1488071944) altimeter: None, appTemp: None, barometer: None, cloudbase: 2870.16645671, dateTime: 1488071944, dewpoint: 63.2912675905, extraHumid3: 70, extraTemp3: 75.02, heatindex: 75.92, humidex: 85.9240450083, inDewpoint: 65.9669846148, inHumidity: 63, inTemp: 79.7, maxSolarRad: None, outHumidity: 65, outTemp: 75.92, pressure: None, rainRate: 0, usUnits: 1, windchill: None
LOOP:   2017-02-25 20:19:33 EST (1488071973) altimeter: None, appTemp: None, barometer: None, cloudbase: 2870.16645671, dateTime: 1488071973, dewpoint: 63.2912675905, extraHumid3: 70, extraTemp3: 75.02, heatindex: 75.92, humidex: 85.9240450083, inDewpoint: 65.9669846148, inHumidity: 63, inTemp: 79.7, maxSolarRad: None, outHumidity: 65, outTemp: 75.92, pressure: None, rainRate: 0, usUnits: 1, windchill: None
LOOP:   2017-02-25 20:20:07 EST (1488072007) altimeter: None, appTemp: None, barometer: None, cloudbase: 2870.16645671, dateTime: 1488072007, dewpoint: 63.2912675905, extraHumid3: 70, extraTemp3: 75.02, heatindex: 75.92, humidex: 85.9240450083, inDewpoint: 65.9669846148, inHumidity: 63, inTemp: 79.7, maxSolarRad: None, outHumidity: 65, outTemp: 75.92, pressure: None, rainRate: 0, usUnits: 1, windchill: None
LOOP:   2017-02-25 20:20:33 EST (1488072033) altimeter: None, appTemp: None, barometer: None, cloudbase: 2870.16645671, dateTime: 1488072033, dewpoint: 63.2912675905, extraHumid3: 70, extraTemp3: 75.02, heatindex: 75.92, humidex: 85.9240450083, inDewpoint: 65.9669846148, inHumidity: 63, inTemp: 79.7, maxSolarRad: None, outHumidity: 65, outTemp: 75.92, pressure: None, rainRate: 0, usUnits: 1, windchill: None
REC:    2017-02-25 20:20:00 EST (1488072000) altimeter: None, appTemp: None, barometer: None, cloudbase: 2870.16645671, dateTime: 1488072000.0, dewpoint: 63.2912675905, ET: None, extraHumid3: 69.5, extraTemp3: 75.02, heatindex: 75.92, humidex: 85.9240450083, inDewpoint: 65.9669846148, inHumidity: 63.0, inTemp: 79.7, interval: 5, maxSolarRad: None, outHumidity: 65.0, outTemp: 75.92, pressure: None, rainRate: 0.0, usUnits: 1, windchill: None, windrun: 0.0
LOOP:   2017-02-25 20:20:41 EST (1488072041) altimeter: None, appTemp: None, barometer: None, cloudbase: 2870.16645671, dateTime: 1488072041, dewpoint: 63.2912675905, extraHumid3: 70, extraTemp3: 75.02, heatindex: 75.92, humidex: 85.9240450083, inDewpoint: 65.9669846148, inHumidity: 63, inTemp: 79.7, maxSolarRad: None, outHumidity: 65, outTemp: 75.92, pressure: None, rainRate: 0, usUnits: 1, windchill: None
LOOP:   2017-02-25 20:29:31 EST (1488072571) altimeter: None, appTemp: None, barometer: None, cloudbase: 2870.16645671, dateTime: 1488072571, dewpoint: 63.2912675905, extraHumid3: 69, extraTemp3: 75.02, heatindex: 75.92, humidex: 85.9240450083, inDewpoint: 65.9669846148, inHumidity: 63, inTemp: 79.7, maxSolarRad: None, outHumidity: 65, outTemp: 75.92, pressure: None, rainRate: 0, usUnits: 1, windchill: None
```

this is a setup.py install of 3.7.0b2 on macosx with ws3000 driver

tkeffer, 2017-03-05:
Fixed (I think) in commit e34a79e99352244d39bc67dadc5d8ab25925edda

I have no way of testing this (I'm in a hotel in Ft. Lauderdale), so give it a good workout!

tkeffer, 2017-03-08:
Matthew: is this patch working for you? Can we close this issue?

matthewwall, 2017-03-09:
in two days of testing it has not happened again.  however, i'm not sure what sequence of events caused it the one time it happened.

closed 2017-03-09 by matthewwall

tkeffer, 2017-03-09:
I'm not positive, but I think the exception could happen if there were no loop packets received during an archive period. Would that be possible with your setup?

matthewwall, 2017-03-09:
that would have been possible - i was testing a new driver for a usb device (fine offset hp3000 t/h sensors).  i can test specifically for that case, with and without the patch.  but i won't be able to do it until friday evening at the earliest.

gjr80, 2017-03-09:
Sorry, was bored this afternoon so I loaded 3.7.0b2 onto a VM and let it run for a few minutes under simulator with an archive interval of 60 seconds. All was good. Simulated no loop data in an archive interval by changing `[Simulator]` `loop_interval` to 90 seconds. When the next loop packet came in the following was thrown:

```
Mar  9 15:58:50 jessie3 weewx[1989]: engine: Caught unrecoverable exception in engine:
Mar  9 15:58:50 jessie3 weewx[1989]:     ****  'NoneType' object has no attribute 'getRecord'
Mar  9 15:58:50 jessie3 weewx[1989]:     ****  Traceback (most recent call last):
Mar  9 15:58:50 jessie3 weewx[1989]:     ****    File "/home/weewx/bin/weewx/engine.py", line 879, in main
Mar  9 15:58:50 jessie3 weewx[1989]:     ****      engine.run()
Mar  9 15:58:50 jessie3 weewx[1989]:     ****    File "/home/weewx/bin/weewx/engine.py", line 201, in run
Mar  9 15:58:50 jessie3 weewx[1989]:     ****      self.dispatchEvent(weewx.Event(weewx.POST_LOOP))
Mar  9 15:58:50 jessie3 weewx[1989]:     ****    File "/home/weewx/bin/weewx/engine.py", line 223, in dispatchEvent
Mar  9 15:58:50 jessie3 weewx[1989]:     ****      callback(event)
Mar  9 15:58:50 jessie3 weewx[1989]:     ****    File "/home/weewx/bin/weewx/engine.py", line 570, in post_loop
Mar  9 15:58:50 jessie3 weewx[1989]:     ****      self._software_catchup()
Mar  9 15:58:50 jessie3 weewx[1989]:     ****    File "/home/weewx/bin/weewx/engine.py", line 633, in _software_catchup
Mar  9 15:58:50 jessie3 weewx[1989]:     ****      record = self.old_accumulator.getRecord()
Mar  9 15:58:50 jessie3 weewx[1989]:     ****  AttributeError: 'NoneType' object has no attribute 'getRecord'
Mar  9 15:58:50 jessie3 weewx[1989]:     ****  Exiting.
```

which looks familiar.

Installed 3.7.0b4, deleted `weewx.sdb` and started up again with the same archive interval and loop interval settings, all seems good:

```
Mar  9 16:01:46 jessie3 weewx[27124]: engine: Starting up weewx version 3.7.0b4
Mar  9 16:01:46 jessie3 weewx[27124]: engine: Clock error is -0.04 seconds (positive is fast)
Mar  9 16:01:46 jessie3 weewx[27124]: engine: Starting main packet loop.
Mar  9 16:03:17 jessie3 weewx[27124]: cheetahgenerator: Generated 14 files for report StandardReport in 0.43 seconds
Mar  9 16:03:17 jessie3 weewx[27124]: imagegenerator: Generated 12 images for StandardReport in 0.07 seconds
Mar  9 16:03:17 jessie3 weewx[27124]: copygenerator: copied 14 files to /home/weewx/public_html
Mar  9 16:04:47 jessie3 weewx[27124]: manager: Added record 2017-03-09 16:04:00 AEST (1489039440) to database 'weewx.sdb'
Mar  9 16:04:47 jessie3 weewx[27124]: manager: Added record 2017-03-09 16:04:00 AEST (1489039440) to daily summary in 'weewx.sdb'
Mar  9 16:04:47 jessie3 weewx[27124]: cheetahgenerator: Generated 14 files for report StandardReport in 0.15 seconds
Mar  9 16:04:47 jessie3 weewx[27124]: imagegenerator: Generated 12 images for StandardReport in 0.07 seconds
Mar  9 16:04:47 jessie3 weewx[27124]: copygenerator: copied 0 files to /home/weewx/public_html
Mar  9 16:06:17 jessie3 weewx[27124]: manager: Added record 2017-03-09 16:05:00 AEST (1489039500) to database 'weewx.sdb'
Mar  9 16:06:17 jessie3 weewx[27124]: manager: Added record 2017-03-09 16:05:00 AEST (1489039500) to daily summary in 'weewx.sdb'
Mar  9 16:06:17 jessie3 weewx[27124]: cheetahgenerator: Generated 14 files for report StandardReport in 0.16 seconds
Mar  9 16:06:17 jessie3 weewx[27124]: imagegenerator: Generated 12 images for StandardReport in 0.06 seconds
Mar  9 16:06:17 jessie3 weewx[27124]: copygenerator: copied 0 files to /home/weewx/public_html
```

I note the odd 'Added record' times but I guess that is a consequence of the oddball settings I am using. At least it looks like the 'no loop packets in an archive interval' issue is fixed. Of course whether that was Matthew's cause or not I don't know.

tkeffer, 2017-03-09:
Thanks, Gary! I feel better now that we have a handle on the cause and that this commit fixes it.