weewx/weewx #223: adopt a cheetah

matthewwall, 2017-03-05:
the cheetah documentation has been unavailable for some time now (since late 2016?)

we should see about hosting the cheetah docs at weewx.com

tkeffer, 2017-03-05:
I agree.

I wish someone would take ownership of the project. I really don't want to, but it's an incredibly valuable piece of software. I'll w rite rtyler and ask his intentions.

awcolley, 2017-03-09:
Is this not the current documentation? https://pythonhosted.org/Cheetah/users_guide/index.html

Or is this a fork of the Cheetah used in weewx?

I was using this earlier today and what I learned there worked in weewx (though I didn't go very wide or deep in Cheetah).

matthewwall, 2017-03-09:
there were a couple of references to cheetahtemplate.org in the weewx docs.  these have been replaced with the pythonhosted.org url as of commit 9b37e15

tkeffer, 2017-03-09:
> Is this not the current documentation? https://pythonhosted.org/Cheetah/users_guide/index.html

It is, and there seems to be more information there than the last time I looked, but perhaps my memory is faulty.

I've had an exchange with the original Cheetah author. He passed the code on to someone else, who seems to be disengaged. There are other groups in a similar situation. If someone in the weewx community wanted to take Cheetah over, the original author is very willing to pass it on.

tkeffer, 2017-03-17:
The link to the pythonhosted documentation will serve for now. Closing, but we can reopen if someone has an interest in taking on Cheetah. I don't.

closed 2017-03-17 by tkeffer