weewx/weewx #229: daily temperatures hourly average not working

paolobenve, 2017-03-13:
I'm trying to implement the example given in the Customizing guide in the section "Including the same SQL type more than once in a plot":

```
[[[daytemp_with_avg]]]
    [[[[avgTemp]]]]
        data_type = outTemp
        aggregate_type = avg
        aggregate_interval = 3600
        label = Avg. Temp.
    [[[[outTemp]]]]
```

but it only produces the temperature plot, not the average one:

[image]

tkeffer, 2017-03-13:
The plot is working properly. If you look closely, you'll see red dots on the hour boundaries.

Your value for `line_gap_fraction` is too small. Try 0.1.

closed 2017-03-13 by tkeffer

paolobenve, 2017-03-13:
ok, let me try

but shouldn't the average value be plotted along all the period it refers to, in order to plot something like a line?