weewx/weewx #233: should installer touch skins?

matthewwall, 2017-03-15:
is it ok for the installer to touch skins?

specifically, should we fix the installer so that it modifies any search_list_extension that refers to examples.xsearch?

proposed behavior would be to look for a skin.conf in each subdirectory in the skins directory, then replace examples.xsearch.MyXSearch with user.stats.MyStats, and copy examples.stats.py to the user directory.

this could be good 'practice' for how we will deal with installing the replacement for the Standard skin...

refer to this thread:

https://groups.google.com/forum/#!topic/weewx-user/ydhQftFQ2Ys

tkeffer, 2017-03-17:
If I had to do it over again, I would leave behind a dummy directory with symbolic links to the new location. Of course, "the new location" varies depending on install method, so maybe that wouldn't be so simple.

Yes, we could do this, but it sounds complicated.  I'd rather spend my time on other things.

I don't see how it serves as practice. When we introduce a new skin, it will have a new name and live in the `skins` subdirectory. It should be up to the user to activate it.

dcapslock, 2017-03-18:
I suggest that skins be looked at but not touched. Rather than any changes, provide an upgrade report that details any upgrade issues found. Can the different upgrade methods put up some kind of information message at the end?

tkeffer, 2017-03-20:
I like that idea better.

We should first, "Do no harm." If we can't, then warn the user about any changes s/he should do.

tkeffer, 2017-03-25:
I'm not sure what the "action item" is for this issue. Is there anything in particular we should be doing?

matthewwall, 2017-03-25:
this issue was created in response to people who were using examples from the examples directory instead of copying to the user directory.  probably not much need for adding a warning for that.

i'll open another issue for how to deal with newskin

closed 2017-03-25 by matthewwall