weewx/weewx #235: Cannot establish comms over MQTT network.

mriotimpactlabs, 2017-03-16:
Hello,

I am trying to run Weewx 3.7.0 with the MQTT client extension.  Everything seems to be working out just fine but I have been unsuccessful in establishing communication between Weewx and any MQTT client.  I have no username or password.  My topic is new_mountain_test.  

Below is my weewx.conf,  and syslog data:

# WEEWX CONFIGURATION FILE
#
# Copyright (c) 2009-2015 Tom Keffer <tkeffer@gmail.com>
# See the file LICENSE.txt for your rights.

##############################################################################

# This section is for general configuration information.

# Set to 1 for extra debug info, otherwise comment it out or set to zero
debug = 0

# Root directory of the weewx data file hierarchy for this station
WEEWX_ROOT = /

# How long to wait before timing out a socket (FTP, HTTP) connection
socket_timeout = 20

# Do not modify this. It is used when installing and updating weewx.
version = 3.7.0

##############################################################################

#   This section is for information about the station.

[Station]
    
    # Description of the station location
    location = "Santa's Workshop, North Pole"
    
    # Latitude and longitude in decimal degrees
    latitude = 41.008
    longitude = -71.005
    
    # Altitude of the station, with unit it is in. This is downloaded from
    # from the station if the hardware supports it.
    altitude = 0, meter
    
    # Set to type of station hardware. There must be a corresponding stanza
    # in this file with a 'driver' parameter indicating the driver to be used.
    station_type = Simulator
    
    # If you have a website, you may specify an URL
    #station_url = http://www.example.com
    
    # The start of the rain year (1=January; 10=October, etc.). This is
    # downloaded from the station if the hardware supports it.
    rain_year_start = 1
    
    # Start of week (0=Monday, 6=Sunday)
    week_start = 6

##############################################################################

[Simulator]
    # This section is for the weewx weather station simulator
    
    # The time (in seconds) between LOOP packets.
    loop_interval = 2.5
    
    # The simulator mode can be either 'simulator' or 'generator'.
    # Real-time simulator. Sleep between each LOOP packet.
    mode = simulator
    # Generator.  Emit LOOP packets as fast as possible (useful for testing).
[... 332 lines cut ...]
    [[Services]]
        # This section specifies the services that should be run. They are
        # grouped by type, and the order of services within each group
        # determines the order in which the services will be run.
        prep_services = weewx.engine.StdTimeSynch
        data_services = ,
        process_services = weewx.engine.StdConvert, weewx.engine.StdCalibrate, weewx.engine.StdQC, weewx.wxservices.StdWXCalculate
        archive_services = weewx.engine.StdArchive
        restful_services = weewx.restx.StdStationRegistry, weewx.restx.StdWunderground, weewx.restx.StdPWSweather, weewx.restx.StdCWOP, weewx.restx.StdWOW, weewx.restx.StdAWEKAS, user.mqtt.MQTT
        report_services = weewx.engine.StdPrint, weewx.engine.StdReport

sudo tail -f /var/log/syslog | grep "MQTT" 

Mar 16 16:23:25 DarkstarTerminal weewx[1888]: restx: MQTT: Published record 2017-03-16 16:23:25 EDT (1489695805)
Mar 16 16:23:27 DarkstarTerminal weewx[1888]: restx: MQTT: Published record 2017-03-16 16:23:28 EDT (1489695808)
Mar 16 16:23:30 DarkstarTerminal weewx[1888]: restx: MQTT: Published record 2017-03-16 16:23:30 EDT (1489695810)
Mar 16 16:23:32 DarkstarTerminal weewx[1888]: restx: MQTT: Published record 2017-03-16 16:23:33 EDT (1489695813)

*Note:  (Debug is set to 0 but there is no useful information when I use "1" or "2")

Why cant I connect?  It is saying that it publishes and I get no errors, yet I see nothing on the other end.  Thanks in advance.

Regards,

Michael

mriotimpactlabs, 2017-03-16:
Sorry about the annoying text formatting.  It did this after I submitted the issue.

matthewwall, 2017-03-16:
the issues list is for tracking bugs and development issues.

your request for help with the mqtt extension belongs in the user group

closed 2017-03-16 by matthewwall

mriotimpactlabs, 2017-03-16:
Oops, sorry.