weewx/weewx #239: Inconsistent use of LOOP packets when recording daily highs & lows

tkeffer, 2017-03-19:
I've noticed that in my own station the daily highs and lows seem to almost always fall on the archive interval boundaries. If LOOP packets were being used to calculate highs and lows, this should happen only rarely.

Not sure, but I think the problem is that the "old accumulator" is getting thrown away before starting a new packet loop, so it's not available when constructing the record. So, the record values only are getting used.

This issue is a place holder until I get it figured out.

gjr80, 2017-03-19:
Just thinking aloud, but wouldn't use of loop packets for highs/lows be an all or nothing situation; ie loop data is either being used or it's not, it's not a case of it being used sometimes? I have looked at my page (not db data) and of the 19 highs/lows I display 8 are on 5 minute boundaries at the moment (statistically you would expect around 20% I guess). Maybe a statistical aberration?

tkeffer, 2017-03-19:
> wouldn't use of loop packets for highs/lows be an all or nothing situation

You would think so. That's why I'm puzzled by my page: I'm calculating 22 highs and lows and all but two are on five minute boundaries. That's way more often than chance would suggest.

The two that are not on a five minute boundary, `barometer` and `dewpoint`, don't seem to be anything special.

I've changed the code a bit so the "old accumulator" does not get thrown away. I'll let it run overnight and see what happens.

tkeffer, 2017-03-20:
I have an operating theory. 

Both archive records *and* the accumulator are used to update the daily highs and lows. However, right now, the record is used first, then the accumulator. 

If, say, the record contains a new high, then the time of that high, which is always on an archive interval boundary, will be first recorded in the daily summary. If the accumulator value is the same high value, and it usually is, then its time does not replace it, even though it is more fine-grained and accurate.

In v3.6, the two updates were done as two separate transactions, leading to a possible skew between archive and daily summaries. In v3.7, they were brought into the same transaction, but the ordering was inadvertently changed.

The fix is to simply swap the update ordering: first update using the accumulator, then the record.

I'll try the change tonight and see what happens.

tkeffer, 2017-03-21:
Fixed in commit https://github.com/weewx/weewx/commit/158ec4a1657be3e9a863e18e5d0402e2d6a5dff5

closed 2017-03-21 by tkeffer