weewx/weewx #248: how to merge newskin into weewx master

matthewwall, 2017-03-29:
what else does the newskin branch need before we merge it into master?

this is what it does as of 29mar17 (see the newskin branch):

* Standard is replaced by an entirely new skin
* the mobile and smartphone skins have been split out into separate skins
* defaults for all skins can be specified in [StdReports]

the branch includes the changes for taking advantage of the defaults in StdReports.  for example, when you set units using wee_config, those units apply to *all* reports.  then you can customize per-report if desired.

i see at least these issues:

1) how to incorporate the new standard skin.  should it be installed in every upgrade?  if so, then using what name (so it does not clobber StandardReport)?  or should users get the new skins only if they do a clean install?

2) what other features/functionality are necessary in the new standard skin before it is ready for release?

3) what to call the new standard skin?

tkeffer, 2017-03-30:
Looks like the merge conflicts were never resolved. If I run it, I get:

~~~
Traceback (most recent call last):
  File "./bin/wee_reports", line 15, in <module>
    import weewx.station
  File "/home/tkeffer/git/weewx/bin/weewx/__init__.py", line 9
    <<<<<<< HEAD
     ^
SyntaxError: invalid syntax
~~~
Easy enough to fix. Pressing on ...

I get the error

~~~
Mar 29 18:12:08 nuc wee_reports[10232]: cheetahgenerator: Generate failed with exception '<class 'NameMapper.NotFound'>'
Mar 29 18:12:08 nuc wee_reports[10232]: cheetahgenerator: **** Ignoring template /home/weewx/skins/Standard/index.html.tmpl
Mar 29 18:12:08 nuc wee_reports[10232]: cheetahgenerator: **** Reason: cannot find 'extraTemp1' while searching for 'obs.label.extraTemp1'
Mar 29 18:12:08 nuc wee_reports[10232]: ****  Traceback (most recent call last):
Mar 29 18:12:08 nuc wee_reports[10232]: ****    File "/home/tkeffer/git/weewx/bin/weewx/cheetahgenerator.py", line 329, in generate
Mar 29 18:12:08 nuc wee_reports[10232]: ****      print >> _file, compiled_template
Mar 29 18:12:08 nuc wee_reports[10232]: ****    File "/usr/lib/python2.7/dist-packages/Cheetah/Template.py", line 1005, in __str__
Mar 29 18:12:08 nuc wee_reports[10232]: ****      rc = getattr(self, mainMethName)()
Mar 29 18:12:08 nuc wee_reports[10232]: ****    File "_home_weewx_skins_Standard_index_html_tmpl.py", line 215, in respond
Mar 29 18:12:08 nuc wee_reports[10232]: ****    File "/usr/lib/python2.7/dist-packages/Cheetah/Template.py", line 1615, in _handleCheetahInclude
Mar 29 18:12:08 nuc wee_reports[10232]: ****      self._CHEETAH__cheetahIncludes[_includeID].respond(trans)
Mar 29 18:12:08 nuc wee_reports[10232]: ****    File "_home_tkeffer_git_weewx_skins_Standard_hilo_inc.py", line 505, in respond
Mar 29 18:12:08 nuc wee_reports[10232]: ****  NotFound: cannot find 'extraTemp1' while searching for 'obs.label.extraTemp1'
~~~

Removed reference to `extraTemp`. Pressing on...

1. Definitely no. Lots of people have spent lots of time tuning their skin, and we shouldn't just step on their efforts. I would say we should call it something else besides `Standard`, say `Spring`. If the subdirectory `Spring` does not exist, we install, otherwise we leave what's there alone.

2. It's pretty darn close! Suggestions:
 * I know I would miss more celestial information, but that could be provided in a separate page.
 * I'd like to see the NOAA pages have a `next` and `prev` link, so one doesn't have to go back up to the main menu to get to the next or previous month.

3. I just through out `Spring`, and Gary may want to call it `Fall`, but I'm feeling pretty good about the weather now~ :-)

matthewwall, 2017-03-30:
how about 'Seasons', then i'll include 4 color schemes.  the colors will automatically change as the year progresses :)

celestial details are included on a separate page - click on the word 'Celestial'.  similarly for sensors, the sensor details are on a separate page.

i do not see the HEAD conflict when i checkout the newskin branch (btw, this is one reason i dislike the use of ====== as a separator in source code - it makes it harder to search for the ==== of a merge/diff)

i'll work on the navigation for noaa pages...

gjr80, 2017-03-30:
>3. I just threw out `Spring`, and Gary may want to call it `Fall`, but I'm feeling pretty good about the weather now~ :-)

Tom, you disappoint me, just when I thought I had you full bottle on the lingo.... `Fall` is what you do when you have had too many beers, `Autumn` is the time of year when the leaves fall off the trees :-)

And no I have no suggestions, `Seasons` sounds fine to me, as does `Spring`.

tkeffer, 2017-03-30:
> celestial details are included on a separate page - click on the word 'Celestial'. similarly for sensors, the sensor details are on a separate page.

I missed that. It's not obvious the headings are links. Perhaps use an underline? Or a right triangle (▶)?

> i do not see the HEAD conflict when i checkout the newskin branch

See: https://github.com/weewx/weewx/blob/newskin/bin/weewx/__init__.py

matthewwall, 2017-04-28:
newskin has been updated with the following:
- fixed HEAD conflict
- NOAA reports are embedded in a page with header and nav elements

tkeffer, 2017-04-28:
Still getting this:

```
Apr 28 05:31:37 nuc wee_reports[28173]: cheetahgenerator: Generate failed with exception '<class 'NameMapper.NotFound'>'
Apr 28 05:31:37 nuc wee_reports[28173]: cheetahgenerator: **** Ignoring template /home/weewx/skins/Standard/index.html.tmpl
Apr 28 05:31:37 nuc wee_reports[28173]: cheetahgenerator: **** Reason: cannot find 'extraTemp1' while searching for 'obs.label.extraTemp1'
Apr 28 05:31:37 nuc wee_reports[28173]: ****  Traceback (most recent call last):
Apr 28 05:31:37 nuc wee_reports[28173]: ****    File "/home/weewx/bin/weewx/cheetahgenerator.py", line 329, in generate
Apr 28 05:31:37 nuc wee_reports[28173]: ****      print >> _file, compiled_template
Apr 28 05:31:37 nuc wee_reports[28173]: ****    File "/usr/lib/python2.7/dist-packages/Cheetah/Template.py", line 1005, in __str__
Apr 28 05:31:37 nuc wee_reports[28173]: ****      rc = getattr(self, mainMethName)()
Apr 28 05:31:37 nuc wee_reports[28173]: ****    File "_home_weewx_skins_Standard_index_html_tmpl.py", line 215, in respond
Apr 28 05:31:37 nuc wee_reports[28173]: ****    File "/usr/lib/python2.7/dist-packages/Cheetah/Template.py", line 1615, in _handleCheetahInclude
Apr 28 05:31:37 nuc wee_reports[28173]: ****      self._CHEETAH__cheetahIncludes[_includeID].respond(trans)
Apr 28 05:31:37 nuc wee_reports[28173]: ****    File "_home_tkeffer_git_weewx_skins_Standard_hilo_inc.py", line 505, in respond
Apr 28 05:31:37 nuc wee_reports[28173]: ****  NotFound: cannot find 'extraTemp1' while searching for 'obs.label.extraTemp1'
```

BTW, `newskin` should really be branched off of `development`, not `master`.

matthewwall, 2017-04-30:
extraTemp1 should be fixed now.  i put extraTemp1, extraTemp2, and extraTemp3 into a single additional plot, and added them to the hilo and current tables.  newskin is in sync with development.

i think we agreed to give it a new name so there is no confusion about Standard vs "new Standard".  current candidate is 'Seasons' - any second thoughts or better names?

how do we deploy this?  anyone who updates weewx will not get the new skin, since upgrading using setup.py does not modify existing skins.  if someone wants to use Seasons, s/he must copy the skins/Seasons directory from the weewx distribution tarball.  any new installations will get the Seasons skin as the default.

deb and rpm upgrades will get Seasons, Mobile, and Smartphone alongside existing Standard (whether or not it has been modified).  Seasons is enabled, Mobile and Smartphone are disabled.  units are in [default_options] stanza.  existing [Standard] is untouched.  there is a conflict in output directory between Seasons and Standard, as they both default to /var/www/html/weewx.  (i have tested this on debian8)

remember that Smartphone and Mobile have been pulled out of the Standard skin into their own, standalone skins.

we will also have to remove all of the stuff in the customization guide that refers to Standard skin and make it refer to Seasons instead...

gjr80, 2017-04-30:
At the risk of creating a duplicate, could 'Seasons' be provided as an extension and existing users just install the 'Seasons' extension?

tkeffer, 2017-04-30:
I don't see any reason why we can't have `setup.py` check to see if the user already has Seasons in the `skins` subdirectory and, if not, copy it over. Then, if the user wants to activate it, it's a simple matter of adding the appropriate stanza to `[StdReports]`.

This would honor our promise not to modify existing skins.

Not sure if this would work with DEB and other distributions.

As for the conflict between outputs, users already manage to deal with this with existing 3rd party skins. If we really think it's going to be an issue, we could have Seasons put its images in a separate `images` subdirectory, and its index page to `default.html`. However, I think this would create more issues than it solves.

Another option would be to hardwire in a check in the report generator. A kludge, but it would work.

tkeffer, 2017-04-30:
One issue: when I select a month or year for the NOAA reports, it says...

```
Loading NOAA/NOAA-2011.txt
```

but then nothing happens.

tkeffer, 2017-04-30:
Opening up the JavaScript console in Chrome reveals the reason why:

```
seasons.js:122 XMLHttpRequest cannot load file:///home/weewx/public_html/NOAA/NOAA-2016-02.txt. Cross origin requests are only supported for protocol schemes: http, data, chrome, chrome-extension, https.
```﻿
So, things should work fine under an http connection.

matthewwall, 2017-05-02:
i don't yet have a good answer for the cross origin request problem.  it is only an issue on chrome, and only when you are browsing use file://.  it might be an issue with most recent versions of firefox as well.  there is a workaround for chrome - when you launch chrome, specify --allow-file-access-from-files.  or install a web server and browse using http/https.

options include:

1) iframe, with javascript that changes the src.  unfortunately, iframe does not resize properly, although there are ways to do it using jquery, and probably ways to do it using raw javascript.

2) full set of html pages with NOAA tables.  this would results in a bunch of html files instead of a bunch of text files.  

3) use XMLHttpRequest to embed text into a single html div.  this is how newskin currently does it.

4) raw text with no navigation elements.  this is how the old standard skin does it.

tkeffer, 2017-05-02:
It's not a problem at all.

I've got Seasons up and running on my own [weather station](http://www.threefools.org/weewx/).

My wife's comment was, "busy."

matthewwall, 2017-05-02:
so she was too busy to take a look? ;)

indeed, Seasons is rather information-dense, but hopefully the styling and layout keep it from being too overwhelming.

i'm tempted to release a 'lite' skin with less content.  the problem is that there are two competing use cases.  the forecast use case ("what will the weather be later today or later this week?") tends to be where a 'lite' skin is desirable, i.e., only the current info is necessary.  the 'historical' use case ("what has the weather been doing the past day/week/month", or "how did last month compare to the same month of last year") requires more information.

the Smartphone skin is basically a 'lite' skin, but it might be nice to make a version of that which is less iOS-from-2007-ish.

thanks for running it on a site with real data - sinusoids get a bit monotonous after awhile :)

gjr80, 2017-05-11:
I guess I am a bit late on this, most of my time lately has been taken up getting my [Odroid XU-4 Cloudshell2](http://www.hardkernel.com/main/products/prdt_info.php?g_code=G149142327280) setup and then moving my web server 'stuff' over to it. I installed Seasons on my weeWX machine this morning and all seems good - generates about 6 times faster than my modified Standard skin. As part of my server changeover I have been cutting everything across to https, I think things are working from 'outside' and you should be able to see my Seasons page [here](https://weather.therodericks.id.au/seasons).

matthewwall, 2017-05-12:
gary, i just get an empty page when i try to view your seasons page.  btw, i am on a boat until mid june, with only occasional internet.  tom is just starting the oregon offshore race.  thus the lull in weewx activity :)

gjr80, 2017-05-12:
>gary, i just get an empty page when i try to view your seasons page.

Interesting. I just used my iPad on mobile internet to pull up this issue the link above and it worked fine. I just looked at my server log and saw a few 403 errors from an IP that when I look it up come from the US Virgin Islands? Using a Mac? That was about 12 May 01:05 GMT (1 minutes ago). Someone else was there around 11 May 13:08 GMT from a US based IP, and it seems to have worked then. Might get my folks to see if they can see it as a double check. You haven't got some ddogy setup have you? :)

> btw, i am on a boat until mid june, with only occasional internet. tom is just starting the oregon offshore race. thus the lull in weewx activity :)

Ah, that explains it, thought it must have been something I said.

andylittle, 2017-05-12:
I see the page just fine on my Android phone.

andylittle, 2017-05-12:
Up and running [here](https://www.andylittle.us/weewx/)

tkeffer, 2017-05-21:
One suggestion: leave out the "max" lines for radiation and UV for day plots. They aggregate over the last hour, making for some strange looking plots:

[image]

The week, month, and year plots look as expected.

rbeerster, 2017-05-30:
Up and running [here](http://weather.onu.edu).  Note, the "Change" under Celestial and let me know if you want the diffs.

tkeffer, 2017-05-31:
Commit https://github.com/weewx/weewx/commit/d662b20f6ebd2081f22fd2cf34fdec438b8f01e0 has something similar, although you have to click on the "Celestial" link to see it.

Suggestion: put a unary sign (`+` or `-`) in front of the change to reinforce that this is a delta.

rbeerster, 2017-05-31:
Tom, I like your code for this.

tkeffer, 2017-06-06:
One little alignment issue. This on an iPad, using Safari, although I'm seeing it on Chrome on a MacBook Air as well.

[image]

gjr80, 2017-06-06:
> One little alignment issue. This on an iPad, using Safari, although I'm seeing it on Chrome on a MacBook Air as well.

Ditto Chrome on an iPad.

tsoodla, 2017-08-03:
Fix of the alignment issue: [8bb2c97](https://github.com/weewx/weewx/pull/265/commits/8bb2c9750e89abe07b374f8f12dc269be7c523fb)

matthewwall, 2017-08-08:
what was the intent of commit 007622495dd931d06f80d540f4eda75ec8e585e0 (the br after the year select)?

gjr80, 2017-08-08:
> what was the intent of commit 0076224 (the br after the year select)?

For me it seem to have correctly right aligned the - Select Month - and - Select Year - drop downs on my iPad (in landscape - not sure if the issue exited in portrait). Doesn't seem to have made any difference to my PC, it was already aligned correctly.

tsoodla, 2017-08-08:
[image]
There is an alignment issue on mac (OS X 10.11.6) browsers.
Safari 10.1.2
Chrome 60.0.3112.90

tkeffer, 2018-10-07:
I've included a "rain year." Matthew may not like the formatting, but it's the best I could come up with.

I also included a reference to the `Standard` skin in `weewx.conf`, but left it unenabled by default.

Still to do:
* `Seasons` uses the old tag formatting scheme (in particular, it uses `.formatted`). These should be
changed to `.format()`.
* The install script should install `Seasons` into the `skins` directory if its not already there. Same with `Mobile` and `Smartphone`. Otherwise, it doesn't touch anything.
* Documentation needs to be updated.
  - Change references to `Standard` skin to `Seasons`.
  - Mention the `Mobile` and `Smartphone` skins.

Incidentally, what the difference between `Mobile` and `Smartphone`?

matthewwall, 2018-10-08:
the extension installer is really dumb - it simply iterates through each tuple defined by the dict in the extension's install.py to create each file and directory for the extension.  it does not know anything about enable/disable.  it will only install into 'bin' (for bin/user) or 'skins'.

if the extension developer included the files for a 'skins/Standard', then the extension installer would overwrite the original 'skins/Standard'.  (it does make a backup copy of the configuration file, but i don't think that it backs up the skins or bin/user directories, or anything in them that it modifies)

tkeffer, 2018-10-08:
I was referring to the package installers (deb, rpm, etc.)

tkeffer, 2018-10-08:
With commit https://github.com/weewx/weewx/commit/ef31de7fd7e414dd3342f024595f5033e3eb8c81, `setup.py` now installs only any missing skins

matthewwall, 2018-10-10:
regarding the use of a 'defaults' stanza for StdReport (part of the newskin implementation) that will have parameters that apply to all reports.  this will let users set metric units (and other parameters) for all reports, with ability to override those defaults for individual reports.

i guess i would formalize the override sequence like this:

```
parameters for a skin 'SKIN_NAME' are determined in the following order:

1 - hard-coded defaults
2 - override with values from skin.conf
3 - override with values from [StdReport] in weewx.conf
    3a - any scalars from [StdReport] (for backward compatibility)
    3b - any scalars or non-scalars from [StdReport][[defaults]]
4 - override with values from [StdReport][[SKIN_NAME]] in weewx.conf

if a value (scalar or non-scalar) in 3 is not recognized by a skin, it is silently ignored.
```

## how do we do this?

### option 1: literal mapping
```
[StdReport]
    [[defaults]]
        SKIN_ROOT = skins
        HTML_ROOT = public_html
        data_binding = wx_binding
        [[[Units]]]
            [[[[Groups]]]]
                group_altitude = foot
                group_degree_day = degree_F_day
                group_pressure = inHg
                group_rain = inch
                group_rainrate = inch_per_hour
                group_speed = mile_per_hour
                group_speed2 = mile_per_hour2
                group_temperature = degree_F
    [[Standard]]
        ...
    [[FTP]]
        ...
```

### option 2: messy mapping (only non-scalars in [[defaults]])
```
[StdReport]
    SKIN_ROOT = skins
    HTML_ROOT = public_html
    data_binding = wx_binding
    [[defaults]]
        [[[Units]]]
            [[[[Groups]]]]
                group_altitude = foot
                group_degree_day = degree_F_day
                group_pressure = inHg
                group_rain = inch
                group_rainrate = inch_per_hour
                group_speed = mile_per_hour
                group_speed2 = mile_per_hour2
                group_temperature = degree_F
    [[Standard]]
        ...
    [[FTP]]
        ...
```

i'm leaning toward option 1, with recognition of scalars in [StdReport] for backward compatibility.  any scalars in [StdReport] would be overridden by scalars in [[defaults]].

separate but related: there is overlap with [StdRESTful] inasmuch as the log_success and log_failure should also be options in [StdReport], applying to any generator or service that gets its settings from [StdReport] tree.

i know it is not part of this problem, but we probably want consistent behavior between [StdReport] and [StdRESTful].  for [StdRESTful] we have a few root-level parameters, all of which are scalars:

```
[StdRESTful]
    log_success = (True|False)
    log_failure = (True|False)
    post_interval = None
    max_backlog = sys.maxint
    state = None
    timeout = 10
    max_tries = 3
    retry_wait = 5
    retry_login = 3600
```
i think that these are all handled properly right now.

i can imagine having defaults apply to each uploader, too.  for example, there are the observation-to-uploadname mappings, the units and unit systems, and the per-observation formatting.  but although there are a few uploaders that share patterns, most are so different that propagating options from the root is probably not necessary.

tkeffer, 2018-10-10:
A problem with both options is that they are a mix of defaults and overrides. For example, the semantics of an option like `log_success` in `[StdRESTful]` is that it is a default. It can be overridden by individual uploaders. By contrast, the semantics of `[[defaults]]` is that it unilaterally overrides whatever value an individual skin might provide.

At the very least, we should call it something else, like `[[overrides]]`, but I still don't like this mix of semantics.

A third option is to go back to what we had. It still has a mix of overrides and default semantics, but at least it's one level less deep.

tkeffer, 2018-10-10:
Another option. It has some big negatives, but I'll just throw it out there.

### option 3: Pure default semantics
```ini
[StdReport]
    [[defaults]]
        SKIN_ROOT = skins
        HTML_ROOT = public_html
        data_binding = wx_binding
        [[[Units]]]
            [[[[Groups]]]]
                group_altitude = foot
                group_degree_day = degree_F_day
                group_pressure = inHg
                group_rain = inch
                group_rainrate = inch_per_hour
                group_speed = mile_per_hour
                group_speed2 = mile_per_hour2
                group_temperature = degree_F
    [[Standard]]
        ...
    [[Mobile]]
        HTML_ROOT = public_html/mobile
        ...
```

This has the same structure as option 1, but different semantics. All options truly are defaults. They can be overridden by individual skins. 

However, individual skins would typically not have a `[Units][[Groups]]` section. They would inherit options from farther up the hierarchy. This allows a single unit system to be specified for all skins.

This would break backwards compatibility, but it has simple semantics.

matthewwall, 2018-10-10:
> By contrast, the semantics of [[defaults]] is that it unilaterally overrides whatever
> value an individual skin might provide.

just to be clear: items in the [[defaults]] section would override anything in skin.conf, but anything in [StdReport][[Standard]] would override anything in [[defaults]]

there is still a hierarchic precedence of settings

the only differences between [StdReport] and [StdRESTful] are:

1 StdReport needs to have non-scalar items in its config defaults

2 the reports have a skin.conf file in addition to weewx.conf, but StdRESTful services do not

(i may not be using 'override' and 'default' in the same way that you are...)

matthewwall, 2018-10-10:
take a use case from the docs - make a single report render as two different unit systems.  this would still happen as before:

```
# explicitly
[StdRESTful]
    [[StandardImperial]]
        HTML_ROOT = public_html/imperial
        ...
        [[[Units]]]
            [[[[Groups]]]]
                group_altitude = foot
                group_degree_day = degree_F_day
                group_pressure = inHg
    [[StandardMetric]]
        HTML_ROOT = public_html/metric
        ...
        [[[Units]]]
            [[[[Groups]]]]
                group_altitude = meter
                group_degree_day = degree_C_day
                group_pressure = mbar
```

```
# implicitly with defaults
[StdRESTful]
    [[defaults]]
        [[[Units]]]
            [[[[Groups]]]]
                group_altitude = foot
                group_degree_day = degree_F_day
                group_pressure = inHg
    [[StandardImperial]]
        HTML_ROOT = public_html/imperial
        ...
    [[StandardMetric]]
        HTML_ROOT = public_html/metric
        ...
        [[[Units]]]
            [[[[Groups]]]]
                group_altitude = meter
                group_degree_day = degree_C_day
                group_pressure = mbar
```

```
# implicitly with no defaults (imperial defaults to whatever is in skin.conf)
[StdRESTful]
    [[StandardImperial]]
        HTML_ROOT = public_html/imperial
        ...
    [[StandardMetric]]
        HTML_ROOT = public_html/metric
        ...
        [[[Units]]]
            [[[[Groups]]]]
                group_altitude = meter
                group_degree_day = degree_C_day
                group_pressure = mbar
```

matthewwall, 2018-10-10:
> By contrast, the semantics of [[defaults]] is that it unilaterally overrides whatever
> value an individual skin might provide.

i think i understand our difference in view now.  i am assuming that anything in [[defaults]] will take precedence over anything in skin.conf, but anything in [StdReport][[REPORT]] will take precedence over anything in [[defaults]].  this is so that a user can keep his/her customizations to a skin by putting those customizations in weewx.conf.  although we have a policy of not modifying the skins directory, i think the extension installer violates that an overwrites skin.conf instead of datestamping it like the weewx upgrade process does to weewx.conf.

tkeffer, 2018-10-13:
Thinking about this... the problem with option 3 is that it breaks backwards compatibility. However, it would not be hard to cobble together an update program that would fix the problem. 

The compatibility problem arises when a user has overridden a `skin.conf` spec in `weewx.conf`. Indeed, this is something the installer regularly does. To fix, we have to comment out any specs in the `skin.conf` file that were overridden in `weewx.conf`. This allows the defaults in `weewx.conf` to take hold. We do this for all skins that have an override clause.

This should work well *except* for one problem: 3rd party skins and their installers. If any of them rely on overrides in `weewx.conf`, they would no longer work. One option is that the installer could look for these cases and patch them.

Overall, I feel like we should wait on this. If we are going to break backwards compatibility with WeeWX v4, that would be the time to change skin option semantics. Do all the breaks at once.

matthewwall, 2018-10-13:
are you saying that options 1 and 3 have identical syntax, but differ in how parameters are set, like this:
```
option 1:
1. start with hard-coded defaults
2. override with anything from skin.conf
3. override with anything from [StdReport][[defaults]]
4. override with anything from [StdReport][[REPORT]]

option 3:
1. start with hard-coded defaults
2. override with anything from [StdReport][[defaults]]
3. override with anything from [StdReport][[REPORT]]
4. override with anything from skin.conf
```

matthewwall, 2018-10-13:
the difference between StdReport and StdRESTful is skin.conf.  StdReport has it, but StdRESTful does not.  i thought that weewx has always used option 1 for StdReport, but without a [[defaults]] section; the default values have always come from skin.conf, then get overridden by parameters in weewx.conf, if they exist.  so option 1 is backward compatible - it is an extension of the existing sequence of overrides.  am i missing something?

matthewwall, 2018-10-13:
i guess i resolved the scrambled semantics by having skin.conf be the decision tree.  skin.conf has always given me a bit of a rash - i never understood why one *must* have a skin.conf, even if the skin.conf is trivial (like FTP or rsync) and only defines things that no one would ever modify.  seems like the degenerate case of no skin.conf should simply use the skin's default values.  i guess that is why i picture the skin.conf as the root of the tree, not the leaves.

tkeffer, 2018-10-16:
I think the confusion is precisely my point --- we have two different tree hierarchies running in opposite directions.

For a given skin, `skin.conf` is acting like something near the root, in the sense that it is getting overridden by something else (a 'leaf', if you will) in `weewx.conf`. But, in the configuration file hierarchy, it's a leaf in the sense that its actions apply only to the skin it's attached to, not to other skins.

My option 3 is to bring these two views into concordance. Options in `[[defaults]]` would apply to *all* skins and are therefore near a root. They could get overridden all the time by things out towards the leaves, that is, options in `[StdReport][[Report]]` and, finally, the individual `skin.conf` files.

Your desire to not require a `skin.conf` would get satisfied by hardwiring defaults that apply to all skins. These hardwired defaults wouldn't appear in `weewx.conf` (although, I suppose they could), but can always get overridden.

Taking a specific example,  the FTP generator, the hardwired defaults would look like (I'm using ConfigObj notation here, but it would actually be Python code):

```ini
  [Generators]
        generator_list = weewx.cheetahgenerator.CheetahGenerator, weewx.imagegenerator.ImageGenerator, weewx.reportengine.CopyGenerator
```

while the FTP section would look like

```ini
[StdReport]
  [[FTP]]
    [[[Generators]]]
      generator_list = weewx.reportengine.FtpGenerator
```

There would be no `Ftp` subdirectory under directory `skins`.

tkeffer, 2018-10-21:
I have reverted the `[[default_options]]` code and docs. https://github.com/weewx/weewx/commit/adb848568f4d76c8c61255587a81020ab6dc7518

tkeffer, 2018-11-03:
First cut at moving defaults into `weewx.conf` in branch [`big_default`](https://github.com/weewx/weewx/tree/big_default).

I actually went one step farther and moved the defaults into their own top-level section in `weewx.conf`. 
I am thinking that things like `log_success`, `log_failure`, and `socket_timeout` could go in there. For now, it looks something like this:

```ini
##############################################################################

#   Default values for skins

[Defaults]

    # The following section is for managing the selection and formatting of units.
    [[Units]]

        # The following section sets what unit to use for each unit group.
        # NB: The unit is always in the singular. I.e., 'mile_per_hour',
        # NOT 'miles_per_hour'
        [[[Groups]]]
            group_altitude     = foot                 # Options are 'foot' or 'meter'
            group_degree_day   = degree_F_day         # Options are 'degree_F_day' or 'degree_C_day'
            group_direction    = degree_compass
            group_moisture     = centibar
            group_percent      = percent
            group_pressure     = inHg                 # Options are 'inHg', 'mmHg', 'mbar', or 'hPa'
            ... etc
```

tkeffer, 2018-11-25:
I've written a patch procedure that:
1. Adds a `[Defaults]` section to an old `weewx.conf`.
2. If there is an "override" in the reports section, it comments out the corresponding scalar in `skin.conf`. This allows the "override" to work with the new semantics. I put "override" in quotes, because it is no longer an override. It's just the default value for the skin, which takes affect unless there is an explicit entry in `skin.conf`.  Hence, the need to comment out the scalar in `skin.conf`.
3. If there is a value in `skin.conf` that is identical to a value in the `[Defaults]` section, it comments out the latter. This allows one to change a value in `[Defaults]` and have it take affect.
4. It adds a version number to the `skin.conf`, signalling that it assumes the new semantics.
5. It backs up the old `skin.conf` with a timestamp.

Net, net, if an `weewx.conf` starts out looking like this:
```ini
        ...
        [[[Units]]]
            [[[[Groups]]]]
                group_altitude = meter
                group_speed2 = meter_per_second2
                group_pressure = mbar
                group_rain = mm
                group_rainrate = mm_per_hour
                group_temperature = degree_C
                group_degree_day = degree_C_day
                group_speed = meter_per_second
```

It will end up looking like

```ini
        ...
        (as before)
        [[[Units]]]
            [[[[Groups]]]]
                group_altitude = meter
                group_speed2 = meter_per_second2
                group_pressure = mbar
                group_rain = mm
                group_rainrate = mm_per_hour
                group_temperature = degree_C
                group_degree_day = degree_C_day
                group_speed = meter_per_second
...
#   Default values for skins

[Defaults]
    
    # The following section is for managing the selection and formatting of units.
    [[Units]]
        
        # The following section sets what unit to use for each unit group.
        # NB: The unit is always in the singular. I.e., 'mile_per_hour',
        # NOT 'miles_per_hour'
        [[[Groups]]]
            group_altitude = foot    # Options are 'foot' or 'meter'
            group_degree_day = degree_F_day    # Options are 'degree_F_day' or 'degree_C_day`
            ...
        ...
    ...
```    

Meanwhile, `Standard/skin.conf`, which started out looking like this:
```ini
[Units]
    # This section is for managing the selection and formatting of units.
    
    [[Groups]]
        # For each group of measurements, this section sets what units to
        # use for it.
        # NB: The unit is always in the singular. I.e., 'mile_per_hour',
        # NOT 'miles_per_hour'

        group_altitude     = foot                 # Options are 'foot' or 'meter'
        group_degree_day   = degree_F_day         # Options are 'degree_F_day' or 'degree_C_day'
        group_direction    = degree_compass
        group_moisture     = centibar
        ...
```

will end up looking like:
```ini
skin_version = 2
[Units]
    # This section is for managing the selection and formatting of units.
    
    [[Groups]]
    
    [[StringFormats]]
    
    [[Labels]]
    
    [[TimeFormats]]
    
    [[Ordinates]]
    ...
```
As you can see, a limitation on the patching procedure is that sections in the `skin.conf` may end up with no comments (as there is no scalar to attach them to). I'll see if I can work something out to avoid that.

On the TODO list:
- Figure out some way to preserve comments when upgrading a skin.conf that has nothing
 left in a section.
 - Extension installer needs to patch old skins on the fly.
 - When setting a unit system, installer should now modify `[Defaults]` section rather than doing overrides in individual reports.

matthewwall, 2018-11-26:
all of this patching makes me anxious.

1) what is the intent of the 'skin_version' in skin.conf?  is that intended to be the version of the skin.conf instance, so skin_version might be 1.0, 1.1, 3.0, etc?  or is it the version of how skins are processed, i.e., somehow related to setup.py?  i used to put a version in the skin.conf of skins that i wrote, but then i kinda gave up on that, since people modify the skin.conf a lot, so any version tag quickly becomes meaningless.

2) why not just eliminate any empty stanza when patching?  if there are no scalars (or non-scalars) left inside a stanza, then delete the stanza.

3) this means we will modify stuff in the skins directory.  are you sure we want to go there?  we could leave existing systems as-is - do not insert `[Defaults]`, do not modify any 'overrides', and do not modify any skin.conf files.  if someone wants to take advantage of the [Defaults], s/he could modify his/her files.  unfortunately, if s/he makes no changes, things would behave as before only if s/he had no 'overrides' in weewx.conf.  rats!

just to confirm, this is the order in which things would be applied:

1) start with hard-coded values
2) update with anything specified in `[Defaults]` from weewx.conf
3) update with anything specified in a specific section, e.g., `[StdReport][[foobar]]` from weewx.conf
4) update with anything specified in skin.conf

if i have learned one thing from all of this, it is to get the semantics and processing of your configuration files right before you do anything else :)

tkeffer, 2018-11-26:
Makes me anxious too. At this point, I'm still experimenting to see what it would look like --- I'm not 100% sure I want to go there. To answer your questions:

1. The version number versions the semantics of defaults, not the skin itself. Maybe I should change the name to `skin_semantics` or something like that.
2. That could be done. Ideally, I'd prefer commented out options so a user can easily figure them out.
3. I went through the same thought process. Another option is to just introduce the new semantics and let things fall where they will. That means a lot of support questions like, "Hey! My units are wrong! What happened?" and then let the user change things, possibly with the support of a tool like `wee_config`.

Your application order is correct, except there are no hardcoded values. The starting point for everything is `[Defaults]`. 

What motivated this in the first place is the configuration process. With the old method, with the introduction of the new skins, it's necessary to do an override for each one of them. Then, if the user wants to change them later, the config program is now looking at possibly evaluating and changing up to four different overrides. With the new system, everything is in `[Defaults]`.

Agree  with your conclusion about getting things right from the start! We mostly did ... this is the exception. These changes would bring skin configuration into alignment with the rest of the program.

tkeffer, 2018-11-28:
I've added a routine to patch skin extensions to V3.9 semantics on the fly. I've successfully installed the `amphibian` extension and it acts as expected. Except for some documentation changes, I've pretty much completed branch `big_default`. 

Personally, I like it. Choosing between metric and US is super simple: you just modify `[Defaults]`. No overrides necessary (unless you want a mix of US and metric).

Please take a few minutes and try it out and let me know what you think. My inclination is to go ahead with this.

tkeffer, 2018-12-05:
Hearing no objections (well, more precisely, hearing nothing at all), I've merged `big_default` into `development`.

The hardest part was refactoring the Customizing Guide to accommodate the new semantics. It's still not completely done.

matthewwall, 2019-01-01:
finally digesting the changes you made.  i'm finding the description in 'upgrading' a bit hard to follow.  it is clearly written, and the example helps.  but the behaviors are complicated.  i think i will have to make a diagram to illustrate the behaviors before vs after.

the label 'ObservationDefaults' smells funny - 'ReportDefaults' might be more appropriate?  except this does not just apply to reports, does it?  or maybe we should introduce that pattern - ReportDefaults apply to StdReport, RESTfulDefaults apply to StdRESTful, etc.  (don't want to over-think this - just looking at it from a noob and non-english pov)

matthewwall, 2019-01-01:
in v3.9, what is the recommended way to customize a skin?

use cases:
* some people prefer to put all of the changes in weewx.conf instead of modifying many different skin.conf files
* when a skin extension is upgraded, it overwrites skin.conf, even if user has changes

caveats:
* this highlights a fundamental problem with the pre-3.9 model: when you upgrade an skin extension, it overwrites the skin.conf.  but weewx promises to never modify anything in skin.conf.
* for weewx post-3.9, authors of skin extensions should never specify time formats, units, in the skin.conf since that would have precedence over anything in weewx.conf

matthewwall, 2019-01-01:
i think there is a bug in the customizing guide (and in the code?) regarding: [Units][[Generic]]

i would expect this hierarchy:
```
[Defaults]
    [[Units]]
        [[[Groups]]
            group_altitude = foot
        [[[Labels]]]
            inch     = ' in'
        [[[StringFormat]]]
            inch     = %.2f
        [[[TimeFormats]]]
            day        = %X
        [[[Ordinates]]]
            directions = N, NNE, NE, ENE, E, ESE, SE, SSE, S, SSW, SW, WSW, W, WNW, NW, NNW, N
        [[[DegreeDays]]]
            heating_base = 65.0, degree_F
        [[[Trend]]]
            time_delta = 10800
    [[Labels]]
        inTemp = Inside Temperature
    [[Almanac]]
    [[Extras]]
```
notice that there is no need for a `Generic` subsection in `Labels`.

the `[[Labels]]` section can also be used for i18n - labels can be used by the skin author not only for observations, but also for any arbitrary text that might show up in the skin.  (of course complicated templates should use full i18n, probably by having a separate .tmpl file for each language.  but for number-oriented skins the generic use of `[[Labels]]` will save a lot of duplication)

matthewwall, 2019-01-01:
btw, i find too many cross-links in the documentation to be rather distracting - i would try to eliminate them whenever possible.  the narrative style of writing is very welcoming, but you cannot assume that everyone will want to follow the same narrative path.

matthewwall, 2019-01-01:
i assumed that 'skin_semantics=2' means "compliant with 3.9 behavior", but that is not really true, is it?  is this correct:

* if a skin.conf contains 'skin_semantics=2' then the contents of that skin.conf will override anything already specified.

* if a skin.conf does not contain 'skin_semantics' then the contents of that skin.conf will be applied first, overridden by anything in weewx.conf

in that case, 'skin_semantics=2' is a mechanism for a skin author to say "these are critical for the skin to operate" versus "these are suggested for the skin"

it sounds like we have added a new option (skin_semantics) whose value changes the way skin.conf contents are applied.  if so, we need to document it.

tkeffer, 2019-01-05:
Answering some of your questions:

1. I don't like the name `ObservationDefaults` either. I also don't like the pattern of having separate defaults sections, named after their operative section (*e.g.*, `[ReportDefaults]`) --- if we were to do that, you might as well just use a subsection `[[Defaults]]` under the appropriate top-level section.

    I was hoping `[ObservationDefaults]` would be useful for more than just `StdReport`. Maybe we should just call it a simple `[Defaults]`. 

2. No reason you can't include an `[[Extras]]` section under `[Defaults]`. If you can't, it's a bug.

3. Not sure what you mean by "when a skin extension is upgraded..." Do you mean that literally? That is, when the extension itself is upgraded? Or, do you mean when my upgrade procedure is applied to `skin.conf`? If the former, that is no different from before --- we have no mechanism to upgrade a skin. If the latter, the upgrade process proceeds only if the option `skin_semantics` is omitted, so it's a one-time thing. The process is also designed to warp the options to match the new semantics. It doesn't actually change any options. Just moves them, if necessary, to `weewx.conf`.

4. In general, you're right: authors of skin extensions will not want to specify generic options in `skin.conf`. That would break the whole point of the new skin semantics: to allow global options to be specified. Nevertheless, there may be occasions when you want to do so. Take a look in Season's `skin.conf` and you'll find one such instance. It overrides the `[Labels]` section because it includes some of the text in the template itself.

5. (Bug in Customizing guide). I think you're right. I've changed it in commit https://github.com/weewx/weewx/commit/4360ac61db39a35206dbebf0f58b3b395eedcd9e.

6. I thought about having the reporting system use whatever semantics the `skin.conf` specified, but rejected it as too hard to debug and support. So, all `skin_semantics` *really* means is that it's been patched to the V3.9 semantics, so there's no reason to do it again. The actual reporting code makes no use of it.

7. I like your organization for the Customizing guide. I'm not working at it now, if you want to take a crack at it.

8. I was sorely tempted to change section names and the ordering, but resisted because the upgrade process would be much more complicated. Right now, the worst that can happen is that something will appear with the wrong units. But, if we reorder things, a report could get totally garbled, or even unrunnable. The sections are not terribly consistent, but they work, and they've been unchanged for nearly 10 years!

Now my turn: is this a good idea? I hate the idea of breaking that long, long string of upwards compatibility, but I didn't see any other way of easily supporting multiple skins.

tkeffer, 2019-01-06:
Changed from `[ObservationDefaults]` to a simple `[Defaults]`.

I've changed the installer so it no longer attempts to patch incoming skins. This means, of course, that the units used in the new, installed skin will not reflect the choices made in `[Defaults]`.

That does mean there's an asymmetry between skins that were "upgraded" (which will have their `skin.conf` modified), and those that were installed (no modifications). That doesn't feel right.

I guess we could go all the way and not modify upgraded `skin.conf` files at all. Make the user do it.

matthewwall, 2019-01-06:
first draft of refactored customization guide is in at commit a8ae4e15

* i think we should eliminate the need for skin.conf, especially for the FTP and RSYNC cases
* need to add section to extension writers that enumerates best practice, i.e., do not specify anything in skin.conf unless necessary, explanation of what happens to skin.conf and template files when an extension is installed (everything is overwritten), what else?
* the bracket nesting is a mess.  see the example in 'how options work'.  somehow we ended up with the worst case (3 different layerings for a single specification)
* there is now no way to override imagegenerator or cheetahgenerator options in weewx.conf.  i do this a lot at some of my sites, for example in exfoliation - it makes it easier to integrate multiple extensions.  now this is broken.  we should discuss.

tkeffer, 2019-01-06:
> the bracket nesting is a mess.

If we want to cut down the number of layers, I see only three options:
1. Have no `skin.conf` at all. Everything is specified in `weewx.conf`. PRO: simple semantics. CON: we are too far down the path to go back; makes `weewx.conf` enormous; makes it harder for skin writers to specify customizations.
2. Do not allow `[StdReport]` overrides. PRO: simple semantics. CON: not possible to use the same skin for multiple unit system (the example used in the Customizing Guide). 
3.  Everything must be specified in `skin.conf`. No overrides, no `[Defaults]` section. PRO: very simple semantics; cut-and-paste becomes easy. CON: limited flexibility and reuse of skins; not possible to use the same skin for multiple unit systems, at least without a lot of cutting and pasting.

My conclusion: what we have is the best of a  bunch of so-so options.

> there is now no way to override imagegenerator or cheetahgenerator options in weewx.conf.

If I'm understanding you correctly, you want to specify which images and templates get processed by overriding in `[StdReport]`. If that's the case, it should be possible, although I have to admit, I haven't tried it.

> using skin.conf to define the defaults instead the leaves

Not following your thinking.

Epic issue thread we've got going here...

[5 more replies]

closed 2019-01-31 by tkeffer