weewx/weewx #249: Rain accumulation inaccurate from Netatmo

burtamcgna, 2017-03-30:
The Netatmo weather station and my.netatmo.com site is reporting accumulation of 0.033 and when I use Meteoware as the service to push to Wunderground or PWSWeather, the results are reported accurately and things are fine.

If I disable Meteoware and use Weewx, all of the accumulation and rain totals are losing their precision or simply not being calculated properly. Weewx is reporting 3.33 accumulation.  I ran a test and can see clearly that Weewx is not measing as hundredths, only tensths.

.001 reported by the station and Netatmo results in Weewx reporting .01

matthewwall, 2017-03-30:
what are the units of 0.33 from Meteoware (what is Meteoware?)

what are the units of 3.33 reported by weewx?

what are the units of 0.001 'reported by the station'?

what are the units of 0.01 reported by weewx?

burtamcgna, 2017-03-30:
All units are "inches" or US based measurements

Meteoware is a publishing service similar to Weewx that uploads data to Wunderground and PWSWeather and a couple of other sites.  It is publishing the data from Netatmo (my.netatmo.com) and reflects exactly what Netatmo is reporting.  I am sorta of new to this PWS stuff, so not sure about all the terminology yet :)

matthewwall, 2017-03-30:
if the discrepancy is exactly a factor of 10, then i'm guessing that the netatmo docs are wrong.

netatmo says that the netatmo api reports rainfall in mm.  so when weewx-netatmo gets a value x from netatmo, it puts a value of 0.1 * x into the packet that it yields to weewx, with unit system METRIC (METRIC expects cm for rain).

but you are saying that the values in weewx are 10 times too big.

that would imply that netatmo is reporting 10ths of mm not mm.

if you do one bucket tip, what does the netatmo web site report, in metric units?

(i do not have a netatmo rain gauge, so i have no way of testing any of this)

burtamcgna, 2017-03-30:
I will rerun a test on this later this afternoon and let you know.  That’s what it appears to be to me as well.  When I first looked at the WU site this morning around 7 it showed we had received 2.5 inches of rain at my house and I was looking for my raft ☺

matthewwall, 2017-03-30:
i posted netatmo-0.10rc1 on the weewx user group.  please give that a try.

closed 2017-04-17 by matthewwall

MichaelSulzer, 2019-08-17:
Hello, why isn't the version mentioned in your post on github available? I just wondered why the accumulated rainfall is so much higher in weewx.

Updating the github would be very helpful. I had to search the user group to find the mentioned version. 

Thank you very much in advance!