weewx/weewx #250: `StdWXCalculate` calculates `windrun` as a 'day' value rather than a 'per period' value

gjr80, 2017-03-31:
`StdWXCalculate` can optionally (it is disabled by default) augment each archive record with a calculated `windrun` field. At present the calculated value is the cumulative `windrun` for the day. This means that most `windrun` aggregates (based on either archive or daily summaries) are meaningless eg `$day.windrun.sum` would be expected to give the total `windrun` for the day so far, but it currently returns the sum of the day `windrun` so far for each of today's archive records. `$hour.windrun.sum` would be expected to give the total `windrun` seen so far in the current hour, but it results in the sum of the day `windrun` so far for each of the archive records in the current hour.

`windrun` should operate in a similar way to `rain` where the value is the 'amount since the last value' rather than being day cumulative.

`windrun` is currently only provided for archive records, consideration should be given to including windrun in loop packets; however, this will present additional challenges for stations that emit partial packets.

tkeffer, 2017-11-16:
Another question: is windrun the vector or scalar sum? That is, if the wind blows to the east at 10 knots for an hour, then to the west for an hour, is the windrun zero (vector sum) or 20 nm (scalar)?

matthewwall, 2017-11-16:
pretty sure it is scalar sum.  it gives you an idea of how windy a place is, independent of direction.  i think of it as analogous to area under a curve.

nellyt, 2017-11-20:
My reference book "The Wx Observers Handbook" by Stephen Burt says .. wind run  is a measure of mean wind speed expressed in terms of "distance covered over a given period" .....
so yes its a scalar sum AFAIK

PR #399 (2019-04-18): fixed windrun calculation to be for current period only
I've made some changes to wxservices.py so that calc_windrun only calculates the run for the current archive period. This I believe fixes  #250. The method is somewhat simpler now as it doesn't have to query the db to find out previous windrun values, which I think is the new intent.

Being new to this whole process I'm not quite sure how to test it works. I flicked the code onto my test system and it didn't throw any errors, however, I can see that windrun is not actually in my db schema and I'm not actually sure how to get it there?

Co [... cut]
closed 2019-04-18 by tkeffer