weewx/weewx #257: FineOffset USB Lockup: a possible way to explore

thunderdome2, 2017-05-28:
Hello.

I'm using a FineOffset WH3080 connected on a Raspberry Pi 3 and WeeWX installed on it to upload data on wunderground.com. Since my station is located on a small town, very few users consult it. However, a REPRODUCTIBLE issue is that, when a user consults the weather station during the data upload, the USB lockup issue arrives. From my humble opinion, a way to explore could be that: even if the RapidFire mode is disabled, the "weather station chain" try to answer to a request, which can lock the USB receiver since an upload is already in progress.
Again, it's an hypothesis, not a developer approach. If I can help to do tests or reproduce the issue, please let me know.

thunderdome2, 2017-08-11:
Here is a small screenshot displaying the weather stations around me. The first one is mine.
[image]
But the RapidFire is disabled in the weewx.conf file:
`grep -i rapid /etc/weewx/weewx.conf`
`        # Set the following to True to have weewx use the WU "Rapidfire"`
`        rapidfire = False`

tkeffer, 2017-08-11:
I am now following your hypothesis. Are you saying that if someone browses your station on the Weather Underground while an update is in process, it causes a USB lockup?

Or, am I completely misunderstanding you?

thunderdome2, 2017-08-11:
You're perfectly understood.
In fact, since the USB lockup issue has also been pointed out with other softwares, I'm quite curious to know if someone can try to reproduce it with Cumulus or another one, namely browsing WeatherUnderground.com during upload and monitor if he sees a USB lockup some minutes after.

tkeffer, 2017-08-11:
Well, demonstrating such a link would be quite extraordinary. It's hard to see how a coupling would work between your RPi, the Weather Underground's database server, and their webserver that could cause this. But, you're welcome to knock yourself out trying to prove it.

I suppose it's possible that if you did the browsing on your RPi it would create enough of a load on it that it could cause the usb to lock up. But, that would happen with *any* browsing, not just to the Weather Underground.

In any case, this list is for demonstrable bugs in weewx. There's nothing here that suggests there's a bug in the code, so I am closing this issue.

closed 2017-08-11 by tkeffer