weewx/weewx #259: MySQL write failure loses record

tkeffer, 2017-06-01:
If MySQL goes offline, weewx will raise a `WeeWxIOError`, which causes the program to restart from the top. Unfortunately, this means the record at hand is lost. 

Alternatively, it could reset the database connection and try again in the driver. It should probably limit the retries to 60 seconds or so.

tkeffer, 2017-06-25:
I did some experiments with retrying the write of a record. Unfortunately, if the MySQL server goes away, the connection does have to be remade --- it's not enough to simply retry the write.

This isn't easy with the present architecture of weewx. Database connections are considered scarce resources so they are recycled and held in caches. So, the cache has to be invalidated, or the reconnection has to be made in a manner that is transparent to the cache. Neither is simple.

I'm thinking of just doing away with the database cache, but that will require some more experimentation.

dl1rf, 2017-07-24:
This is an important issue to my setup.
The MySQL is running at a NAS station which is connected via wlan to my Raspberry Pi with weewx.
My wlan is stable with 99%, but in the passed I often missed to shut down my weewx while updating NAS firmware which causes rebooting and than I lost the MySQL connection.
So I started to think about a cache mechanism at my PI, if MySQL connection is gone away.
My wish is that weewx shall not restart, but run anyway and cache archive packets and shall try to reestablish connection to the MySQL database.
If reestablished weewx shall post the cached archive records to MySQL and delete them at the Pi.
So far my theory.
I am thinking and playing here.
And you know it better, that's not an easy task.
I have some ideas how to do and will try them.
Maybe we can discuss?

moonpyk, 2020-09-02:
Maybe another way to take the things : if the database connexion is lost, write the record on the disk at a known location in a serialized format. 
When weewx takes back normally with the actual system, it could look in that place for pending save records and save them back. 

This kind of pattern is implemented in a lot of MQTT related libraries to recover from application crashes without losing any message.

tkeffer, 2020-09-02:
LOL! Yes. Maybe it could write the records to say... a sqlite database?

moonpyk, 2020-09-02:
Hehe, why not 🤣 
I'll have a bit of time during the weekend and encountered this issue yesterday while doing maintenance on my MySQL server. 
Luckily for me, nothing got lost, the Davis Vantage took back where it was, the things wouldn't have been the same with another type of station. 
I'll make a code proposal, without going to the sqlite extreme, we could have something nice, rugged and implying only simple changes in the engine.

tkeffer, 2020-11-17:
One option is to queue archive records until they can be written to the database, *then* kick off the reporting cycle. Something similar would have to be done for all the RESTful services. Then there are some services (wxservices comes to mind), that need access to the database in order to do their work. They cannot wait until the database is available and would need to either use old data, or synthesize old data with the current records in the queue.

I am concluding that fixing this would require a near total rewrite of weewxd. Not worth it.

closed 2020-11-17 by tkeffer