weewx/weewx #261: Ensure that max and min values are not dropped in report graphs

jrleeman, 2017-06-22:
It appears that the downsampling for report graphics does not preserve maximum and minimum daily values, which are often of interest to track visually. In the example shown, the daily temperature maxed out at just over 100 F, but the on the weekly report graph, it appears to only get into the upper 90's.
[image]
[image]

tkeffer, 2017-06-22:
This is a misunderstanding of what you are looking at.

The daily graph uses no aggregation: every single point is plotted.

By default, the weekly graph uses one hour (3600 seconds) averages. So, it will average all values within the hour, which, naturally, will cause the loss of individual highs and lows.

If this is not what you want, you have two choices:

1. Do not do any aggregation for your weekly plot. Plot every single point. This will result in a lot of points, but your machine should be able to handle it.

2. Instead of plotting the average for each hour, plot the max and min as separate lines. This can be a little confusing to look at, but it will do what you want.

In any case, this is not a bug, and it's not clear what enhancement you are asking for.

jrleeman, 2017-06-22:
I see why the hourly averaging would cause the smoothing, attenuating the high and low values. I suppose the enhancement that would make the most sense to me would be to plot the hourly average except that the daily max or min occurs during that hour, then that value would be plotted.

This preserves the daily significant events, but does not require every single point to be plotted. Seemingly a sensible default. I believe this is the approach taken by folks like the Mesonet (example: http://www.mesonet.org/index.php/weather/meteogram/nrmn/30d) 

This would also help the presentation of the monthly/yearly graphics. For example, the yearly graphic in this scenario shows the maximum temperature at just over 80.

tkeffer, 2017-06-22:
> I believe this is the approach taken by folks like the Mesonet (example: http://www.mesonet.org/index.php/weather/meteogram/nrmn/30d)

I don't think so. Looking at the daily graph it shows points every 5 minutes. Max temperature on 21-June was 92.84 at 14:40. 

Looking at the 30 day graph you linked to, it shows 3 hour aggregations of some unknown type. The max value it shows is 91.796 at 14:00, so the max value was lost.

I get what you're saying. You're proposing a rather unusual aggregation type. Not only would an aggregation interval and type have to be specified, but also some sort of meta interval over which mins and maxes would also be inserted into the resultant series. 

Not a priority, but if you want to take a stab at a pull request... go for it. But, let me warn you: it's not a trivial project as the meta interval has to respect things like DST.

jrleeman, 2017-06-23:
I've bugged a few other meteorologists at work who all agree that loosing the max/min is not ideal. That being said the real issue is that averaging is occurring. The Mesonet (nor any professional organization I'm aware of) is not averaging that data - they simply downsample. The monthly plot shows the ob obtained at the top of the hour every 3 hours for example.

That being said, I'd advocate for that to be a standard practice:

* Daily - each data point plotted
* Weekly - data plotted every 15 minutes (<700 points)
* Monthly - Hourly data plotted (<750 points)
* Yearly - Every 3 hours plotted (<3000 points)

Or some semblance of that. That would likely also make the PR a bit more straightforward.

tkeffer, 2017-06-23:
It would be more straightforward, but it would also not do what you want. There is no guarantee that downsampling will preserve the daily min and max. Indeed, the Mesonet example that you referenced did not.

While these kinds of data don't have much high-frequency energy, subsampling also introduces a risk of spectral aliasing. Averaging does not.

tkeffer, 2017-06-23:
I don't know that it's "far preferable," but it's easy enough to add an aggregation type `last`. Indeed, it's "sort of" in there already, but not fully implemented.

jrleeman, 2017-06-23:
Wouldn't it make a sensible default? Averaging is very misleading and over such a large time span is physically invalid. In the case above the yearly graph shows the low 80's as the highest data point. That number is simply meaningless. Here's a look at the NCAR Mesa Lab which uses simple decimation https://www.eol.ucar.edu/cgi-bin/weather.cgi?site=ml&units=metric&period=monthly

 I'll try to get a PR on this, but it may be awhile.

tkeffer, 2017-06-23:
I guess I would disagree with this. Averaging is very meaningful and quite physically valid. It can be used for conservation of physical properties, such as heat or moisture.

By contrast, with subsampling, the value you get depends on which, of many, data points you happen to choose. If there is a lot of high-frequency energy within the aggregation interval, the retained value can vary wildly from interval to interval, solely because you are subsampling at different points in the amplitude cycle. This is called spectral aliasing. The results would be useless for conservation statements.

Having spent a career in the earth sciences, I have a bit of experience in this.

In any case, I have started completing the code for `last` aggregations. It's not very complicated.

jrleeman, 2017-06-23:
Our organization's (UCAR/NCAR) sole purpose is meteorological data distribution and visualization. I agree that averaging can be a useful tool, but am lost at how showing a graph that has a 20 F discrepancy in values based on the time window is useful in a meteorological context. The high frequency content in temperature is not really present and certainly not meaningful, especially in a simple visualization. Mathematically valid? Probably not, there are the aliasing problems you've mentioned, but they are immaterial for a plot like this.

jrleeman, 2017-06-23:
(Not trying to be argumentative here, just trying to get the visualization that most accurately represents the contents of the data. I really appreciate all the work in WeeWx and find it a most useful utility!)

tkeffer, 2017-06-23:
Look at it this way: as the amount of high-frequency energy vanishes, the average value and the subsampled value will converge. The only difference is that the former is conserved, the latter is not. The only way there could be a 20F discrepancy is precisely the case where there is a lot of high-frequency energy.

I know NCAR well --- I did a sabbatical there.

Beautiful place!

In any case, don't worry: you will be able to subsample! I'm just not ready to make it the default.

jrleeman, 2017-06-23:
Excellent. If you ever make it down Boulder way again, let me know. I owe you a beer or two for all of your work on this package.

tkeffer, 2017-06-25:
Fixed in commit https://github.com/weewx/weewx/commit/07bc50b7ccf2dd31505891893e40d0290239d46d

Give it a try. The file [`manager.py`](https://raw.githubusercontent.com/weewx/weewx/07bc50b7ccf2dd31505891893e40d0290239d46d/bin/weewx/manager.py) should be a drop in replacement unless you're running a really old version of weewx.

closed 2017-06-25 by tkeffer

jrleeman, 2017-06-26:
Thanks! I'll get in on my system and give you a shout if there are any issues!