weewx/weewx #266: Make addRecord syslog log level configurable from weewx.conf

rjsparks, 2017-08-11:
LOG_NOTICE seems wrong as a default for addRecord - it's a normal and frequently event. I would suggest a default of LOG_INFO, but even better would be to add the plumbing to allow that log level to be controlled from the conf file.

matthewwall, 2017-08-11:
i agree.  we have log_failure and log_success options for restful and reporting.  perhaps we should do that for archive as well?

the log_failure option is useful for restful services where the destination may not be available, but you don't care.  log_failure is probably less desirable for the archive service.  so maybe just a log_success = (true|false) option for archive service?

it should default to logging each addRecord - that is very helpful for diagnosing noob issues.

tkeffer, 2017-08-11:
I agree. It really should be `LOG_INFO`. 

I would much rather have a few course "logging" or "debug" levels that can be selected from the .conf file rather than allow logging to be turned on and off for individual services. 

Perhaps the value of DEBUG should select the logging level:

| DEBUG | Log level |
|--|--|
|0 | LOG_NOTICE
|1 | LOG_INFO
|2 | LOG_DEBUG

tkeffer, 2019-10-06:
Version 4 will use the Python `logging` module, and will be able to do this, albeit using a slightly different approach. See the Wiki article [WeeWX V4 and logging](https://github.com/weewx/weewx/wiki/WeeWX-v4-and-logging).

closed 2019-10-06 by tkeffer