weewx/weewx #269: cannot run on systemd with user weewx

g-vidal, 2017-08-24:
Hi all,
I have tried to  configure weewx with systemd.
I changed a little bit the proposed conf file to fit my debian stretch standard installation but I am getting an error due the lack of of permissions fo weewx user to access. I have tried various suggestions from the web (adding an udev rule...) but nothing worked.
has anybody  a clue ?

```shell
weewx.service - weewx weather system
   Loaded: loaded (/etc/systemd/system/weewx.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Thu 2017-08-24 17:32:24 CEST; 59s ago
  Process: 5690 ExecStart=/usr/bin/weewxd --daemon --pidfile=/var/run/weewx/weewx.pid /etc/weewx/weewx.conf (code=exited, status=4)
 Main PID: 5690 (code=exited, status=4)

août 24 17:32:24 raspwife3 weewx[5690]: te923: sensor map is {'extraBatteryStatus1': 'bat_2', 'outTemp': 't_1', 'outHumidity': 'h_1', 'extraHumid4': 'h_5', 'uvBatteryStatus':
août 24 17:32:24 raspwife3 weewx[5690]: te923: Found device on USB bus= device=
août 24 17:32:24 raspwife3 weewx[5690]: te923: release interface failed: [Errno 13] Access denied (insufficient permissions)
août 24 17:32:24 raspwife3 weewx[5690]: te923: Unable to claim USB interface 0: [Errno 13] Access denied (insufficient permissions)
août 24 17:32:24 raspwife3 weewx[5690]: import of driver failed: [Errno 13] Access denied (insufficient permissions) (<class 'weewx.WeeWxIOError'>)
août 24 17:32:24 raspwife3 weewx[5690]: engine: Unable to load driver: [Errno 13] Access denied (insufficient permissions)
août 24 17:32:24 raspwife3 weewx[5690]:     ****  Exiting...
août 24 17:32:24 raspwife3 systemd[1]: weewx.service: Main process exited, code=exited, status=4/NOPERMISSION
août 24 17:32:24 raspwife3 systemd[1]: weewx.service: Unit entered failed state.
août 24 17:32:24 raspwife3 systemd[1]: weewx.service: Failed with result 'exit-code'.
```

Thanks

g-vidal, 2017-08-24:
Auto-answer (partial)
Found a udev.rules that seems to work but weewx.service has to be changed, not yet found the right parameter to have /var/run/weewx directory created at launch of the service.

tkeffer, 2017-09-16:
I suspect the issue was that you uncommented the 2 lines 

```
#User=weewx
#Group=weewx
```

which told `systemd` to run as the user `weewx`. However, that user did not have access to the USB port. You fixed the problem by giving the user access.

This is explained in the Wiki section on [systemd](https://github.com/weewx/weewx/wiki/systemd)

Closing this issue as no bug is involved.

closed 2017-09-16 by tkeffer