weewx/weewx #274: Wrong solar radiation values in weewx

Blisk, 2017-09-16:
On my WeatherRanger which is clone of ambientweather, it show real data on my console, but on weewx I get insane values like this  - Solar Radiation   9580 W/m²
on console I get 46,19 W/m²

Is there a settings in weewsx to change to get real data?

tkeffer, 2017-09-16:
It's possible your console is emitting lux. The conversion factor between lux and W/m² is 0.005, about the factor you are seeing.

Matthew is the expert on this.

Blisk, 2017-09-16:
Conversion factor on my console is 126,7lux and console shows right value, and I have set to show w/m2, it must be something in weewx which gives wrong value. But really don't know where to search for solution.

matthewwall, 2017-09-17:
which firmware is the weatherranger running?

please post the raw output from the station.  it should look something like this:

ID=XXX&PASSWORD=XXX&tempf=20&dewptf=18...&action=updateraw&realtime=1&rtfreq=5

you can see this raw output by setting debug=1 in weewx.conf then restarting weewx.  the raw station output will be in log lines that begin with 'raw data: '

in particular, i need to see the 'light', 'radiation' or 'solarradiation' field along with the other parameters.  some stations report 'light' when in fact the value is solar radiation, not luminosity.  seems to be due to buggy firmware, but to make the interceptor driver work properly i need to know how different firmware versions behave.

matthewwall, 2017-09-17:
taking this back to weewx-user so that others may comment or share their station firmware.  probably not a bug in the interceptor as much as not knowing how each firmware version behaves.

closed 2017-09-17 by matthewwall