weewx/weewx #276: Tag for whether it's daylight or not

tkeffer, 2017-09-23:
User Antony K. proposed a tag like

    $current.isDaylight

Seems reasonable.

windcrusader, 2019-04-15:
I'm keen to help with this project. I have moderate Python skills and have been tinkering with my own install of WeeWX for a couple of months. This seems like an easy enough project to start with. 

I guess the methodology would be checking to see if the current time is less than the sunset time and greater than the sunrise time? There would be some complexity I imagine depending on whether the almanac was installed, but this would just be a simple check right?

What am I missing?

tkeffer, 2019-04-15:
That would be about it.

But, it occurs to me that a more general solution would be appreciated by the solar-power fans. Given latitude, longitude, and time-of-day, it would return the current theoretical solar radiation in W/m<sup>2</sup>, as something like

    $almanac.sun.radiation

If this value is non-zero, then it must be daylight, so it would satisfy the original requirement.

This would be an extension to `almanac.py`, which is probably the most sophisticated Python in WeeWX, so it would make a good learning experience if you want to get more proficient at the language!

It would also take a bit of research on how to calculate solar radiation.

I don't see it as a problem if this tag requires pyephem.

What think you? Are you up for it?

gjr80, 2019-04-16:
A couple of observations...

I think having this as a `$almanac` based tag makes sense 

Being a 'solar power fan' I take the point about using current theoretical solar radiation as a yardstick for whether it is day or night, but a default WeeWX install that includes pyephem will already include/calculate the field `maxSolarRad` in loop packets and archive records. In such cases the tag `$current.maxSolarRad` and derivatives will give access to this field in reports whether the schema includes `maxSolarRad` or not.

Being an 'astronomy fan' as well I like to deal with the various twilights. There are definitions for the various twilights based on how far the sun is below the horizon. If we based the tag on the sun's position relative to the horizon as the pyephem almanac does for `.rise` and `.set` we could extend the tag to show whether we are in daylight bounded by civil, nautical or astronomical twilight. Something like:

`$almanac.sun.is_day` - True if sunrise < now < sunset otherwise False
`$almanac($horizon=-6).sun($use_center=1).is_day` - True if morning civil twilight < now < evening civil twilight otherwise False
`$almanac($horizon=-12).sun($use_center=1).is_day` - True if morning nautical twilight < now < evening nautical twilight otherwise False
`$almanac($horizon=-18).sun($use_center=1).is_day` - True if morning astronomical twilight < now < evening astronomical twilight otherwise False

I believe current theoretical solar radiation will only be able to be calculated if pyephem is installed, so a theoretical solar radiation based approach will mean that non-pyephem users will not be able to distinguish day/night via a single tag. If you revert to the sun rise/set based approach non-pyephem users will have access to a single tag that will distinguish day/night just as they have access to non-pyephem based sun rise and set times (though of course they will not have access to the twilight based approach previously mentioned)

I know not too many folks will use the twilight based 'is day' (though I am not sure there is a big demand for a sun rise/set 'is day' either) and it will be more complex to implement but I think it's worthwhile doing. Likewise I think there is value in a sun rise/set approach to include non-pyephem users.

tkeffer, 2019-04-16:
Geez, I totally forgot about `maxSolarRad`! So, we already have something that satisfies the original requirements. Thanks, Gary, for reminding me!

But, it would be useful to indicate the sun's status. How about something analogous to `moon_phase`?

    $almanac.sun.phase

This would return one of `dark`, `astronomical twilight`, `nautical twilight`, `civil twilight`, or `light`.

gjr80, 2019-04-16:
I guess we do already have the ability, but doing something like:
```
#if $current.maxSolarRad.raw > 0
it's day time
#else
it's might be night time
#end if
```
doesn't seem very intuitive.

Something analogous to `moon_phase` would do the job, though but for the average user who just wants day/night and doesn't care about twilights `$almanac.sun.phase == 'light'` means day but to get night the user would need to use `$almanac.sun.phase == 'night' or 'twilight' in $almanac.sun.phase`. Again not too intuitive.

I think the the simplest test (or tag) should be what the average user will be using, eg:
`#if $almanac.sun.is_day` or `#if not $almanac.sun_is_day`
or
`#if $almanac.sun.phase == 'day'` or `#if $almanac.sun.phase == 'night'`

The more complex tag/code should be for the specialised user, eg:
`#if $almanac($horizon=-6).sun($use_center=1).is_day`
or
`#if $almanac.sun.phase == 'night' or 'astronomical' in $almanac.sun.phase`

Then again if it ends up messy/complex no matter how we do it we could always go back to strict sunset/sunrise only without the twilights. Just seemed like an opportunity to follow a bit of a standard we already had in place.

PeterQuinn925, 2019-11-02:
Can't something like sun_altitude be used for this?

closed 2024-06-07 by github-actions[bot]