weewx/weewx #279: One week plot aggregations should start on Sunday

tkeffer, 2017-10-03:
Presently, weekly plots show an aggregation over seven days, but the start day is almost arbitrary --- it's effectively the present day.

It should consistently aggregate starting on Sunday (or whatever is the locale's start of the week)

gjr80, 2017-10-04:
I am a little lost on this one, by saying:

> but the start day is almost arbitrary --- it's effectively the present day.

I presume you mean the current day of the week one week ago ie if today is Wednesday the 11th the start is Wednesday the 4th?

> It should consistently aggregate starting on Sunday (or whatever is the locale's start of the week)

Are you saying that if generated on a Wednesday the 'week' plot should have no more than 5 days (assuming of course that Sunday is used as the start of the week) of data plotted? This would seem to me to provide somewhat of an odd plot.

Perhaps I am missing the obvious, or something particularly subtle, but I am having trouble seeing the real issue here. I accept there is an inconsistency between a tag with an aggregation period of 1 week eg `$week.rain.sum` and what are known as the 'weekly' or 'week' plots. The former shows data since (usually) Sunday whereas the latter shows data over the last 7+ days. 

Would not the same issue exist for month and year? Maybe we could just change what we call the 'week' plots, perhaps '7 day' plots.

Of course if I have missed the issue completely....

tkeffer, 2017-10-09:
Sorry, I wasn't very clear, and my diagnosis was a bit off.

If you ask for a month plot with daily aggregations, you get this:
[image]

Note how all aggregations start and end at midnight. If you try it for a week plot, you get this:

[image]

Note how the weeks do not start on Sunday. They start on a Thursday.

gjr80, 2017-10-09:
Ok, understand, I took weekly plots to mean plots that cover 1 week in total where in fact the issue is the aggregattion period.

closed 2024-06-06 by github-actions[bot]