weewx/weewx #285: mixed content blocked when opening the smartphone pages

unriccio, 2017-11-21:
Hi,
I'm using weewx and I push the generated pages to a webserver serving https.
The smartphone pages of the standard skin includes css and js resources served over plain http.

Nowadays browsers complain or block this behaviour:
https://developer.mozilla.org/en-US/docs/Web/Security/Mixed_content
https://developers.google.com/web/fundamentals/security/prevent-mixed-content/what-is-mixed-content
(resulting in ugly and less functional pages)

Could you please include the jquery resources via https? (it should be painless when pages are served over plain http)
Do you prefer a pull request?

Thanks for weewx!
Riccardo

tkeffer, 2017-11-21:
I have to plead ignorance here, cause I don't understand the issues.

If you're pushing the pages up to the webserver using https, why are they mixed content when they are served back down? Are you using https for this? If so, wouldn't this be a local, certification problem?

And, where does jQuery fit in? The WeeWX standard skin doesn't use jQuery.

gjr80, 2017-11-21:
The standard skin smartphone templates include a http call to load some jquery css, visitors to a weeWX site that uses https will see mixed content warnings or it may be blocked when viewing the smartphone pages. It's just a simple matter of changing some http entries to https in the smartphone templates. Should have no effect on sites that use http only, the jquery css will just load via https. I have some time so will take care of it.

unriccio, 2017-11-21:
Thanks @gjr80 that's what I was meaning.

gjr80, 2017-11-21:
Smartphone html templates changed to use https to loade jQuery scripts/css vide commit https://github.com/weewx/weewx/commit/804ab433578b1b4017ac7ac837c2a1b47d5f7046.

http is used in all of the Standard skin top level html templates, some instances of which could be probably be changed to https, but none of which currently cause a mixed content warning/error (at least in my testing they didn't). In keeping with the "if it ain't broken don't fix it" mantra I will leve these alone.

closed 2017-11-21 by gjr80