weewx/weewx #286: wiki page for transfer from sqlite to mysql has invalid `wee_database --backfill-daily --binding=dest_binding`

gearoidgit, 2017-11-25:
The wiki [page](https://github.com/weewx/weewx/wiki/Transfer%20from%20sqlite%20to%20MySQL) has an invalid `wee_database --backfill-daily ` example.
Is there a way to to a transfer of the daily summaries to a new mySQL database ?
I'm happy to correct the wiki if there's a way to do this or if not to remove the section.

gjr80, 2017-11-25:
You are correct, the `wee_database` utility received a makeover in v3.7.0 and the `--backfill-daily` action was renamed `--rebuild-daily`. For the purposes of the SQLite to MySQL wiki entry it is simply a case of using `wee_database --rebuild-daily` Instead of `wee_database --backfill-daily`.

Anybody with a github account should be able to edit the wiki, so feel free to fix the error.

Just to clarify one point, `wee_database` does not actually transfer the daily summaries from one database to another, rather it rebuilds the daily summaries of the database referred to by the binding `db_binding` in `weewx.conf` (or the `wee_database` option `--binding` if used) from the archive data in the same database.

gearoidgit, 2017-11-26:
I've updated the wiki [section](https://github.com/weewx/weewx/wiki/Transfer-from-sqlite-to-MySQL#using-wee_database).  If another contributor reckons it's now correct I'll close this issue

gjr80, 2017-11-26:
There are two occurrences of `--backfill-daily` just above the heading 'Using wee_database', you might as well update those as well. I also noticed another unrelated typo; the reference to `archive_sqlite` in the second para after the 'Using wee_database' heading should in fact be `archive_mysql`. Would be appreciated if you fixed that one as well. Otherwise looks good to me.

gearoidgit, 2017-11-26:
I've completed those, I see `--backfill-daily` is also in the MySQL->sqlite page so I'll correct that too.

gjr80, 2017-11-26:
Looks good. Thanks very much for your help.

closed 2017-11-26 by gjr80