weewx/weewx #287: wunderfixer should use UTC times to find missing records

tkeffer, 2017-11-27:
See [this thread](https://groups.google.com/d/msg/weewx-user/YoKGSiUpRZY/Mf9kop8HBQAJ).

Right now, wunderfixer uses the first returned column to figure out what records are missing. This time is in *local time*, as determined by the WU. If there is a difference between the time offset held by the WU and the time offset used by the wunderfixer host, then the calculated time in UTC will be wrong.

Instead, wunderfixer should use the last column, which is in UTC.

andrewmoni, 2017-11-28:
As this is probably a rare occurrence - an alternative might be to allow the wunderfixer command line to have an optional 'host offset' parameter where the user can specify an offset of wunderfixer host to data source.

tkeffer, 2019-07-19:
No longer an issue with commit https://github.com/weewx/weewx/commit/32c35ce380b766054b05a7b64d1d82ab113a5d73, which uses the new WU API to download records. 

Also, see issue #414

closed 2019-07-19 by tkeffer