weewx/weewx #288: weewx wmr300 not reading data

scotta1980, 2017-11-29:
hi am using the latest weewx but cant get weewx to read the wmr300 weather station error below 

Nov 29 10:13:15 raspberrypi weewx[1868]: wmr300: reading records since ******* N/A *******     (    N/A   )                    
Nov 29 10:13:15 raspberrypi weewx[1868]: wmr300: usb failure: [Errno 110] Operation timed out                                  
Nov 29 10:13:15 raspberrypi weewx[1868]: engine: Caught WeeWxIOError: [Errno 110] Operation timed out                          
Nov 29 10:13:15 raspberrypi weewx[1868]:     ****  Waiting 60 seconds then retrying...

matthewwall, 2017-12-24:
a pre-release of wmr300 is now available as version 0.19rc1 on the master branch at commit 1135f093

merry christmas!

matthewwall, 2017-12-31:
wmr300 version 0.19rc2 has been running reliably now on armhf ubuntu 16.04.03

```
dpkg --list | grep usb
ii  libusb-0.1-4:armhf 2:0.1.12-28 armhf   userspace USB programming library
ii  libusb-1.0-0:armhf 2:1.0.20-1 armhf    userspace USB programming library
ii  python-usb         1.0.0~b2-2 all      USB interface for Python
ii  usbutils           1:007-4    armhf    Linux USB utilities
```

feedback from users with these systems would be helpful:
```
any x86_64 systems
any armel systems
python-usb 0.4.x
libusb-0.1 (no libusb-1.x installed)
libusb-1.0.11 (ships with debian 7)
```

matthewwall, 2017-12-31:
request for feedback made to weewx-user group

closing this issue since it seems to be resolved

closed 2017-12-31 by matthewwall