weewx/weewx #290: Apply "stale" test to images

tkeffer, 2017-12-02:
See [this post](https://groups.google.com/d/msg/weewx-user/JybGVQfbFVA/ksq1Ui8IBgAJ).

Proposed fix:

```
--- imagegenerator.py.old	2017-11-23 00:36:48.000000000 +1100
+++ imagegenerator.py	2017-12-02 16:21:22.767704720 +1100
@@ -83,7 +83,19 @@
                 # Check whether this plot needs to be done at all:
                 if skipThisPlot(plotgen_ts, ai, img_file) :
                     continue
-                
+
+                # skip files that are fresh, but only if staleness is defined
+                stale = to_int(plot_options.get('stale_age'))
+                if stale is not None:
+                    t_now = time.time()
+                    try:
+                        last_mod = os.path.getmtime(img_file)
+                        if t_now - last_mod < stale:
+                            syslog.syslog(syslog.LOG_DEBUG, "imagegenerator: Skip '%s': last_mod=%s age=%s stale=%s" % (img_file, last_mod, t_now - last_mod, stale))
+                            continue
+                    except os.error:
+                        pass
+
```

gjr80, 2017-12-05:
`stale_age` support added to image generator at commit https://github.com/weewx/weewx/commit/9213582d76e21cadcb5b182156875f7d9c4b8206, that was easy. I've had the view for a while that there is a disconnect between the `stale_age` and `report_timing` options. Whilst one operates at the report engine level and the other at the generator level, both exert varying degrees of control over when something is generated in the report cycle, yet there is no linkage in the Customization Guide (arguably appropariate since until this morning `stale_age` only applied to cheetah generated files). In fact when this issue first came up Tom forgot about the `stale_age` option so posibly expecting a bit much of users to link/find the two.

We have a [Customizing the report generation time](http://weewx.com/docs/customizing.htm#customizing_gen_time) section in the Customization Guide, and given that `stale_age` now applies to both cheetah and image generators, I was wondering if we should mention the use of the `stale_age` option in this section (it only needs a paragraph). The `[CheetahGenerator]` and `[ImageGenerator]` sections further down should both list the `stale_age` option but with a short one liner that links to the coverage in the [Customizing the report generation time](http://weewx.com/docs/customizing.htm#customizing_gen_time) section.

Thoughts before I go ahead and make wholesale changes?

tkeffer, 2017-12-05:
Agreed.

Also, there's no reason why `stale_age` cannot propagate down from root level, just like any other `skin.conf` option (such as `HTML_ROOT` and others).

tkeffer, 2018-04-28:
Any progress on this? I'd like to close this issue.

gjr80, 2018-04-29:
Sorry no. It's only relatively minor documentation change so shouldn't take too long. I will get back onto it.

gjr80, 2018-06-15:
Added a note to `stale_age` and reworked the lead in para to _Customizing the report generation time section_ in the _Customization Guide_ in development. Would appreciate a proof read or Matthew's critical eye :)

evilbunny2008, 2018-06-15:
It might be worth while checking self.record is None and skip the stale check.

tkeffer, 2018-06-15:
@gjr80 looks good to me.

@evilbunny2008: Not sure what you're referring to. The image generator gets its data from the database. Indeed, because it is plotting *time series*, and not an individual record, it has to. Or, am I missing something?

evilbunny2008, 2018-06-15:
In #17 it was determined to ignore report_timing if wee_reports is being run, in stale_age there is no check currently if it was called by wee_reports, so images are skipped if they aren't deemed to be stale

tkeffer, 2018-06-15:
I get your point. 

But, I don't think it's so bad if the semantics of `stale_age` is still getting honored, even if `wee_reports` is what launched the image generation. That's what the Cheetah generator does.

closed 2018-06-27 by tkeffer