weewx/weewx #291: Caching for Rapidfire updates not working

tkeffer, 2017-12-25:
See [this thread](https://groups.google.com/d/msg/weewx-user/K0A_3U94U2k/rFHwYqVzBgAJ).

Looking through the log that Warren provided and picking out some random posts:

```
Dec 24 23:24:48 NODE1 user.debug weewx[24947]: restx: Ambient: url: http://rtupdate.wunderground.com/weatherstation/updateweatherstation.php?action=updateraw&ID=KAZPHOEN538&PASSWORD=XXX&softwaretype=weewx-3.8.0&realtime=1&humidity=019&dateutc=2017-12-24%2023%3A24%3A49&rtfreq=2.5&dailyrainin=0.00&tempf=67.1&windspeedmph=0.0&rainin=0.00
Dec 24 23:25:06 NODE1 user.debug weewx[24947]: restx: Ambient: url: http://rtupdate.wunderground.com/weatherstation/updateweatherstation.php?action=updateraw&ID=KAZPHOEN538&PASSWORD=XXX&softwaretype=weewx-3.8.0&realtime=1&dewptf=23.5&humidity=019&dateutc=2017-12-24%2023%3A25%3A07&rtfreq=2.5&dailyrainin=0.00&tempf=67.1&windspeedmph=2.2&rainin=0.00
Dec 24 23:25:24 NODE1 user.debug weewx[24947]: restx: Ambient: url: http://rtupdate.wunderground.com/weatherstation/updateweatherstation.php?action=updateraw&ID=KAZPHOEN538&PASSWORD=XXX&softwaretype=weewx-3.8.0&realtime=1&humidity=019&dateutc=2017-12-24%2023%3A25%3A25&rtfreq=2.5&dailyrainin=0.00&winddir=180&tempf=67.1&windspeedmph=2.2&rainin=0.00
Dec 24 23:25:26 NODE1 user.debug weewx[24947]: restx: Ambient: url: http://rtupdate.wunderground.com/weatherstation/updateweatherstation.php?action=updateraw&ID=KAZPHOEN538&PASSWORD=XXX&softwaretype=weewx-3.8.0&realtime=1&humidity=019&dateutc=2017-12-24%2023%3A25%3A25&rtfreq=2.5&dailyrainin=0.00&tempf=67.1&windspeedmph=2.2&rainin=0.00
Dec 24 23:25:43 NODE1 user.debug weewx[24947]: restx: Ambient: url: http://rtupdate.wunderground.com/weatherstation/updateweatherstation.php?action=updateraw&ID=KAZPHOEN538&PASSWORD=XXX&softwaretype=weewx-3.8.0&realtime=1&baromin=29.902&dewptf=23.5&humidity=019&dateutc=2017-12-24%2023%3A25%3A43&rtfreq=2.5&dailyrainin=0.00&tempf=67.1&windspeedmph=1.7&rainin=0.00
Dec 24 23:26:01 NODE1 user.debug weewx[24947]: restx: Ambient: url: http://rtupdate.wunderground.com/weatherstation/updateweatherstation.php?action=updateraw&ID=KAZPHOEN538&PASSWORD=XXX&softwaretype=weewx-3.8.0&realtime=1&humidity=019&dateutc=2017-12-24%2023%3A26%3A01&rtfreq=2.5&dailyrainin=0.00&winddir=202&tempf=67.1&windspeedmph=1.1&rainin=0.00
```

We can see that the partial packets have not been turned into full packets. For example, barometer (`baromin`) appears in only one of them.

matthewwall, 2017-12-25:
the root cause here is that StdWXCalculate inserts values for None even when it does not have any of the variables it needs to do a calculation.  for example, given a partial packet that does not contain any of pressure, barometer, or altimeter, it will insert None for each of pressure, barometer, and altimeter.

i think that the correct behavior should be to insert a None for pressure, barometer, and/or altimeter only if it has the variables to do a calculation, but those variables are None.

in this case, acurite stations report pressure.  for partial packets that contain no pressure, StdWXCalculate should not add any fields for pressure, barometer, or altimeter.  for partial packets that contain pressure, StdWXCalculate should enter a value for barometer and a value for altimeter, and that value should be None only if one of the independent variables on which they depend is None.

matthewwall, 2017-12-25:
i implemented a 'skip None' hack in the rapidfire caching at commit f2e114db

matthewwall, 2017-12-25:
then i implemented what i think is the fix for StdWXCalculate at commit 9164c084

matthewwall, 2017-12-25:
unfortunately, i seem to recall a discussion about the use of None in StdWXCalculate some time ago, and i fear that the 'no unnecessary None' fix (commit 9164c084) conflicts with the outcome of that discussion.

gjr80, 2017-12-25:
Yes, I seem to remeber that conversation (I think I had my fingers in there
somewhere). Unfortunately I am away from home so will have to trawl the
issues in github (if that is where it was discussed).

Gary

tkeffer, 2017-12-25:
As I recall, we decided that `None` was to signify that a sensor is present, but either not working, or emitting an invalid value. If a sensor is not present, then the observation value is simply not include in the packet.

So, your fix would be consistent with this philosophy.

BTW, why consolidate lines 375-379, but unconsolidate line 405?

matthewwall, 2017-12-26:
> BTW, why consolidate lines 375-379, but unconsolidate line 405?

sorry about the formatting changes.  its not about consolidate-everything or consolidate-nothing, its about formatting to fit 80-column screen in chunks that are easy to grok.

when i'm in the field, on the road, or working within virtual machines, my normal development environment is two 80-column terminals, one for editing, and one for tailing a log or running commands.

tkeffer, 2017-12-26:
I haven't used 80 columns since programming Fortran on an IBM 029 keypunch. :-)

I've clarified the rules for using `None` in commit https://github.com/weewx/weewx/commit/8c833cc66aa29470c6ea2e8d44973ae7b8fc57a2

tkeffer, 2017-12-26:
Can we close this issue?

matthewwall, 2017-12-26:
i reverted the rapidfire caching so that it will accept None values, since the real fix was in StdWXCalculate.  so this should be resolved as of commit 6a48f09a

closed 2017-12-26 by matthewwall