weewx/weewx #292: Adding extra sensor records for wundergrund

endmarsfr, 2017-12-28:
Hello ,
In "http://wiki.wunderground.com/index.php/PWS_-_Upload_Protocol" 
there is a way to add extra temperature sensors 
tempf - [F outdoor temperature] 
 * for extra outdoor sensors use temp2f, temp3f, and so on

I need to use it for adding for instance indoor temp in weewx
But I not found how to do with in conf file.
Could you please tell me how to do that ? 

regards

gjr80, 2017-12-28:
Hi,

WeeWX provides the ability to include  indoor temperature and indoor humidity in WU posts by setting the `post_indoor_observations` config option in `weewx.conf`. Unfortunately weeWX only supports posting **both** indoor temperature and indoor humidity not just one or the other. By default the  `post_indoor_observations` option is disabled. To post both indoor temperature and indoor humidity to WU place the following under `[StdRESTful] [[Wunderground]]` in `weewx.conf`:

```
post_indoor_observations = True
```

You will need to stop then start weeWX or do a config reload for the change to take effect.

Also, the github issues are for raising weeWX bugs etc, general support questions should be posted in the weewx-user Google group(https://groups.google.com/forum/m/#!forum/weewx-user)

Gary

endmarsfr, 2017-12-28:
Hi Gary, 

Thank for your answer.
I have added "post_indoor_observations = True" in my conf file.
I restarted weewx service, and signed in to my Wunderground account, but unfortunately, i not succeeded to display indoor temp and humidity  in Wunderground.
Maybe I missed something.

regards

tkeffer, 2017-12-28:
1. Make sure it's in the right location. It should be under the `[[Wunderground]]` subsection. So, you end up with...
    ```
    [StdRESTful]
    ...
        [[Wunderground]]
            ...
            post_indoor_observations = True
    ```

2. They will appear as Indoor and Outdoor temperature (not extra temperature sensors). Whether the WU will display them is another matter --- the WU is full of bugs. To make sure the WU is receiving your data, look on the "Table" tab, or even the "Download" tab. Don't rely on the graphs.

3. Along the same lines, "Extra temperatures" do not work on the WU. It displays them as soil moisture. See [this thread](https://groups.google.com/forum/#!searchin/weewx-user/upload$20extratemp1%7Csort:date/weewx-user/jUucXbjShdM/Jk1Z4W2LBAAJ) for the gory details.

Closing this issue as it is not a weewx bug. If you're still having problems, please continue on [weewx-user](https://groups.google.com/forum/#!forum/weewx-user)

closed 2017-12-28 by tkeffer