weewx/weewx #293: wee_device does not recognize Davis Envoy - (Unknown hardware type 79)

jbehrends, 2017-12-29:
I have a Davis Envoy (used with a Vantage Pro2) which weewx can pool just fine.  However when I attempt to use the wee_device utility to interact with it, I get the follow stack trace:

```
$ sudo wee_device --info
Using configuration file /etc/weewx/weewx.conf
Using Vantage driver version 3.0.10 (weewx.drivers.vantage)
Unable to wake up console... sleeping
Unable to wake up console... retrying
Traceback (most recent call last):
  File "/usr/bin/wee_device", line 54, in <module>
    main()
  File "/usr/bin/wee_device", line 51, in main
    device.configure(config_dict)
  File "/usr/share/weewx/weewx/drivers/__init__.py", line 69, in configure
    self.do_options(options, parser, config_dict, prompt)
  File "/usr/share/weewx/weewx/drivers/vantage.py", line 1968, in do_options
    station = Vantage(**config_dict[DRIVER_NAME])
  File "/usr/share/weewx/weewx/drivers/vantage.py", line 488, in __init__
    syslog.syslog(syslog.LOG_DEBUG, "vantage: Hardware name: %s" % self.hardware_name)
  File "/usr/share/weewx/weewx/drivers/vantage.py", line 1253, in hardware_name
    raise weewx.UnsupportedFeature("Unknown hardware type %d" % self.hardware_type)
weewx.UnsupportedFeature: Unknown hardware type 79
$
```

tkeffer, 2017-12-29:
I also use a Davis Envoy and have not had any problems with it. It uses the same hardware type ID as the VantagePro. Also, the same device driver is used in wee_database as in weewx itself, so that should not cause a difference.

Try running this small program. Stop weewx first.

```
import serial
port = serial.Serial("/dev/ttyUSB0", 19200)

port.write("WRD" + chr(0x12) + chr(0x4d) + "\n")

ack = port.read()
if ack != chr(0x06):
    print "Unable to get ACK. Got %d instead." % ack

htype = port.read()

print "Hardware type is %d" % htype
port.close()
```

You may have to adjust the port. I've set it to `/dev/ttyUSB0`. Yours may differ. 

I do not have access to my Envoy (I'm away from home), so I am unable to test this. We may have to fiddle to get the test program working.

Let me know what you find.

jbehrends, 2017-12-31:
I threw the above code into a script and ran it, and got an error:

```
$ sudo ./test.py
Traceback (most recent call last):
  File "./test.py", line 14, in <module>
    print "Hardware type is %d" % htype
TypeError: %d format: a number is required, not str
```

tkeffer, 2017-12-31:
Change the print statement to read

    print "Hardware type is %d" % ord(htype)

and give it another try.

jbehrends, 2017-12-31:
That worked!

```
$ sudo ./test.py
Hardware type is 76
$
```

I own two of these at two different sites, and ran the script against both, and both return the same value.

tkeffer, 2017-12-31:
Now I'm really confused. When run in WeeWX you're getting a hardware type of 79, but when running this simple program to just read the EEPROM, you're getting 76.

These are the standard Envoys, not an Envoy 8X right? That is, this [part](https://www.davisnet.com/product/wireless-weather-envoy/)?

Let me think about this a bit.

jbehrends, 2017-12-31:
Yup, that's what I'm using.  

So I just tried running "wee_device --info" again and it's now works!  

*sigh*  I haven't changed anything at all so I'm stumped as to why it now seems happy.  I'll close this issue and let you know if I see that happen again.

closed 2017-12-31 by jbehrends

tkeffer, 2017-12-31:
If you had one station I'd say you have a hardware problem. But, it's hard to explain getting the same results on two separate Envoys. I have no explanation.