weewx/weewx #300: NOAA format reports only display day/month data when there is `barometer` data for that day/month

gjr80, 2018-02-10:
Refer to [this user group thread](https://groups.google.com/d/topic/weewx-user/pN6KA5_N2S0/discussion).

gjr80, 2018-02-10:
Ok, not as easy as it first seems. `$month.dateTime.count.raw` won't work; `dateTime` does not have a daily summary table so you can't pull out a `count` aggregate on it. I guess that is why `barometer` was 'chosen'. Using `.has_data` seems like an obvious choice but `$month.dateTime.has_data` won't work, it comes up true every time becasue month iterates over all the months in the year. We could use another obs_type, say `outTemp` but then users who don't have any `outTemp` data for a day or month will be similarly affected. Left field solution; we could do away with the conditional and just override the `NONE` string format for the NOAA format reports so we don't display `N/A` on all fields without data. The downside is that we will evaluate a whole lot of aggregates that are presently not evaluated using the if conditional (though the number would decrease as we move through the month/year).

Or perhaps there is another solution I am mising......

tkeffer, 2018-02-10:
Oops. Didn't think of that.

The only idea that comes to me is to have a special check in `DaySummaryManager.has_data()` for type `dateTime`. Instead of doing a database lookup, it would instead see if `dateTime` is beyond the last date in the day summary (by checking meta-variable `lastUpdate`).

What the NOAA template is trying to accomplish is to see if there is *any* data for the given timestamp, so this honors that spirit.

I'm sure there's something I'm overlooking, but this should get you started.

gjr80, 2018-02-10:
I did briefly think of somehow using the `lastUpdate` meta-variable but whilst it will do the trick for a 'perfect database' anytime you regenerate old NOAA reports on a database with a gap of more than a day with no data you will end up with a row of N/As somewhere.

What if we implemented a `has_data` attribute for `TimeSpanBinder` objects, so we could do something like `$month.has_data` or `$day.has_data` that would return true if there was at least one data record (ie count > 0) for at least one obs type in the period concerned? If you recall we had a similar issue come up when we were patching the daily summaries from v1.0 to v2.0, from memory the algorithm I used was to iterate over all of the available obs daily summary tables (though I can't remember what test I did). It was a little cumbersome but you only need to iterate until you find a daily summary table with count > 0. In fact, since the NOAA format reports are only interested in a few obs types, you could have an optional parameter passed to `has_data` that tells `has_data` what obs types to look at, the default could be all. 

I know you are not a fan of complexity but.....

tkeffer, 2018-02-10:
I see your point about data gaps if we use `lastUpdate`.

Regarding your idea of introducing tags of the sort `$month.has_data`, that certainly is more intuitive than `$month.dateTime.has_data`.  

Perhaps I'm missing something, but I don't see why we have to iterate over all the observation types in the day summary to see if there is any data for the given day/month. Why can't we just see if there is a `dateTime` entry for the given time span? If there is a data gap, there shouldn't be one.

gjr80, 2018-02-10:
>  Perhaps I'm missing something, but I don't see why we have to iterate over all the observation types in the day summary to see if there is any data for the given day/month. Why can't we just see if there is a `dateTime` entry for the given time span? If there is a data gap, there shouldn't be one.

Yep, see what you mean, was focussing on the daily summaries but the data is the issue not the daily summaries. That will certainly pick up the cases where there is a gap in the data.

Another corner case to consider; data exists on a given day but there is no `outTemp` for the day (let's call this 'partial data'). `$day.has_data` returns true but `$day.outTemp.max` gives `None` so `N/A` is displayed for the days max temp (as is the case for all other 'day' temperature aggregates). All 'future' days of the month concerned will return false for `$day.has_data` so those lines (days) will be skipped (blank). Are we happy for days with 'partial data' to display `N/A` instead of blanks where the obs concerned is missing?  Don't want to seem pedantic but the original issue was due to missing (QC removed) `barometer` data so I think having one of the required obs missing is a realistic scenario to consider.

Thinking of another solution, maybe the NOAA report issue could be solved by changing the if condition to something like:
```
#if ($day.barometer.count.raw or $day.outTemp.count.raw or $day.rain.count.raw etc....)
```

A similar (but simpler) approach could be used for the other conditionals in the NOAA year report.

tkeffer, 2018-02-11:
The intention all along was to produce a good looking table that didn't have scores of rows with nothing but `N/A` for dates in the future. 

Your big, comprehensive conditional would do this. It makes explicit what is needed to show a useful row in the table. Plus, it's easy to implement.

gjr80, 2018-02-11:
Just had a good couple of hours on the ride-on mower to think about this, and in this case I agree the best solution is to work on the template. I've updated the conditionals in both the month and year NOAA templates, year has 3 conditionals each with a single condition. Month has 1 conditional with 3 conditions; so I don't think it has ended up too unwieldy. The result is empty historical periods are blank as are future periods. Periods for which there is some partial data (eg has `wind` but no `outTemp`) results in N/A for the non-existent data - I did not realise that was hard coded into the templates. Attached are Feb 2018 and 2018 NOAA reports for a db that has simulator data for 10 and 11 Feb but which contains no `outTemp` data. It is identical to the output from the unmodified NOAA templates but without the `barometer` dependency.

Depending on how pedantic you are heating and cooling degree days could require some finessing; when there is no `outTemp` data heating degree and cooling degree days are both 0.0. Maybe they should be N/A as well?

As for `$month.has_data`, implementing it by checking for a `dateTime` in the span should be trivial, implementing it would have no performance impact and I'm sure it will be of use to someone (inventing a problem for a solution?), so if OK with you I will code it IDC but put it in the development branch. As the template changes are trivial and fix a current bug, when agreed I will put them straight into master.

tkeffer, 2018-02-11:
Thanks, Gary

I'd rather put the code fix in the `development` branch, along with the template changes that use them. Nothing in the `master` branch. The reason is that we can't upgrade templates --- they are part of the skins. So, once we change them, we're stuck with them.

Alternatively, you could put everything in the `master` branch, but this fix will require a documentation change (the addition of the `$month.has_data` notation), which we try to avoid in `master`.

gjr80, 2018-02-11:
Tom,

Just to be clear the only change to the month template is to change:
```
#if $day.barometer.count.raw
```
to
```
#if $day.outTemp.count.raw or $day.rain.count.raw or $day.wind.count.raw
```
The implementation of `$period.has_data` is not used, no need with the modified conditional. Same deal in the year template. So the template, whilst changing, will be fully backwards compatible, transparent to the user and require no documentation changes.

Implementing `$period.has_data` is independent of the template changes and will require documentation changes; but as I said, it is really a solution looking for a problem but worthwhile nonetheless. It won't be used anywhere in the distributed weeWX package.

That was my reasoning for putting the template changes straight in to master and the new `has_data` into development. If you still want them in the same branch I will put them into development and just add a post to the originating thread pointing users there.

tkeffer, 2019-03-04:
Another user bitten by this: https://groups.google.com/forum/#!topic/weewx-user/TFylRQIuFpo

gjr80, 2019-03-04:
I have emailed user Michael with above mentioned modified NOAA templates for both Seasons and Standard skins. If they work OK and subject to a final check (been a while since I touched this) I will commit them to development.

Will get onto the `$period.has_data` in due course.

johnny2678, 2019-08-10:
I've been scratching my head trying to figure out why my NOAA reports weren't getting generated.  Stumbled on the issue and remembered that my barometer died around the same time reports stopped being generated.

Will try the fix posted above by @gjr80 and report back.

johnny2678, 2019-08-11:
> I've been scratching my head trying to figure out why my NOAA reports weren't getting generated. Stumbled on the issue and remembered that my barometer died around the same time reports stopped being generated.
> 
> Will try the fix posted above by @gjr80 and report back.

The fix worked... sort of.  After making the changes to the NOAA tmpl files in the Seasons skin, I know have NOAA report data for this month, but it still isn't populating last month's.  I lost my barometer on July 20, 2019 and the July NOAA report stops on that date.  August is now populating (just noted that a new daily record was created for today, 8/11). 

Do I need to do something else to get last month to populate after July 20?

gjr80, 2019-08-11:
> Do I need to do something else to get last month to populate after July 20?

Yes, you need to delete the July NOAA format report and all subsequent months reports (ie July and August 2019) as well as the 2019 report. WeeWX only generates NOAA monthly reports after the most recent monthly report that exists. The current months report is always generated. Likewise for yearly reports. You can force regeneration by deleting some/all NOAA format reports.

johnny2678, 2019-08-11:
Thanks @gjr80 ! Deleting the reports fixed the issue with the Monthly Reports.

Yearly report are still blank, sort of. The totals on the summary line are correct but the monthly entries are  blank:
```
                  PRECIPITATION (in)

                  MAX         ---DAYS OF RAIN---
                  OBS.               OVER
 YR  MO  TOTAL    DAY  DATE   0.01   0.10   1.00
------------------------------------------------
2019 01
2019 02
2019 03
2019 04
2019 05
2019 06
2019 07
2019 08
2019 09
2019 10
2019 11
2019 12
------------------------------------------------
         32.39   2.21   May     75     50     10
```

Is this the correct logic for the  Yearly tmpl file?
```
#for $month in $year.months
#if $day.outTemp.count.raw or $day.rain.count.raw or $day.wind.count.raw
$month.dateTime.format($YM) $month.rain.sum.format($Rain,$NONE,add_label=False) $month.rain.maxsum.format($Rain,$NONE,add_label=False)  $month.rain.maxsumtime.format($D,$NODAY) $month.rain.sum_ge($Trace).format($Count,$NONE,add_label=False) $month.rain.sum_ge($SomeRain).format($Count,$NONE,add_label=False) $month.rain.sum_ge($Soak).format($Count,$NONE,add_label=False)
#else
$month.dateTime.format($YM)
#end if
#end for
------------------------------------------------
        $year.rain.sum.format($Rain,$NONE,add_label=False) $year.rain.maxsum.format($Rain,$NONE,add_label=False)  $year.rain.maxsumtime.format($M,$NODAY) $year.rain.sum_ge($Trace).format($Count,$NONE,add_label=False) $year.rain.sum_ge($SomeRain).format($Count,$NONE,add_label=False) $year.rain.sum_ge($Soak).format($Count,$NONE,add_label=False)

```

johnny2678, 2019-08-11:
Whoops! Just re-read the code I posted above.  Didn't see the for loop was iterating using the  $month variable.

Replacing with the following works:
```
#if $month.outTemp.count.raw or $month.rain.count.raw or $month.wind.count.raw
```

tkeffer, 2019-11-28:
Fixed, using the solution proposed @gjr80. Will appear in V4.0

closed 2019-11-28 by tkeffer