weewx/weewx #301: reports should display local time even when computer clock is set to UTC

beaniecopter, 2018-02-14:
I would like reports to display local time, but my computer clock is set to UTC.  I am in the USA Central time zone.  A Davis VantagePro2 is reporting via USB to my desktop machine, which runs Debian GNU/Linux.  A work-around would be to use an old computer with the system clock set to local time.  Would the calibration methods used for temperature and wind direction work for time offset?

tkeffer, 2018-02-14:
I don't think that would work. You'd be messing with UTC times, trying to convince Python that the Greenwich meridian is now located in the central US.

Allowing an arbitrary time zone would require the time zone module `pytz`. Some clever engineering might allow `pytz` to be optional, but otherwise, this would mean introducing another dependency. All date objects would also have to become time zone aware. 

This is a low priority, but a pull request would be welcome if you want to give it a try. It would be a challenging project.

beaniecopter, 2018-02-14:
I thank you for the reply.  weewx is marvelous, nonethess, and I am grateful for the foresight and care and work which went into its design and implementation.  

But this is a bit beyond my programming capability, and an old machine is inexpensive...

tkeffer, 2018-02-14:
OK, we'll mark this `wontfix`, but it can be resurrected if someone wants to give it a try.

closed 2018-02-14 by tkeffer

gccobservatory, 2020-03-21:
I would also like to see this added.  My weewx runs on the same pc as my observatory control software, and I would prefer to have the computer clock set to UTC while the reports output in local time (EST/EDT for me).  Thanks.