weewx/weewx #311: ubuntu 18.04 breaks dpkg packaging of weewx

vinceskahan, 2018-04-02:
weewx 3.8.0 is currently packaged for dpkg systems in a way that requires 'python-imaging' which is not available in ubuntu 18.04 (due out late April 2018).   In addition python 2.7 is not installed by default on ubuntu-18.04 either.  Result is the dpkg variant of weewx won't install currently.

Suggest tweaking the dpkg requirements to require:

- python-pil          (to replace python-imaging)
- python-minimal (additional, in order to cause python 2.7 to install)

I did a little research and these tweaks seem to be ok on ubuntu 16.04, 17.10, and the current pre-release 18.10

vinceskahan, 2018-05-02:
update - did a little more research today under Docker.  It's ugly.

- debian7 has only python-imaging
- debian8+9 have python-imaging, which requires python-pil as a corequisite.  You can also select only python-pil and it'll just install that
- ubuntu 14.04 and 16.04 act like debian 8+9
- ubuntu 18.04 has only python-pil.  If you try to install python-imaging, it fails and suggests (but does not autoselect) python-pil

So it looks to me like you'll have to pick whether one dpkg will fail on old debian7 or new ubuntu-18.04 as there isn't a middle ground it seems.....

jeremyb31, 2018-05-05:
Please fix this for Ubuntu 18.04 as it's derivatives will be releasing new releases based on it soon

tkeffer, 2018-05-05:
> Please fix this for Ubuntu 18.04 as it's derivatives will be releasing new releases based on it soon

We're all volunteers here. If you want to speed things up, how about a pull request?

vinceskahan, 2018-05-05:
there are MANY workarounds for this one....

Install the imaging package:

   - install the imaging package your distro/version has available in it
   - or use 'pip install' to do so

Install weewx:

   - specify the ignore dependencies flag your command (apt, apt-get, or
   dpkg) has

jeremyb31, 2018-05-06:
Or changing the debian control file so that it is the following for depends
`Depends: python (>= 2.5), python-configobj (>= 4.5), python-cheetah (>= 2.0), python-imaging(>=1.1.6) | python-pil(>=5.1.0-1), python-serial (>= 2.3), python-usb (>= 0.4)`

Mori, 2018-06-10:
I have the same issue. Upgrading to Ubuntu 18.04 has bricked my weather station for now. I followed @vinceskahan 's advice above and installed the replacement imaging package with pip install. But now /etc/init.d/weewx just immediately exits with no effect whatever the command. I don't know how tot find a debian control file as @jeremyb31 suggsted, but I'll start crawling down that rabbit hole. 

Does anyone have any other suggestions for the "MANY workarounds" to this?

Thanks for your work on this software. It's been running great for years, until yesterday.

vinceskahan, 2018-06-11:
Impossible to say what you're experiencing without debug=1 logs or
systemctl logs.

If you follow the debian setup.py instructions, it'll tell you which file
to install from the sources to get startup files that should work.   Look
under the 'utils' directory in the sources for init.d and systemd variant
startup files.

vinceskahan, 2018-06-11:
https://github.com/vinceskahan/weewx-vagrant has a working provision.sh that you can run on a vanilla bionic VM and it should all work.   Works for me under vagrant.  Use that as your example for how to install using setup.py on bionic for the interim.

jeremyb31, 2018-06-11:
I installed python-pil and did sudo dpkg --ignore-depends=python-imaging -i weewx_3.8.0-1_all.deb

tkeffer, 2018-06-16:
V3.8.1 will address these issues, RSN.

matthewwall, 2018-06-22:
fix applied at commit 50bd96c665768facfd4024da30519be04bbd45c9

tested with dpkg/apt-get on a virgin, minimal ubuntu 18.04

closed 2018-06-22 by matthewwall