weewx/weewx #317: weewx www-site down

Calathia, 2018-04-19:
www.weewx.com has been down for a while. Any idea why?

matthewwall, 2018-04-19:
confirmed that weewx.com is not responding to ssh or http requests as of at least 05:30 on 19apr2018 (us eastern time)

tkeffer, 2018-04-19:
Works for me.

Frankly, this happened on Sunday as well. I ended up rebooting the VM. However, now I wonder if it was just being unresponsive, but still running.

tkeffer, 2018-04-28:
We think we've identified the problem as a runaway process that was using all the CPU cycles. 

Closing issue.

closed 2018-04-28 by tkeffer

MichaelAronsen, 2019-07-26:
this seems to be happning again?

tkeffer, 2019-07-27:
Tough crowd! 

The site was down for about 20 minutes this morning because of an unintended webserver "upgrade."

MichaelAronsen, 2019-07-27:
Well you told me to upgrade my weewx and since there are no mirrors, that was hard