weewx/weewx #326: report_timing implementation

evilbunny2008, 2018-06-10:
I added report_timing to my weewx config files and found that in Standard/skin.conf, doesn't seem to have an effect:

[CheetahGenerator]
    [[ToDate]]
        [[[lastyear]]]
            report_timing = @yearly
            template = lastyear.html.tmpl

The report occurs on every archive record. report_timing seems to slightly work under:

[ImageGenerator]
    [[day_images]]
        [[lastyear_images]]
            last_year = true
            report_timing = @yearly

It looks like it regenerated at midnight, rather than the first of January...

gjr80, 2018-06-11:
What you see is expected behaviour, the `report_timing` option is [only valid in `weewx.conf` and does not work in skin config files](http://weewx.com/docs/customizing.htm#_________The_report_timing_option_______). If there is a specific report or image that you wish to control using `report_timing` then these reports or images need to be included in their own skin and the `report_timing` option configured for that skin in `weewx.conf`.

evilbunny2008, 2018-06-11:
In that case, is there any reason that report_timing can't be expanded to work in skin.conf as well?

gjr80, 2018-06-11:
> In that case, is there any reason that report_timing can't be expanded to work in skin.conf as well?

When I first looked at implementing `report_timing` the first test code I put together actually worked at the `CheetahGenerator` level which meant the relevant config options could be used at the skin level. For various reasons; mainly technical, architectural and for simplicity the `report_timing` option code was put in the `StdReportEngine` which meant the skin level support for `report_timing` was not available. One of the key reasons was the availability (or rather lack of) of key weeWX properties/parameters in the `CheetahGenerator`. The fine grained control over individual templates could still be met by judicious allocation of templates to skins.

If you want any further info you can wade through issue #17.

Could it be revisited; I guess it could, but given individual template control already exists and all of the other tasks that exist it won't be too high on my priority list.

evilbunny2008, 2018-06-11:
I'm now generating reports for last month and last year, both of only need to be generated once a month or year respectively, yet the configs are nearly identical.

Having report_timing in weewx.conf the current code doesn't check if files exist or not and skips generating if the timing isn't matched. wee_reports will generate them regardless, but this doesn't seem user friendly.

I'll read through #17 and get back to you...

evilbunny2008, 2018-06-12:
I'm after your feedback on this to make sure I'm in the ballpark at least, I put the following code together in imagegenerator.py above skipThisPlot() call and seems to work. While stale_age is nice, more exacting methods of when to generate images and templates seem like a better idea to me, eg long running processes over night when no one is using the website.

```
                archive_interval = to_int(self.config_dict['StdArchive'].get('archive_interval', 300))
                timing_line = plot_options.get('report_timing', None)
                if timing_line is not None:
                    timing = weewx.reportengine.ReportTiming(timing_line)
                    if timing.is_valid:
                        if timing.is_triggered(plotgen_ts, plotgen_ts - archive_interval) is True:
                            syslog.syslog(syslog.LOG_DEBUG, "timing is valid, and so is is_triggered")
                        else:
                            syslog.syslog(syslog.LOG_DEBUG, "timing is valid, but not in timing_report")
                    else:
                        raise ValueError("Invalid report_timing values '%s'" % timing_line)
```

evilbunny2008, 2018-06-12:
> The code may well work but I think there are some higher levels issues that need to be clearly thought through. I am not quite clear here what problem is you are trying to solve. It seems you want plot by plot/template by template control by report_timing but two posts back you mention the current code does not check file existence and skips generating if timing isn't matched. That is indeed the case (file existence is not checked) and remains the case with your proposed code.

I only posted proof of concept code to see if it would work at all. All it does at the moment is output debug lines or an exception. For actual implementation the code in skipThisPlot() should be used.

I've not studied the current code beyond what could be reused, but I suspect it would be much more difficult to have file existence checking at that level from what I've read.

> How does this tie in with the other generators (except CopyGenerator), will they need similar changes? In particular FTP and Rsync are peculiar beasts.

At this stage I was only thinking of templates and graphs, that is things in the standard skin, as other generators need their own report settings so I hadn't considered them. 

> You now effectively have two sets of 'CRON-like control', one at service level and one at generator level, what is the interaction? The service reaches down into skin.conf for certain config information, if report_timing is such that the decision is not to generate then the service skips the entire report, not just the template/plot. Is that an issue?

The testing I've done with report_timing added to various places in the config files. The timing information seems to flow down to the plot level.

I don't think of it as multiple cron instances, but more like multiple settings in cron. The code for stale_age already creates a similar conundrum.

gjr80, 2018-06-13:
Such an implementation also has the potential to impact on how `wee_reports` operates. The decision was taken that during development of `report_timing` that `wee_reports` would ignore/not use `report_timing`, essentially because `wee_reports` is typically used to generate reports/plots for a particular point in time. I am not sure that the generators know whether it is `StdReport` or `wee_reports` that has called them.

evilbunny2008, 2018-06-14:
After some more digging it seems you can tell if wee_reports called by checking self.record is/isn't None, and this works regardless if a timestamp is given or not.

```
                archive_interval = to_int(self.config_dict['StdArchive'].get('archive_interval', 300))
                timing_line = plot_options.get('report_timing', None)
                if not os.path.exists(img_file):
                    syslog.syslog(syslog.LOG_DEBUG, "imagegenerator: file doesn't exist, skip report_timing checks")
                elif timing_line is not None and self.record is None:
                    syslog.syslog(syslog.LOG_DEBUG, "imagegenerator: self.record is None, skip report_timing checks as this is most likely being run from wee_reports")
                elif timing_line is not None:
                    timing = weewx.reportengine.ReportTiming(timing_line)
                    if timing.is_valid:
                        if timing.is_triggered(plotgen_ts, plotgen_ts - archive_interval) is True:
                            syslog.syslog(syslog.LOG_DEBUG, "imagegenerator: timing is valid, and in is_triggered range")
                        else:
                            syslog.syslog(syslog.LOG_DEBUG, "imagegenerator: timing is valid, but not in timing_report range")
                    else:
                        raise ValueError("imagegenerator: Invalid report_timing values '%s'" % timing_line)
```