weewx/weewx #329: Weather underground and "stale data"

RPeters52, 2018-06-19:
It appears that WU is now posting zeros for data that is missing in a post from weewx.  IF the base has lost contact with the weather station for some time we see the following on the log:

Jun 12 15:49:35 raspberrypi weewx[969]: acurite: Found station at bus=001 device=004
Jun 12 15:49:35 raspberrypi weewx[969]: acurite: R1: ignoring stale data (rssi indicates no communication from sensors): 01 0e 7a 71 02 62 00 3e 00 00
Jun 12 15:49:35 raspberrypi weewx[969]: acurite: next read in 18 seconds
Jun 12 15:49:53 raspberrypi weewx[969]: acurite: Found station at bus=001 device=004
Jun 12 15:49:53 raspberrypi weewx[969]: acurite: R1: ignoring stale data (rssi indicates no communication from sensors): 01 0e 7a 71 02 62 00 3e 00 00
Jun 12 15:49:53 raspberrypi weewx[969]: acurite: next read in 18 seconds

so some part of weewx knows there is NO DATA.  In this case, if weewx connects with WU and transmits NO WEATHER DATA, I get zeros posted on WU for that time period.  It would be better if weewx did NOT connect with WU and we had missing data.  (Plus it seems silly to connect and transmit "NO WEATHER DATA")

See the discussion at https://groups.google.com/forum/#!topic/weewx-user/jynPPrd8YIY for lots more info, plots, etc.

tkeffer, 2018-06-19:
As I outlined in the thread, this is an incorrect diagnosis. The Acurite is a *partial packet station*. That means, not every data type appears in every packet. A cache in restx.py makes up for this, but, if the Acurite does not emit temperature often enough, eventually the cached values are declared stale, and the post to the WU will omit it. However, *other values are transmitted.*

So, it is not true that "NO WEATHER DATA" is transmitted. Data are transmitted, just not any stale, missing data.

RPeters52, 2018-06-20:
HOWEVER, in this particular case there is NO connection between the base and weather station for about an hour and I have seen longer gaps (interference from a neighbor or ????) - see the attachments on the initial post  So, I would suggest, that in this case, ALL of the data was stale.  If all of the data is "stale", then maybe we should "do something"

tkeffer, 2018-06-20:
What do you mean, "we"?

RPeters52, 2018-06-20:
OK, you are correct -- I was very imprecise.  I will try again.  

It would be nice if the weewx software could  take care of this and similar "special cases"  I have spent about 40 years writing software in Fortran, Basic, Lisp, C and C++, and Java -- but not python.  So I have some idea of the problems associated with "fixing" software for "special cases", let alone the problems associated with precisely defining the "special cases".  I cn help with "debugging" as I have done a lot of it for my own software and other software that I have found on the web -- e.g. LibreOffice

tkeffer, 2018-06-20:
Sorry, I was being ornery. I just don't like my time being taken for granted.

In any case, I've put in a fix with commit https://github.com/weewx/weewx/commit/e04a17e6721e076cfbd9a6408f114bf715bd7f32, to appear in V3.9 some day.

You use it by listing observation types in section `[[[Essentials]]]`:
```
    [[Wunderground]]
        # This section is for configuring posts to the Weather Underground.
        ...        
	[[[Essentials]]]
	    outTemp = True
```

The implementation is very general, but it is only invoked for the Weather Underground.

If you want to test it (and I wish you would), do a checkout from branch `development`.

This summer represents my 50th year of programming. Old timers...

RPeters52, 2018-06-20:
No problem!  I did some "re-calculating" and discovered that I wrote my first Fortran program at college in the Fall of 1970, almost 48 years ago -- whew now I feel old  :^)

I will give your fix a try and report back to you.  THANKS!!!
Ralph

RPeters52, 2018-06-20:
I use .deb files on my raspberry pi to install weewx, so I have not spent any time figuring out where files go.  I am sure that I can figure it out, but first I thought I would ask if you had any simple suggestions for doing this "upgrade"

sycophantic, 2018-06-20:
I think this should be extended to CWOP/APRS posts too, the same behavior is observed with blank posts.

tkeffer, 2018-06-20:
> After doing some looking, it appears that replacing
> bin/weeutil/weeutil.py
> bin/weewx/restx.py
> in my current 3.8.0.1 install may allow me to test your fix. Correct???
> 

Correct.

RPeters52, 2018-06-20:
NO joy Tom....

With 3.8.0.1
At about 9:45am, I powered down the acurite weather station for about 5 minutes to break connection and we see that happened in both WU and weewx plots
[weewxTail.txt](https://github.com/weewx/weewx/files/2120778/weewxTail.txt)
[image]
[image]

I then powered it up and saw everything return to normal.
I stopped weewx.
I installed the 2 new files (restx & weeutil) by deleting the old ones and replacing them with the dev files.
I started weewx and watched it run for a while.
At about 10:30am, I powered down the weather station to see what would happen.  Essentially, on the WU site I saw the same thing as before.

I have attached the tail of the syslog, a copy of WU site, and the weewx plot.  I may have screwed something up, so please take a look.  Questions??  Drop me a note!!

Ralph Peters

RPeters52, 2018-06-20:
A final check on the 2 new files....

pi@raspberrypi:~ $ ls -las /usr/share/weewx/weewx/r*
36 -rw-r--r-- 1 root root 33630 Nov 22  2017 /usr/share/weewx/weewx/reportengine.py
24 -rw-r--r-- 1 root root 20516 Dec 30 16:12 /usr/share/weewx/weewx/reportengine.pyc
76 -rw-r--r-- 1 root root 75551 Jun 20 09:57 /usr/share/weewx/weewx/restx.py
56 -rw-r--r-- 1 root root 55285 Jun 20 10:00 /usr/share/weewx/weewx/restx.pyc

pi@raspberrypi:~ $ ls -las /usr/share/weewx/weeutil/*
16 -rw-r--r-- 1 root root 13050 Nov 22  2017 /usr/share/weewx/weeutil/ftpupload.py
12 -rw-r--r-- 1 root root  8711 Dec 30 16:15 /usr/share/weewx/weeutil/ftpupload.pyc
 4 -rw-r--r-- 1 root root   142 Nov 22  2017 /usr/share/weewx/weeutil/__init__.py
 4 -rw-r--r-- 1 root root   163 Dec 30 16:12 /usr/share/weewx/weeutil/__init__.pyc
 4 -rw-r--r-- 1 root root  1417 Nov 22  2017 /usr/share/weewx/weeutil/Moon.py
 4 -rw-r--r-- 1 root root  1571 Dec 30 16:12 /usr/share/weewx/weeutil/Moon.pyc
 8 -rw-r--r-- 1 root root  6592 Nov 22  2017 /usr/share/weewx/weeutil/rsyncupload.py
 8 -rw-r--r-- 1 root root  4843 Dec 30 16:15 /usr/share/weewx/weeutil/rsyncupload.pyc
20 -rw-r--r-- 1 root root 18207 Nov 22  2017 /usr/share/weewx/weeutil/Sun.py
20 -rw-r--r-- 1 root root 18189 Dec 30 16:12 /usr/share/weewx/weeutil/Sun.pyc
52 -rw-r--r-- 1 root root 52687 Jun 20 09:58 /usr/share/weewx/weeutil/weeutil.py
52 -rw-r--r-- 1 root root 52189 Jun 20 10:00 /usr/share/weewx/weeutil/weeutil.pyc

tkeffer, 2018-06-20:
Hmmm. I think I know what happened. Try this version of `restx.py`: https://raw.githubusercontent.com/weewx/weewx/development/bin/weewx/restx.py

RPeters52, 2018-06-21:
Still no joy  I disconnected power to the weather station between about 7pm and 7:30pm.  I have attached the syslog for a bit more that this time period.

[tail2.txt](https://github.com/weewx/weewx/files/2121711/tail2.txt)

tkeffer, 2018-06-21:
I guess I don't understand what you did. You say you "disconnected power to the weather station," yet the station is clearly running from your log.

sycophantic, 2018-06-21:
Seems to be working for me with fileparse.

`Jun 21 13:19:23 weewx weewx[10293]: restx: Wunderground-PWS: Skipped record 2018-06-21 09:19:00 EDT (1529587140): Observation type outTemp missing`

RPeters52, 2018-06-21:
Hi Tom,

Again, i appears that I have not been clear enough in my communications -- sorry about that.  I will try to clearly and completely describe the situation in "excruciating" detail.

In order to reproduce my problem, I have disconnected power to the outdoor acurite weather station, but NOT the indoor acurite weather base that communicates with the weather station.  So, there is zero data flowing from the weather station to the base station,  The base station is connected to the raspberry pi computer running weewx.  The weewx plots clearly show there is no communication between the base and the weather station (see bottom plot in attached file) and no outdoor weather data -- at least there is a gap in the plots on 3 occasions yesterday.
[image]  

It appears that the restx module in weewx is continuing to send data to WU, even though some other module in weewx understands that the outdoor weather data is stale and (probably) does not put this stale outdoor weather data into the database to plot at a later date.  Again, the latter is an assumption on my part based on looking at the plots.

This situation is the same as the original problem that I reported at the start of this discussion  - "a break in communication between the acurite weather station and its base station which results in bad data at WU".  

Have I answered your question clearly?  If not, try again!!  Sometimes it is difficult to discuss a problem using "email" where a face-to-face discussion would be easy.... 

Thanks again for working on this!!!

Ralph

tkeffer, 2018-06-21:
Try this version: [`restx.py`](https://raw.githubusercontent.com/weewx/weewx/bd73b3819b300dbf39c47f8489e05dce59ddfd98/bin/weewx/restx.py)

It will log the "essentials" dictionary.

Make sure you run with `debug=1`

RPeters52, 2018-06-21:
And the results are attached.
[tail3.txt](https://github.com/weewx/weewx/files/2124687/tail3.txt)
I did a listing of weewx.conf and put it at the end of this file.  It has debug=1

I have attached the relevant WU page also
[image]

RPeters52, 2018-06-21:
So, i looked in syslog and found

Jun 21 10:35:58 raspberrypi weewx[14437]: engine: Loading service weewx.restx.StdWunderground
Jun 21 10:35:58 raspberrypi weewx[14437]: restx: WU essentials: {}
Jun 21 10:35:58 raspberrypi weewx[14437]: restx: Wunderground-PWS: Data for station KNMALBUQ335 will be posted
Jun 21 10:35:58 raspberrypi weewx[14437]: engine: Finished loading service weewx.restx.StdWunderground
Jun 21 10:35:58 raspberrypi weewx[14437]: engine: Loading service weewx.restx.StdPWSweather

RPeters52, 2018-06-22:
OK, now we are getting somewhere....
I vaguely remember you saying something about modifying weewx.conf

I looked on the raspberrypi
   locate weewx.conf
/etc/weewx/#weewx.conf.cur#
/etc/weewx/weewx.conf
/etc/weewx/weewx.conf-3.5.0
/etc/weewx/weewx.conf-3.6.1
/etc/weewx/weewx.conf-3.6.2
/etc/weewx/weewx.conf-3.7.1
/etc/weewx/weewx.conf-3.8.0
/etc/weewx/weewx.conf.1
/etc/weewx/weewx.conf.2
/etc/weewx/weewx.conf.35
/etc/weewx/weewx.conf.cur
/etc/weewx/weewx.conf.current
/etc/weewx/weewx.conf.dist
/etc/weewx/weewx.conf.dpkg-dist
/etc/weewx/weewx.conf.old
/etc/weewx/weewx.conf.prev1
/etc/weewx/weewx.conf.prevSD
/etc/weewx/weewx.conf~
/etc/weewx/apache/conf.d/weewx.conf
/etc/weewx/logwatch/conf/logfiles/weewx.conf
/etc/weewx/logwatch/conf/services/weewx.conf
/etc/weewx/rsyslog.d/weewx.conf
/home/pi/weewx.conf.prevSD
/home/pi/Desktop/weewx.conf
/var/lib/dpkg/info/weewx.conffiles
/var/lib/dpkg/info/weewx.config

 I think that I need to modify my /etc/weewx/weewx.conf  file.   I think that I need to add the following to weewx.conf

  [[Wunderground]]
        # This section is for configuring posts to the Weather Underground.
        ...        
	[[[Essentials]]]
	    outTemp = True

I guess that I goofed by not putting it all together....

I will give it a try and get back to you
Ralph

RPeters52, 2018-06-22:
so I made the modifications -----

    [[Wunderground]]
        # This section is for configuring posts to the Weather Underground.
        
        # If you wish to do this, set the option 'enable' to true,
        # and specify a station (e.g., 'KORHOODR3') and password.
        enable = true
        station = KNMALBUQ335
        password = 6g0uKl9nYxP1

        [[[Essentials]]]
	outTemp = True
	
        # Set the following to True to have weewx use the WU "Rapidfire"
        # protocol. Not all hardware can support it. See the User's Guide.
        rapidfire = False

##############################################################################

restarted and I saw

Jun 21 18:58:25 raspberrypi weewx[24928]: manager: Starting backfill of daily summaries
Jun 21 18:58:25 raspberrypi weewx[24928]: engine: Finished loading service weewx.engine.StdArchive
Jun 21 18:58:25 raspberrypi weewx[24928]: engine: Loading service weewx.restx.StdStationRegistry
Jun 21 18:58:25 raspberrypi weewx[24928]: restx: StationRegistry: Station will be registered.
Jun 21 18:58:25 raspberrypi weewx[24928]: engine: Finished loading service weewx.restx.StdStationRegistry
Jun 21 18:58:25 raspberrypi weewx[24928]: engine: Loading service weewx.restx.StdWunderground
Jun 21 18:58:25 raspberrypi weewx[24928]: restx: WU essentials: {'outTemp': 'True', 'rapidfire': 'False'}
Jun 21 18:58:25 raspberrypi weewx[24928]: restx: Wunderground-PWS: Data for station KNMALBUQ335 will be posted
Jun 21 18:58:25 raspberrypi weewx[24928]: engine: Finished loading service weewx.restx.StdWunderground
Jun 21 18:58:25 raspberrypi weewx[24928]: engine: Loading service weewx.restx.StdPWSweather
Jun 21 18:58:25 raspberrypi weewx[24928]: restx: PWSweather: Posting not enabled.
Jun 21 18:58:25 raspberrypi weewx[24928]: engine: Finished loading service weewx.restx.StdPWSweather
Jun 21 18:58:25 raspberrypi weewx[24928]: engine: Loading service weewx.restx.StdCWOP
Jun 21 18:58:25 raspberrypi weewx[24928]: restx: CWOP: Posting not enabled.

so now I can go and test, I think...

Ralph

tkeffer, 2018-06-22:
Still not quite right. The `[[[Essentials]]]` entry is including `rapidfire`, which has nothing to do with the Essentials. 

Your stanza should look like:
```ini
[[Wunderground]]
    # This section is for configuring posts to the Weather Underground.
    
    # If you wish to do this, set the option 'enable' to true,
    # and specify a station (e.g., 'KORHOODR3') and password.
    enable = true
    station = KNMALBUQ335
    password = XXXXX

    # Set the following to True to have weewx use the WU "Rapidfire"
    # protocol. Not all hardware can support it. See the User's Guide.
    rapidfire = False

    [[[Essentials]]]
      outTemp = True

```

Now go change your password.

andylittle, 2018-06-22:
This seems to work, or perhaps I do not fully understand what is supposed to happen.  I grabbed the dev branch yesterday morning.  I have the sdr driver, when I pull the batteries from the temp sensor, no data gets posted now.  Before the update to dev branch temp was zero but other data was posted.   
        # protocol. Not all hardware can support it. See the User's Guide.
        rapidfire = False
        
        [[[Essentials]]]
            outTemp = True 
[wunderground page](https://www.wunderground.com/personal-weather-station/dashboard?ID=KCAMEADO21#history/s20180620/e20180620/mdaily)

The before and after is in this image. 
[image]

tkeffer, 2018-06-22:
That's exactly what is supposed to happen.

Closing the issue.

closed 2018-06-22 by tkeffer

RPeters52, 2018-06-22:
Both done --  fixed weewx.conf and the password.
I will try a test tomorrow.

THANKS!
Ralph

RPeters52, 2018-08-24:
Hi Tom!

Your fix for my acurite weather station works well.  I have had only a few
"signal loss" episodes, so I have not had many tests.

Is this fix in your latest release - 3.8.2?  If not, I will wait to update.

Thanks again for your help and your great software!
Ralph Peters

tkeffer, 2018-08-25:
No, it is not in v3.8.2. It will be in v3.9.

-tk

wimpie007, 2018-08-26:
Im having similar (?) problems since upgrading and moving my weewx from Zyxel NAS to a Pi.
Local generated reports (html) is ok, published data to WU not.
[image]

[image]

The published -17.8 defrees Celcius = 0 degrees fahrenheit, so every value was 0, but is not "cleaned" when uploading to WU. 
[StdQC]

    [[MinMax]]
        barometer = 26, 32.5, inHg
        outTemp = -40, 120, degree_F
        inTemp = 10, 120, degree_F
        outHumidity = 0, 100
        inHumidity = 0, 100
        windSpeed = 0, 120, mile_per_hour
        pressure = 24, 34.5, inHg
is in my config...

regards,

tkeffer, 2018-08-29:
When you say you "upgraded," which version did you upgrade to? This feature will be part of WeeWX Version 3.9, which has not been released yet.

tkeffer, 2018-08-29:
Now you've lost me. You "upgraded" to v2.6.4? That version is over four years old!

The ability to suppress a post to the weather underground if a data type is missing will not appear until v3.9.

wimpie007, 2018-08-29:
Sorry, 
i upgraded FROM 2.6.4. to "weewx-3.8.2". (as reported by WU)

tkeffer, 2018-08-30:
Thanks. You'll need to either use the 'development' branch of the repository, or wait until v3.9 comes out, in order to get this feature.

wimpie007, 2018-08-30:
Ok, no problem, i'll wait.
It just seems a bit strange to me that the [[MinMax]] config used to work with both local and WU data in the (very) old version, and in 3.8.2 not anymore.
Anyway, no biggie and i want to thank you for the great support and energy you are putting into this.
Appreciated!

RPeters52, 2018-11-08:
Hi Tom,

I used wunderfixer to update a day that had lots of missing data.  The data plots looked OK, but were a bit "thin".  

Wunderfixer worked fine, as always, except a few data points with zero-pressure, zero-temperature... were put up on the wunderground site by wunderfixer.  These bad data points corresponded to "NO-SIGNAL" times.

See https://www.wunderground.com/personal-weather-station/dashboard?ID=KNMALBUQ335#history/s20181103/e20181103/mdaily.  

Is there a chance that YOU, in your wizardly persona, could modify wunderfixer to not send data points that were "problematic".

Thanks,
Ralph Peters

tkeffer, 2018-11-12:
OK, wunderfixer now honors the `[[[Essentials]]]` section as well as of commit https://github.com/weewx/weewx/commit/91ab72d91433db871683456f2a22e80c7ca197ea. This will appear in V3.9.

RPeters52, 2018-11-13:
THANKS TOM!
BTW, do you any WAG when 3.9 will be out?

Ralph

tkeffer, 2018-11-13:
Working on it.

In the meantime, if there is some feature you really, really want, the `development` branch in GitHub is almost always runable.

RPeters52, 2019-02-07:
Hi Tom,
I updated to 3.9.1-1 on my Raspberry Pi and everything works.  I pulled the batteries on the outdoor weather station to simulate a loss of signal, waited for 30 minutes, restarted and the reporting to WU is fine.  

I tried wunderfixer and it seemed to do something unexpected (to me, anyway) in that it aborted.  I have attached a file showing the output from a "test" -- "wunderfixer -t -e 300" , and then a "fix" --  "wunderfixer -e 300".  No big problem for me, but I thought that you would want to see this.  

There was a statement on the test that surprised me:
"No results returned from Weather Underground (perhaps a bad station name??)."

Drop me a note when you have time,
THANKS,
Ralph
[wunderfixer.txt](https://github.com/weewx/weewx/files/2842708/wunderfixer.txt)

tkeffer, 2019-02-07:
Fixed in commit https://github.com/weewx/weewx/commit/ecc400a6d3984c9d728264e26b240ff644999a2b.

RPeters52, 2019-02-08:
Works great!  Thanks!
Ralph Peters