weewx/weewx #331: Customisation Guide instructions for _Customizing the Database_

gjr80, 2018-07-02:
The [current instructions](http://weewx.com/docs/customizing.htm#archive_database) cover a use case where a new module has been created to add a new field to the archive record. The instructions do not cover the other common use case where the user wants to add a field to the schema that already exists in the loop packet/archive record.

The existing instructions also have some language that appears to be a hangover from the old two database arrangement in place up until v3.0.0

gjr80, 2018-07-05:
Reworked the _Add a new type to the archive database_  section of the Customization Guide. Was a bit wordier than I hoped, would appreciate a someone giving it a review.

tkeffer, 2018-07-22:
Thanks, Gary. Hope you don't mind, but I simplified, and shortened, some of the writing in commit https://github.com/weewx/weewx/commit/bde766e9dd12f550f48bc3794f1bde00da7e468e

closed 2018-07-22 by tkeffer