weewx/weewx #334: weewx and SQL payload compression?

bctrainers, 2018-07-19:
Hi,

Does weewx support data compression of its payload to and from the mySQL server? If not, are there any plans for compression?

Right now, I'm running into the issue of where my weewx daemon (running on a virtual machine) is requesting data from my dedicated mySQL machine ranging from 85Mbps to 150Mbps for 15 seconds, once every minute. This appears to be causing the weewx daemon to hang while the data is being queried and returned.

With that being said, to further reduce bandwidth from the mySQL to the weewx server, are there any chances to make use of the [mySQL compression protocol](https://dev.mysql.com/doc/internals/en/compression.html)?

Right now, the SQL database for my primary weewx installation just recently surpassed one million rows in the archive table. All other table rows to the database is currently at 704. The size of the database is at 214MB.

tkeffer, 2018-07-20:
No plans to offer compression. I'd be surprised if it made any difference: all SQL queries are very simple and return very small result sets. 

A few questions:

1. What do you mean when you say your weewx daemon "hangs?" What are the symptoms?

2. How do you know that 85-150 Mbps of data is being requested? Is this on a sustained basis? Or, in bursts?

3. What kind of network is between WeeWX and the MySQL server?

Oh, and the "Issues" list is for bug reporting and tracking, not support. These kinds of questions should be directed to the [weewx-user group](https://groups.google.com/forum/#!forum/weewx-user).

bctrainers, 2018-07-20:
Hi,

My apologies on the issues list and this list. As for your questions...

1. The process while it is querying for mySQL data simply stops responding to given commands, be it signal (hup, term, kill, etc) or service commands. More or less, this is probably by design.

2. This is measured at the ethernet port and ip traffic from the running virtual machine, be it via ntop monitoring, and SQL traffic from the user. This is a constant value currently - once every minute. Which seems as if weewx is querying the entire archive database. Which would mean, with time, the data payload will increase notably year-by-year.

3. Weewx is running on a virtual machine - flawless running and stability. VM Server is on a LAGG, so 2Gbps to and from the switch. Switch is 24 port - all gigabit. The MySQL server is on its own dedicated box, with only 1Gbps ethernet active. VM machines are all on one VLAN (44), and the mySQL server is not on a VLAN currently. All hardware is in-house, nothing goes out to the net.

tkeffer, 2018-07-20:
1. All SQL queries are synchronous, so WeeWX will block until it gets the results set back. Or, until a timeout occurs. But, this is a matter of milliseconds, at the most. It should not "hang."

2. My database is at 1.1+ million rows and I've never noticed a problem with query times, despite using an ancient and slow processor. It is sqlite, not MySQL, so perhaps that makes a difference. The database is indexed by timestamp, and all queries are bounded by time, so unless you are asking for "all database" statistics, there is no reason why query time would grow as the database continues to grow in size.

3. Are you running any extensions? Or, just the Standard skin?

You're welcome to try and turn on compression. I'd be very interested in the results of the experiment. I would also be surprised if it made a difference.

tkeffer, 2018-08-14:
Closed, as we are unlikely to add this feature. However, experiments are welcome!

closed 2018-08-14 by tkeffer