weewx/weewx #335: `wee_database` does notproperly handle a read only SQLite database

gjr80, 2018-08-04:
Refer [this weewx-user thread](https://groups.google.com/d/topic/weewx-user/fZtyq_cTyZ4/discussion).

When using the `wee_database --drop-daily` action against a read only SQLite database `wee_database` incorrectly reports to console that the daily summary tables were successfully dropped.

gjr80, 2018-08-04:
The issue has wider implications than just read only SQLite databases. 

The `wee_database` drop-daily code calls the `DaySummaryManager drop_daily` method. This method traps any exceptions of type `weedb.OperationalError` and writes an appropriate entry in the log; however, the `weedb.OperationalError` exception is never raised so the `wee_database` code that calls the `drop_daily` method never sees the exception, despite including a suitable try..except..else statement, and consequently it believes that the daily summary tables were successfully dropped.`wee_database` notifies the user via console that the daily summary tables were successfully dropped.

It is worth noting that if the user were to monitor the WeeWX log the `weedb.OperationalError` exception would be noted, but most users would not monitor the log when using `wee_database`. it is also worth noting this issue does not only apply to read only SQLite databases, but also any error that results in a `weedb.OperationalError` exception being raised.

One fix is to simply have the `drop_daily` method re-raise the `weedb.OperationalError` exception once it is done processing the exception. Since `wee_database` already has appropriate exception handling code around the call to the `drop_daily` method there would be no need for change to `wee_database`. There would need to be change to `weecfg/database.py` as it contains a call to the `drop_daily` method with no exception handler. This should be easy to implement. There are no other calls to the `drop_daily` method that I could find in the WeeWX code base.

I will raise a PR with a suitable fix.

vinceskahan, 2018-08-04:
Gary and I looked into this a bit and I did some experimentation on a pi running Raspbian.

There's a bunch of stuff going on here it seems...

* there 'is' a syslog message going to /var/log/syslog
* but that message doesn't get displayed to the user running wee_database interactively

* wee_database sets a syslog.LOG_UPTO mask at line 136 or so:

```
   # set syslog level based
    if int(config_dict.get('debug', 0)):
        syslog.setlogmask(syslog.LOG_UPTO(syslog.LOG_DEBUG))
    else:
        syslog.setlogmask(syslog.LOG_UPTO(syslog.LOG_INFO))
```

* but that doesn't get honored because the underlying manager.py drop_daily( ) at line 1588 or so hard-codes the logging severity there.  It's unclear to me if that's a bug or not in this case...

```
    def drop_daily(self):
        """Drop the daily summaries."""

        syslog.syslog(syslog.LOG_INFO,
                      "manager: Dropping daily summary tables from '%s' ..." % self.connection.database_name)
        try:
            _all_tables = self.connection.tables()
            with weedb.Transaction(self.connection) as _cursor:
                for _table_name in _all_tables:
                    if _table_name.startswith('%s_day_' % self.table_name):
                        _cursor.execute("DROP TABLE %s" % _table_name)

            del self.daykeys
        except weedb.OperationalError, e:
            syslog.syslog(syslog.LOG_ERR, "manager: "
                          "Drop summaries failed for database '%s': %s"
                          % (self.connection.database_name, e))
        else:
            syslog.syslog(syslog.LOG_INFO, "manager: "
                          "Dropped daily summary tables from database '%s'"
                          % (self.connection.database_name,))
```

gjr80, 2018-08-12:
After looking at this again I think we can leave out any changes to `weecfg/database.py`; a readonly database throws errors in `database.py` much earlier than the call to `drop_daily`. It is made clear to the user in the resulting log trace that the issue is a readonly database:

```
gary@stretch17:~$ /home/weewx/bin/wee_database --update
Using configuration file /home/weewx/weewx.conf
Using database binding 'wx_binding', which is bound to database 'archive_sqlite'
The update process does not affect archive data, but does alter the database.
Continue (y/n)? y
Preparing Interval Weighting Fix...
Daily summary tables are at version 1.0
Calculating interval weights...
Traceback (most recent call last):
  File "/home/weewx/bin/wee_database", line 793, in <module>
    main()
  File "/home/weewx/bin/wee_database", line 165, in main
    update(config_dict, db_binding, options)
  File "/home/weewx/bin/wee_database", line 552, in update
    weight_obj.run()
  File "/home/weewx/bin/weecfg/database.py", line 443, in run
    self.do_fix(_next_day_to_patch_ts)
  File "/home/weewx/bin/weecfg/database.py", line 538, in do_fix
    _cursor)
  File "/home/weewx/bin/weewx/manager.py", line 1601, in _write_metadata
    (key, value))
  File "/home/weewx/bin/weedb/sqlite.py", line 43, in guarded_fn
    raise weedb.OperationalError(e)
weedb.OperationalError: attempt to write a readonly database
```

We could pretty up `database.py` to handle these errors more gracefully, but its going to add a deal more complexity. Unless advised otherwise will leave that for another day/issue.

So resulting PR will simply add a one line raise statement to `manager.py`

gjr80, 2018-08-12:
Ok, that did not go as planned, commit rather than PR. @tkeffer can you please have a look at commit 2f05d248f43a73f4d35c140b23b3115aba603538

gjr80, 2018-08-13:
Revised `wee_database` `dropDaily()` console output and `manager.drop_daily()` log output.

When a readonly db is encountered:
```
gary@stretch17:~$ /home/weewx/bin/wee_database --drop-daily
Using configuration file /home/weewx/weewx.conf
Using database binding 'wx_binding', which is bound to database 'archive_sqlite'
Proceeding will delete all your daily summaries from database 'weewx.sdb'
Are you sure you want to proceed (y/n)? y
Dropping daily summary tables from 'weewx.sdb' ... 
Error 'attempt to write a readonly database'
Drop daily summary tables failed for database 'weewx.sdb'
```
corresponding syslog:
```
Aug 13 10:13:05 stretch17 wee_database[4718]: manager: Dropping daily summary tables from 'weewx.sdb' ...
Aug 13 10:13:05 stretch17 wee_database[4718]: manager: Drop daily summary tables failed for database 'weewx.sdb': attempt to write a readonly database
```

Daily summaries successfully dropped:
```
gary@stretch17:~$ /home/weewx/bin/wee_database --drop-daily
Using configuration file /home/weewx/weewx.conf
Using database binding 'wx_binding', which is bound to database 'archive_sqlite'
Proceeding will delete all your daily summaries from database 'weewx.sdb'
Are you sure you want to proceed (y/n)? y
Dropping daily summary tables from 'weewx.sdb' ... 
Daily summary tables dropped from database 'weewx.sdb' in 0.05 seconds
```

corresponding log output:
```
Aug 13 10:16:06 stretch17 wee_database[4731]: manager: Dropping daily summary tables from 'weewx.sdb' ...
Aug 13 10:16:06 stretch17 wee_database[4731]: manager: Dropped daily summary tables from database 'weewx.sdb'
```

When attempting to drop non-existent daily summaries:
```
gary@stretch17:~$ /home/weewx/bin/wee_database --drop-daily
Using configuration file /home/weewx/weewx.conf
Using database binding 'wx_binding', which is bound to database 'archive_sqlite'
Proceeding will delete all your daily summaries from database 'weewx.sdb'
Are you sure you want to proceed (y/n)? y
Dropping daily summary tables from 'weewx.sdb' ... 
No daily summaries found in database 'weewx.sdb'. Nothing done.
```

There are no corresponding syslog entries in this case as nothing was done.

gjr80, 2018-08-14:
Well I am at a loss to explain how that change crept in. Certainly did not explicitly make such a change, thought I may have installed something other than development on my VM but checked my history and I pulled development when installing WeeWX. Since changing to Linux I have not been entirely happy with my git client, the current one does not preview changes before commits. Probably need to change something, probably the operator.

Fixed at commit https://github.com/weewx/weewx/commit/5c59a31a2446e85eb31812b6fa4f5e23be81f295

tkeffer, 2018-08-15:
This all looks good. Shall we close this issue?

gjr80, 2018-08-15:
Suspect I need to change a few example output lines in the utilities guide. Have to duck out now but will do that this arvo and then close the issue.

vinceskahan, 2018-08-15:
Is it appropriate to close an issue if there hasn't been a weewx actual
release with the fix in it?

tkeffer, 2018-08-15:
Yes

gjr80, 2018-08-15:
Well that was easy, no `wee_database` output screen captures in the docs that I can find. Certainly not going to add any. Issue closed.

closed 2018-08-15 by gjr80