weewx/weewx #338: Problem with inTemp and inHumidity after upgrade to 3.8.2

barcik75, 2018-08-20:
Hi,
I have problem with inTemp and inHumidity after upgrade. My WS is WH 2900, I use driver user.interceptor. My packates from WS looks ok. in every 5 second I got all information from WS. I have set interval in 240s. In weewx.sdb I have all information in every 2 minutes except inTemp and inHumidyty. This data are inregular. One on 10, 14, 20 minutes.
Please a look on attachment file.
Have you any idea where is problem?
[image]

tkeffer, 2018-08-20:
Your archive interval is actually 120 seconds.

Most likely your station is not emitting a value for `inTemp` and `inHumidity` that often, so they cannot be included in the archive interval.

Run weewxd from the command line and see what it's actually emitting in the LOOP packets (marked with `LOOP` in the print out). I'm betting that values for `inTemp` and `inHumidity` are pretty rare.

barcik75, 2018-08-21:
Ok, I check it. But before update all works fine. And my WS emitting value for indoor enviroment every 30s.

barcik75, 2018-08-22:
Problem solved. The new installation has overwritten my modifications in the restx.py file regarding sending data to WU.

closed 2018-08-22 by barcik75

tkeffer, 2018-08-22:
If you want to modify something, you are better off subclassing, adding the functionality you need, then specifying that the subclass be used. This makes your modifications more future proof.

But, I don't see how a restx.py change would affect archiving. It has nothing to do with how things get put in the database.

barcik75, 2018-08-27:
Yes, but my WS (WiFi) has a hard-coded wunderground address under which to send data. I had to set this address in the local network so that the packets were captured by weewx (the user.interceptor driver). And in the restx.py file, replace the address rtupdate.wunderground.com with an IP address to make it work. When the original restx.py file was in place, the inTemp and inHumidity data was saved in a random period of time. Strange that the rest of the data was regularly recorded every 240 seconds.
After changing in the restx.py file, the addresses rtupdate ... to the IP address the problem has stopped appearing.