weewx/weewx #341: Add better support for data vectors

gjr80, 2018-09-07:
The [V3 Database Architecture document](https://docs.google.com/document/d/1CzuCPu7MsV3Rc3J6GbaMVxtrQ6z5F-xz0glvg3sUWjA/edit#), among other things, outlines how some aggregate tags could be adapted to return a vector of data and ways in which this data could be used in WeeWX reports through the introduction of new tags and the extension of some existing tags.

This issue is to track implementation of the above. (ie those concepts on the final page of the V3 Architecture document)

gjr80, 2018-09-07:
Created `json` branch off `development` for the code

Initial commit adds support for:
- `$aggregation_interval` parameter for `sum`,` count`,` avg`,` max`,` min` and `last` aggregates so you can now do:
```
$month.outTemp.min($aggregation_interval=86400)
```
to obtain a vector `ValueHelper` containing the daily minimums for the month to date:
```
-1.1°F,-1.1°F,-1.1°F,-1.1°F,-1.1°F,-1.1°F,-0.7°F,-0.9°F
```

- implemented `.csv` and `.json` optional formatting tags so something like:
```
$month.outTemp.min($aggregation_interval=86400).csv
$month.outTemp.min($aggregation_interval=86400).json
```
returns:
```
-1.1,-1.1,-1.1,-1.1,-1.1,-1.1,-0.7,-0.9
[-1.1102297312430447, -1.1102297312430842, -1.1102297312430645, -1.1102297312430467, -1.1102297312430705, -1.110229731243037, -0.7002893240889208, -0.9461363859357657]
```
- implemented `$fmt` and `$delimiter` parameters for `.formatted` formatting tag allowing use of:
```
$month.outTemp.min($aggregation_interval=86400).formatted
$month.outTemp.min($aggregation_interval=86400).formatted(fmt='delimited')
$month.outTemp.min($aggregation_interval=86400).formatted(fmt='delimited', delimiter=':')
```
to get:
```
-1.1,-1.1,-1.1,-1.1,-1.1,-1.1,-0.7,-0.9
-1.1,-1.1,-1.1,-1.1,-1.1,-1.1,-0.7,-0.9
-1.1:-1.1:-1.1:-1.1:-1.1:-1.1:-0.7:-0.9
```

First commit and still many features and refinements to go.

gjr80, 2018-09-08:
> Huh. That's simpler than I thought it would be.

Yes, I thought I was doing something wrong it was so easy.

> Why was it necessary to include .....

Early days master!

There are a number of issues with the code I have committed, for instance it will error on certain boundary conditions and there is an obvious discrepancy between `.csv` and `.json` output. That first commit was about getting the basics in there (`$aggregation_interval` for one). 

Just working on a lengthy post now to get your clarification on a few things on the JSON page....

gjr80, 2018-09-08:
Perhaps it was the time of day but having studied that last page for a few days now I think a few aspects need to be clarified. 

In the interest of brevity throughout this post I will use `$vector_vh` as representing some timespan based tag that returns a ValueHelper containing a vector of observational data (a 'vector ValueHelper'), eg `$month.outTemp.max($aggregation_interval=86400)` to give a vector ValueHelper of daily max `outTemp` observations for the current month.

Default vector ValueHelper 'output'

When a tag such as `$day.outTemp.min` is used in a report the output is a string something like `12.3°F`. The V3 document indicates that for a vector ValueHelper of the form `$vector_vh` the default output should be a string something like `[21.2, 21.5, 21.8], ‘degree_C’, ‘group_temperature’]`. A more consistent approach would be for `$vector_vh` to output a string `[21.2°F, 21.5°F, 21.8°F]`.

Providing a vector of observational data as a python object

A user who wishes to manipulate a scalar observation might use the `.raw` formatting tag to obtain the raw value of the datum (usually a number) with optional unit conversion so that it can be used in an equation or some other calculation/code. In reality the `.raw` formatting tag gives the user access to the value property of the ValueTuple in the ValueHelper returned by the base tag.

A similar approach should be supported for vector based data, whereby a user can access the raw value property of the ValueTuple in a vector ValueHelper. The existing `.raw` tag should provide access to the python list that is stored in the value property of the ValueTuple in the vector ValueHelper. Unit conversion should also be supported.

csv and json formatted data

The V3 document demonstrates two methods of providing json and csv data, one using the `.json` or `.csv`  format tag and the other using `$fmt='json'` or `$fmt='delimited'` as a parameter in a `.formatted` format tag. This could be seen as duplication but I think we can use the two approaches to meet slightly different requirements.

Firstly, the `.json` and `.csv` format tags could be used just as `.raw` to provide acces to the full precision data in the ValueHelper. `.json` would provide a python object that is fully compatible with the json to python object mapping used in the json module(ie a list in almost all cases, essentially the output of a `json.loads()`). `.csv` would provide similar data but formatted as a string of comma separated values. Optional unit conversion would be supported for both tags.

Secondly, `$fmt='json'` or `$fmt='delimited'` used as a parameter in a `.formatted` format tag would be used to again provide a json or csv format output, but in this case the output would be a string and usual unit conversion and datum formatting is used. With `$fmt='delimited'` the delimiter can be specified with the optional `$delimiter` parameter, the default being a comma. Other string format tags such as `.nolabel`, `.format` should also be supported.

Perhaps some examples may help:

`$vector_vh` would provide a string `[21.2°F, 21.5°F, 21.8°F, N/A]`
`$vector_vh.raw` would provide a list of numerics/None values `[21.236485, 21.521734, 21.88362, None]`
`$vector_vh.json` would provide a list of numerics/None `[21.236485, 21.521734, 21.88362, None]`
`$vector_vh.csv` would provide a string consisting of a comma separated list of numerics/None `21.236485, 21.521734, 21.88362, None`
`$vector_vh.formatted($fmt='json')` would provide a string representation of the data in json format (ie compatible with the output of json.dumps) `[21.2°F, 21.5°F, 21.8°F, null]`
`$vector_vh.formatted($fmt='delimited')` would provide a string representation of the data in comma separated value format `21.2°F,21.5°F,21.8°F,N/A`
`$vector_vh.formatted($fmt='delimited', $delimiter=":")` would provide a string representation of the data in comma separated value format with a colon as the delimiter `21.2°F:21.5°F:21.8°F:N/A`

I have not covered `$pretty_print` or `$header_row`, will deal with those later.

gjr80, 2018-09-09:
Another question to ponder. 

If it is the 8th of the month but we only have data for the 5th of the month and onwards and we use the tag `$month.outTemp.max(aggregation_interval=86400)` to get a vector of daily maximum temperatures for the month. What is the expected output:
```
[None, None, None, None, 67.4, 72.4, 76.4, 68.3]
```
or
```
[67.4, 72.4, 76.4, 68.3]
```
The former seems more intuitive to me, otherwise how do you know what period (in this case day) to which each datum belongs (have yet to develop a vector based maxtime); however, the current `getSqlVectors()` behaviour will give us the latter. What if the previous month has data or if it has no data (ie 5th of this month is the first day of data in our archive).

Interestingly, `$month.outTemp.count(aggregation_interval=86400)` will give us:
```
[0.000000, 0.000000, 0.000000, 0.000000, 288.000000, 288.000000, 288.000000, 167.000000]
```

gjr80, 2018-09-10:
JSON branch should be stable and basically working as intended as of commit 86433b31aac1420c92f0bbc6166ec973b3e8f449. The following code is an extract from the template I am using to test, it gives a fairly good spread over the available combinations and permutations:
```
#errorCatcher Echo
Time=$current.dateTime

##
## confirm .json provides a deserialised JSON object
#import json
#set $j = $month.outTemp.min($aggregation_interval=86400).json
#set $t = $j.__class__.__name__
#echo "checking .json operation\n"
#echo "$month.outTemp.min($aggregation_interval=86400).json returned an instance of %s\n" % ($t, )
#echo "deserialised JSON object=%s\n" % ($j, )
#echo "iterate over JSON object and check each element\n"
#for $e in $j
#set $t = $e.__class__.__name__
#echo "element=%s element is an instance of %s\n" % ($e, $t)
#end for
#echo "end of .json confirmation\n"
##

## confirm .csv returns a string
#set $c = $month.outTemp.min($aggregation_interval=86400).csv
#set $t = $c.__class__.__name__
#echo "checking .csv operation\n"
#echo "$month.outTemp.min($aggregation_interval=86400).csv returned an instance of %s\n" % ($t, )
#echo "$month.outTemp.min($aggregation_interval=86400).csv=%s\n" % ($month.outTemp.min($aggregation_interval=86400).csv, )
#echo "end of .csv confirmation\n"
##

month.outTemp.min=$month.outTemp.min
month.outTemp.min(aggregation_interval=86400)=$month.outTemp.min($aggregation_interval=86400)
month.outTemp.min(aggregation_interval=86400).csv=$month.outTemp.min($aggregation_interval=86400).csv
month.outTemp.min(aggregation_interval=86400).csv(delimiter=":")=$month.outTemp.min($aggregation_interval=86400).csv($delimiter=":")
month.outTemp.min(aggregation_interval=86400).json=$month.outTemp.min($aggregation_interval=86400).json

month.outTemp.min(aggregation_interval=86400).raw=$month.outTemp.min($aggregation_interval=86400).raw

month.outTemp.min(aggregation_interval=86400).string=$month.outTemp.min($aggregation_interval=86400).string
month.outTemp.min(aggregation_interval=86400).string(fmt='delimited')=$month.outTemp.min($aggregation_interval=86400).string($fmt='delimited')
month.outTemp.min(aggregation_interval=86400).string(fmt='delimited', delimiter=':')=$month.outTemp.min($aggregation_interval=86400).string($fmt='delimited', $delimiter=':')
month.outTemp.min(aggregation_interval=86400).string(fmt='json')=$month.outTemp.min($aggregation_interval=86400).string($fmt='json')

month.outTemp.min(aggregation_interval=86400).formatted=$month.outTemp.min($aggregation_interval=86400).formatted
month.outTemp.min(aggregation_interval=86400).formatted(fmt='delimited')=$month.outTemp.min($aggregation_interval=86400).formatted($fmt='delimited')
month.outTemp.min(aggregation_interval=86400).formatted(fmt='delimited', delimiter=':')=$month.outTemp.min($aggregation_interval=86400).formatted($fmt='delimited', $delimiter=':')
month.outTemp.min(aggregation_interval=86400).formatted(fmt='json')=$month.outTemp.min($aggregation_interval=86400).formatted($fmt='json')

month.outTemp.min(aggregation_interval=86400).format("%.3f")=$month.outTemp.min($aggregation_interval=86400).format("%.3f")
month.outTemp.min(aggregation_interval=86400).format(format_string='0.2f', fmt='delimited')=$month.outTemp.min($aggregation_interval=86400).format("%.3f" ,$fmt='delimited')
month.outTemp.min(aggregation_interval=86400).format(format_string='0.2f', fmt='delimited', delimiter=':')=$month.outTemp.min($aggregation_interval=86400).format("%.3f", $fmt='delimited', $delimiter=':')
month.outTemp.min(aggregation_interval=86400).format(format_string='0.2f', fmt='json')=$month.outTemp.min($aggregation_interval=86400).format("%.3f", $fmt='json')

month.outTemp.min(aggregation_interval=86400).nolabel("%.2f")=$month.outTemp.min($aggregation_interval=86400).nolabel("%.2f")
month.outTemp.min(aggregation_interval=86400).nolabel("%.2f", fmt='delimited')=$month.outTemp.min($aggregation_interval=86400).nolabel("%.2f", $fmt='delimited')
month.outTemp.min(aggregation_interval=86400).nolabel("%.2f", fmt='delimited', delimiter=':')=$month.outTemp.min($aggregation_interval=86400).nolabel("%.2f", $fmt='delimited', $delimiter=':')
month.outTemp.min(aggregation_interval=86400).nolabel("%.2f", fmt='json')=$month.outTemp.min($aggregation_interval=86400).nolabel("%.2f",$fmt='json')

month.windDir.avg=$month.windDir.avg
month.windDir.avg(aggregation_interval=86400)=$month.windDir.avg($aggregation_interval=86400)
month.windDir.avg(aggregation_interval=86400).csv=$month.windDir.avg($aggregation_interval=86400).csv
month.windDir.avg(aggregation_interval=86400).csv(delimiter=":")=$month.windDir.avg($aggregation_interval=86400).csv($delimiter=":")
month.windDir.avg(aggregation_interval=86400).json=$month.windDir.avg($aggregation_interval=86400).json

month.windDir.avg(aggregation_interval=86400).raw=$month.windDir.avg($aggregation_interval=86400).raw

month.windDir.avg(aggregation_interval=86400).ordinal_compass=$month.windDir.avg($aggregation_interval=86400).ordinal_compass
month.windDir.avg(aggregation_interval=86400).ordinal_compass(fmt='delimited')=$month.windDir.avg($aggregation_interval=86400).ordinal_compass($fmt='delimited')
month.windDir.avg(aggregation_interval=86400).ordinal_compass(fmt='delimited', delimiter=':')=$month.windDir.avg($aggregation_interval=86400).ordinal_compass($fmt='delimited', $delimiter=':')
month.windDir.avg(aggregation_interval=86400).ordinal_compass(fmt='json')=$month.windDir.avg($aggregation_interval=86400).ordinal_compass($fmt='json')
```

matthewwall, 2019-01-26:
gary, could you provide the output that results from the example template code that you posted 09sep2018?

matthewwall, 2019-01-26:
you posted these as examples of what json output would look like:

> $vector_vh.json would provide a list of numerics/None [21.236485, 21.521734, 21.88362, None]
> ...
> $vector_vh.formatted($fmt='json') would provide a string representation of the data in json format (ie compatible with the output of json.dumps) [21.2°F, 21.5°F, 21.8°F, null]

but i would expect this:

$vector_vh.json
```
{"label": ["21.236485", "21.521734", "21.88362", ""]}
```

$vector_vh.formatted($fmt='json')
```
{"label": ["21.2°F", "21.5°F", "21.8°F", ""]}
```

or maybe i am not understanding your intent?  i thought the reason for json formatting is to emit json output that can be read directly by d3, highcharts, or other json-ready tools?

gjr80, 2019-01-30:
Let me address these in reverse order.

>  i thought the reason for json formatting is to emit json output that can be read directly by d3, highcharts, or other json-ready tools?

The impetuous behind me was Pat's Belchertown skin, it's use of Highcharts and the use of SLEs  from my [Highcharts extension](https://github.com/gjr80/weewx-highcharts) to provide the required JSON. My extension relied on the SLEs to gather the data and return the required arrays of (timestamp, data) pairs and a report template to assemble the required JSON file. The two limitations that became apparent were the complexity and inflexibility of the SLE and the need to excessive amount of manual JSON building in the template (again complexity and flexibility).

In picking up the 'vector thoughts' of V3 Database Architecture document (the 'V3 document') my intent was (1) do away with the need for the SLEs (except for maybe in the most complex/specialised cases) and (2) simplify the construction of a JSON output file. The intent was never to have a single tag $some_smart_tag.options($more_options=zzzz) that would produce the entire JSON file but rather a number of tags along with some literal text (or perhaps inline code if the author was capable) to produce JSON output. Since it was not a quantum leap to format an array of data in csv format we could leverage of the JSON tag requirements and support csv as well (as per the V3 document).

> but i would expect this:
>
>$vector_vh.json
>
>`{"label": ["21.236485", "21.521734", "21.88362", ""]}`
>
>$vector_vh.formatted($fmt='json')
>
>`{"label": ["21.2°F", "21.5°F", "21.8°F", ""]}`

Happy to discuss exactly what the JSON output should be. I most likely came to the table blinkered by my Highcharts extension approach where a lot of the JSON object was constructed in the report template. With hindsight I guess something that claims JSON output should be producing a JSON object whereas I have been producing a JSON array.

In fact, perhaps we need to step back a bit and define what the structure and resulting output of the different 'vector tags' will be. The V3 document uses a `.csv` option but a `.formatted($fmt='json')`; though there is no `.json` option. Should we have `.json` to provide provide a JSON object with no formatting applied as you use in your example; do we then have `.csv` that provides a non-formatted CSV? How do we get the `.raw` equivalent ie the data points are numeric full precision?

Maybe we almost need to go back to the JSON section of the V3 document and flesh it out further. Or perhaps I am reading too much into what you are asking or perhaps I am missing some of the subtleties of the V3 document.

> gary, could you provide the output that results from the example template code that you posted 09sep2018?

Um, ah yes, I have a VM to resurrect. But I suspect the above probably makes this somewhat superfluous now.

tkeffer, 2020-09-21:
Resurrecting this issue.

I started fiddling around with creating JSON vectors, but now I'm wondering if we even need them. I'm thinking `#for` loops work
just as well. For example, suppose we want a JSON vector of daily average temperatures for the month. This can be done with:

```
    [
    #for $day in $month.days
      [$day.start.raw, $day.outTemp.avg.format(add_label=False, None_string='null', localize=False)],
    #end for
    ]
```

This creates a JSON string like:

```json
[ [1598943600, 72.7] , [1599030000, 77.2] , [1599116400, 75.3] , [1599202800, null] , ... ,]
```

The formatting is a bit clumsy, but we could offer a shortcut formatting option, call it `json`:

```
      [$day.start.raw, $day.outTemp.avg.json],
```

Using loops can be more flexible than JSON created with the tag structure. For example, suppose you want a vector with *two* values, the daily minimums and maximums. This can't be done with the close-formed tag structure, but is easily done with a loop:

```
    [
    #for $day in $month.days
      [$day.start.raw, $day.outTemp.min.json, $day.outTemp.max.json],
    #end for
    ]
```

Unfortunately, my `for` loop has the unfortunate property of leaving a trailing comma (`,`) behind, but this can be fixed with:

```
    [
    #set $not_first=False
    #for $day in $month.days
    #if $not_first
    ,
    #end if
    #set $not_first=True
      [$day.start.raw, $day.outTemp.min.json $day.outTemp.max.json]
    #end for
    ]
```
which is kind of clumsy, but it works. It might be possible to come up with some Cheetah construct that does this for you.

Am I missing something here? Are there use cases that cannot be covered with a loop?

gjr80, 2020-09-21:
I have thought on an off about this since I noticed the demise of `getSqlVectors()`. I think you are probably right, in terms of constructing JSON format data that includes arrays we have a rich set of tags and iterators that makes it easy to construct the JSON format data directly in the cheetah template. Equally so for CSV format data. Any `.json` implementation was never going to avoid a degree of manual construction within the template.

One potential hole that I saw this issue resolving was some limitations in the formatting of `ValueHelpers` that contained a vector of data. This is going back some years now, but I (seem to remember I) found I could easily convert between units if I had a vector `ValueHelper`, but it was not so easy to format a vector `ValueHelper`. I would like a couple of days to work through this and bring my memory back up to speed.

Interestingly I have brought my HighCharts extension into the WeeWX 4 era in the last week or two, still a number of SLEs with numerous calls to `getSqlVectors()` and templates that are a pain to read. I think taking your approach I could simplify things a lot and retire a lot of expensive code. Might be a good test case to see how well it can support HighCharts which was one of the original drivers for this issue.

gjr80, 2020-09-23:
Hmm, this is a bit of fun. Have implemented `.json` in `class ValueHelper()` and have been using the following:
```
#set $elm_list=[]
#for $day in $month.days
  #set $elm = ",".join([str($day.start.raw*1000), $day.outTemp.min.json, $day.outTemp.max.json])
  #set $elm_list += ["".join(["[", $elm, "]"])]
#end for
#set $json=",".join($elm_list)
[$json]
```
which gives me:
```
[[1598882400000,-1.1,21.1],[1598968800000,-1.1,21.1],[1599055200000,-1.1,21.1], ...]
```
Some of the Highcharts JSON files are multi-level and I like to obtain the component vectors first before assembling a sub-element later, I find the structure of sub-element is much easier to follow/troubleshoot that way. As you can tell I'm a bit of a fan of `join`. The 1000 multiplier is due to Highcharts wants timestamps in the Javascript millisecond format.

I've tried various Cheetah constructs to try to to push down some of the ugly detail but have not found anything that has the flexibility I think is required. Cheetahs inline python support and directives don't seem to have the flexibility required. Do you have any ideas?

tkeffer, 2020-09-23:
It should be possible to refactor so `element` and `jsonise` are search list extensions. Doesn't need to be a WeeWX SLE; just a regular Cheetah SLE.

Try that.

gjr80, 2020-09-23:
I have always struggled with the operation of Cheetah. I understand what a WeeWX SLE is (and does) and I understand what the SearchList is wrt a Cheetah template, but what is a Cheetah SLE and where/how is one implemented? Happy for a ‘look at xxxxx’ or ‘read xxxxx’ answer.

tkeffer, 2020-09-23:
I just meant it's not necessary to go through the `get_extension_list()` business. They would be hardwired into `cheetahgenerator.py`.

```python
def jsonise(s_list):
  return "[%s]" % ",".join(s_list)
```

then in `CheetahGenerator._getSearchList()`, something like:

```python
searchList = [{'month_name' : time.strftime("%b", timespan_start_tt),
               'year_name'  : timespan_start_tt[0],
               'encoding'   : encoding,
               'jsonise' : jsonise}, 
              self.outputted_dict]
```

gjr80, 2020-09-23:
Thanks, looking in the wrong place again. That works fine, and we now have just four lines in the template to create the JSON element and of course one line to display it:
```
#set $outtemp_list=[]
#for $day in $month.days
  #set $outtemp_list += $element([str($day.start.raw*1000), $day.outTemp.min.json, $day.outTemp.max.json])
#end for
$jsonise($outtemp_list)
```
So maybe the solution is a few basic helper functions in the search list to make a JSON template a little less cluttered/more readable and the use of for loops to give the developer the necessary control. The penalty paid is that the construction of the overall JSON object is still largely a manual process, but one would argue we were never going to fully automate that.

tkeffer, 2020-09-23:
Exactly. I think that's the right way to look at it. The helper functions, plus a wiki article on how to use them.

Incidentally, I've learned that using the `$day` as the looping variable is risky, because it shadows the aggregation period `$day`.

gjr80, 2020-09-23:
> Exactly. I think that's the right way to look at it. The helper functions, plus a wiki article on how to use them.

I have started converting one of my Highcharts templates from a number of behemoth SLEs to the 'new' approach. Will see what helper functions that suggests.

> Incidentally, I've learned that using the `$day` as the looping variable is risky, because it shadows the aggregation period `$day`.
Yep, just another thing that plays with my head when working with Cheetah.

gjr80, 2020-09-24:
Hmm, bit of a performance hit. Converted my existing SLE based Highcharts template and on my RPi3 it takes 24 seconds to generate compared to 2.5 seconds for the SLE based version. The template provides [timestamp, data] pairs for 14 observations over 7 days at a 5 minute archive interval, so it's a fairly heavy template. I am guessing the vast majority of time is being taken up appending lists to lists - roughly 14 * 2 * 288 * 24 * 7 = 1 354 752 list appends. I guess you also pay a price for doing things in Cheetah rather than natively in python.

Dropped the period from 7 days back to 1 day and that reduced the time to generate to 6.3seconds.

tkeffer, 2020-09-24:
More likely it's the need to do 1,354,752 database lookups.

Did your SLE-based templates do a single lookup for a vector? That would involve just 14 lookups, possibly 14 * 2 if retrieving the timestamp is a separate query.

24 seconds is unacceptable. We may have to rethink this.

gjr80, 2020-09-24:
> More likely it's the need to do 1,354,752 database lookups.

I thought `genBatchRows()` hit the database once for all rows rather than once for every row.

> Did your SLE-based templates do a single lookup for a vector?

Yes, it used `getSqlVectors()` so each call gave you the timestamp and the data. So there were 14 `getSqlVectors()` calls.

> 24 seconds is unacceptable. We may have to rethink this.

Yes. I guess you need to draw the line somewhere as to what is acceptable/usable. Being able to generate 24 hours of unaggregated data for multiple obs would seem a definite must do, 7 days of unaggregated/multiple obs data is probably about the limit. I would expect a month on unaggregated/multiple obs data would be over the top, you should be aggregating by then.

gjr80, 2020-09-24:
If the issue is hitting the database then surely the solution is to take a vector approach to hitting the database. Am I correct in that we have no actual tag that produces a vector of data? If we had a tag for a vector we could use a similar approach to what we have been looking at; giving the user/developer a few simple tools to aid constructing the JSON object. In fact the template would be a lot cleaner as the iteration could be pushed into a Cheetah SLE rather than having it in the template as we have now.

tkeffer, 2020-09-24:
That's what I'm thinking, although I don't know what it would look like. 

Before I got distracted by the idea of using `for` loops I was working on tags that looked like

    $month.outTemp.min.by('day')

which would return a JSON vector of (start, stop, daily minimums). Maybe it should just return the raw data, which can then be iterated over?

Roughly (*very* roughly):

```
#for $data_min, $data_max in $zip($month.outTemp.min.by('day'), $month.outTemp.max.by('day')
   [ $data_min[0] , $data_min[2] $data_max[2] ],
#end for
```

gjr80, 2020-09-24:
> That's what I'm thinking, although I don't know what it would look like.

LOL, I started typing 'maybe something like this:' on my previous then deleted it when I realised I didn't know what it looked like!

> Roughly (_very_ roughly):
> 
> ```
> #for $data_min, $data_max in $zip($month.outTemp.min.by('day'), $month.outTemp.max.by('day')
>    [ $data_min[0] , $data_min[2] $data_max[2] ],
> #end for
> ```

That would seem to work, and it would lend itself very easily to a Cheetah SLE that is called with the vectors you want to zip as arguments:
```
$json_array($month.outTemp.min.by('day')[0], $month.outTemp.min.by('day')[2], $month.outTemp.max.by('day')[2])
```
Makes the template look much neater.

gjr80, 2020-09-24:
Actually, that I've done something different, should be:
```
$json_array($month.outTemp.min.by('day')[0], zip($month.outTemp.min.by('day')[2], $month.outTemp.max.by('day')[2]))
```

tkeffer, 2020-09-24:
Can you post one of the loops in your template? One that you think is taking a lot of time?

gjr80, 2020-09-24:
> Can you post one of the loops in your template? One that you think is taking a lot of time?
```
#*

Windchill plot

*#
## get our arrays
#set $windchill = []
#set $heatindex = []
#for $record in $seven_day.records
  #set $windchill += $json_array_element([str($record.dateTime.raw*1000), $record.windchill.json])
  #set $heatindex += $json_array_element([str($record.dateTime.raw*1000), $record.heatindex.json])
#end for
## now construct the windchillplot JSON object
"windchillplot": {"series":
{#slurp
"windchill": {"name": "Wind Chill", "data": $json_array($windchill)},
"heatindex": {"name": "Heat Index", "data": $json_array($heatindex)}},
"units": "$unit.label.windchill"#slurp
#if $varExists('$outTemp_min_range')
, 
"minRange": $outTemp_min_range},
#else
},
#end if
## cleanup our arrays
#set $windchill = []
#set $heatindex = []
```
`$seven_day` is a `TimeBinder` that covers 7 days. `$json_array_element` and `$json_array` are SLE versions of the `#def element` and `#def jsonise` from about 15 posts back. Ignore the "units" stuff onwards.

tkeffer, 2020-09-24:
Can you push a branch that has the definitions of `.json`, `json_array_element()`, etc? I'd like to do some timing runs.

gjr80, 2020-09-24:
Have pushed branch `json_template_master`. It was branched off `master` and should support `.json`, `$json_array_element()` and `$json_array()`.

tkeffer, 2020-09-24:
The branch `json_template_master` appears identical to `master`.

gjr80, 2020-09-24:
Helps if you commit before pushing a new branch, hang on a few minutes.

gjr80, 2020-09-25:
Should be there now

tkeffer, 2020-09-25:
Switching discussion to Slack. Check the #development channel

closed 2021-03-15 by tkeffer